geerlingguy/raspberry-pi-pcie-devices #112: KubeSail PiBox - Pi CM4 NAS Box

geerlingguy, 2021-04-12:
See: https://kubesail.com/blog/pibox-raspberry-pi-powered-nas

[image]

Still in prototype, but here's the synopsis:

> Today we're announcing the PiBox, a Raspberry Pi powered storage server capable of running tons of great software, like self-hosted alternates to Dropbox and Google Photos. We're happy ship it to you with a vanilla OS, or with Kubernetes and the KubeSail agent pre-installed. KubeSail lets you install new software on your PiBox in a single click, via software templates!

geerlingguy, 2021-04-12:
cc @PastuDan (and thanks to @rockaut for highlighting this project).

PastuDan, 2021-04-12:
Thanks @rockaut and @geerlingguy ! We're early in the process of testing some prototype boards, but I'll keep this thread up to date as we make progress!

Right now we've got some boards fully built that prove we're definitely on the right track! We're still having issues with a kernel panic at boot using the JMB582 chip, but the next revision of boards that we're ordering today will use a carrier / daughterboard design that will let us swap out SATA chipsets easily (likely will make a 2-port design around the ASM1061) as well as number of drives.

[image]
[image]

geerlingguy, 2021-04-13:
@PastuDan - Nice; and sorry I haven't gotten around to testing the 2 port chipset—it's literally sitting on top of a small pile of cards I have in the queue after I get through a couple more videos on general CM4 boards that I've been putting off :)

geerlingguy, 2021-07-02:
@PastuDan - What are the two large connector spaces on the bottom left? (To the left of the HDMI port).

closed 2021-07-02 by geerlingguy

PastuDan, 2021-07-02:
> @PastuDan - What are the two large connector spaces on the bottom left? (To the left of the HDMI port).

This is an SD slot and Ethernet port, from left to right. The eth port is significantly smaller on the latest revision.

PastuDan, 2021-07-02:
Latest photos:

[image]
[image]
[image]

geerlingguy, 2021-07-03:
@PastuDan - Ooh, nice! I like the edge connector for the drives; and I was wondering... that port did look like network, but was way deeper than a typical RJ45 jack, but shallower than SFP+ (and I was thinking it would be odd to stick SFP+ in a consumer NAS!).

PastuDan, 2021-09-22:
@miztroh pibox.io is the best place for updates. We'll send an update email soon! We've been heads down finalizing prototypes, and we're on track to launch our kickstarter next month

miztroh, 2021-09-22:
@PastuDan Ok, great.  Thanks for the quick reply.  I'll be on the lookout for the Kickstarter!

commit 9e987c10 (2021-07-02): Fixes #112: Add KubeSail PiBox 2 Mini to site.