geerlingguy/raspberry-pi-pcie-devices #113: BA7902(FRU89Y1712) FireWire 600 Card

jamesy0ung, 2021-04-14:
I have acquired one of these cards and tested it. It does not work and dmesg gives me this information:
`[    1.250396] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.252946] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.255518] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.258086] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.580797] brcm-pcie fd500000.pcie: link down`

I think it is related to the BAR space which is probably because FireWire uses direct memory access.
Honestly, I don't have much motivation to make it work because I don't have many devices that are firewire. I might do the BAR space patch and see how it goes.

jamesy0ung, 2021-04-14:
[image]
Image of the card for reference

Coreforge, 2021-04-21:
It looks like some ranges are wrong. IO BAR issues are usually "No space for bar" and something with IO (not sure about the exact message right now). 
Is it a fresh install? The device tree might be wrong.

jcdutton, 2021-05-18:
"Link down" means the card is not seated in the slot correctly. The Board cannot see the Card.

geerlingguy, 2021-07-02:
@jamesy0ung - Yeah, it seems like you're just seeing what happens when you either don't have a card plugged in, the card doesn't get any electrical signal at all (e.g. seated improperly), or the card isn't getting enough power (this probably isn't the case here, but could be on higher-power cards like GPUs and some storage controllers).

Can you try again, and make sure the power supply you're using is at least 5A @ 12V? I might even go pick up one of these cards and test it myself, since I still have a couple ancient FireWire 400 devices (and one FireWire 800 hard drive, ha!).

geerlingguy, 2021-12-12:
Card is up on the site, though waiting to hear back from @jamesy0ung if he got any more time to re-seat and reboot to see what would happen.

jamesy0ung, 2021-12-16:
I got around to testing it this morning. It appears the card requires IO BAR (unsuprising) and thus does not work. I attached some logs if anyone is interested.

'[    1.244285] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.244314] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.244383] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.244459] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.276920] brcm-pcie fd500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.277311] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.277330] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.277348] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.277439] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.277685] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.281579] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.281859] pci 0000:01:00.0: [1106:3403] type 00 class 0x0c0010
[    1.281927] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000007ff 64bit]
[    1.281958] pci 0000:01:00.0: reg 0x18: [io  0x0000-0x00ff]
[    1.282206] pci 0000:01:00.0: supports D2
[    1.282219] pci 0000:01:00.0: PME# supported from D2 D3hot D3cold
[    1.294247] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.294292] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.294313] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x6000007ff 64bit]
[    1.294354] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0100]
[    1.294367] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0100]
[    1.294382] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.294403] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]

...

[    6.125780] firewire_ohci 0000:01:00.0: enabling device (0000 -> 0002)
[    6.125923] firewire_ohci: probe of 0000:01:00.0 failed with error -12'

closed 2021-12-16 by jamesy0ung

geerlingguy, 2021-12-16:
@jamesy0ung - FYI, there are a lot of cards that try allocating IO BAR space and fail, but the card and its driver still work fine—it's usually a matter of older cards and drivers (that are X86-specific) that fail.

It looks like the main issue is:

```
[    6.125780] firewire_ohci 0000:01:00.0: enabling device (0000 -> 0002)
[    6.125923] firewire_ohci: probe of 0000:01:00.0 failed with error -12
```

And my guess is that is due to the driver trying to use IO space, but it could be something else. But I'll still mark the card as a 'no', since it does seem it's picking up the right module, just not able to load it for some reason :(