geerlingguy/raspberry-pi-pcie-devices #114: DFRobot CM4 IoT Router Carrier Board Mini

geerlingguy, 2021-04-16:
See: https://www.dfrobot.com/product-2242.html

More detail to come.

runborg, 2021-04-19:
Hi Jeff! 

I See you are trying out multiple dual-nic boards for for the cm4. There are now some ongoing work to get the free open source VyOS router distribution(vyos.io) running on the pi4, as for now it is untested on the CM4 but if you want you could try it out using this repository: https://github.com/runborg/vyos-pi-builder , as for now the kernel is not optimized and the wifi nic driver firmware files are not automatically installed, but the onboard ethernet of the pi4 works ootb..  if you are willing to test it out i'm willing to help out fixing issues :)

geerlingguy, 2021-04-19:
@runborg - Thanks for the suggestion! I was going to try out OpenWRT first, but VyOS would also be fun to test out, didn't know there was any effort behind having it run on Pi!

runborg, 2021-04-19:
The armification of VyOS have been ongoing for some time and is mostly a community effort.. as the pi is one of the most accessible platforms, we try to port it to these devices as a start... for now almost all packages and CI scripts are extended to building arm64 packages ,  but for now the kernel with dependencies and the images itself is missing..   so if you give it a go you need to compile the kernel and some kernel dependent packages for now.. And if you get into issues please open a issue on my repository or join on the VyOS slack chat :)

geerlingguy, 2021-05-18:
Testing this out this week :)

geerlingguy, 2021-05-19:
Flashing OpenWRT to the microSD is a tiny bit different than typical; you could use an imager app, but in my case I did the following:

  1. Downloaded DFRobot's version of OpenWRT [from their Wiki](https://wiki.dfrobot.com/Compute_Module_4_IoT_Router_Board_Mini_SKU_DFR0767) (under Tutorial, then System).
  2. Expanded the zip archive so I could see the `.img` file.
  3. Copy the `.img` contents over to the disk:

     ```
     diskutil list  # get disk that corresponds to your microSD or eMMC
     diskutil unmountDisk /dev/disk2  # Unmount the microSD or eMMC
     sudo dd if=~/Downloads/openwrt-bcm27xx-bcm2711-rpi-4-ext4-factory.img of=/dev/rdisk2 bs=1m
     diskutil unmountDisk /dev/disk2
     ```

geerlingguy, 2021-05-19:
I plugged ETH1 (the left-most port if looking at the Ethernet ports plug-side) into my Mac via a separate NIC, and accessed 192.168.1.1 in the browser (or could log in via `root@192.168.1.1` with SSH).

Checking over some settings:

```
root@OpenWrt:~# dmesg | grep pci
[    0.099005] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.099029] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.099066] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x0603ffffff -> 0x00f8000000
[    0.099099] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x003fffffff -> 0x0400000000
[    0.130091] brcm-pcie fd500000.pcie: link up, 2.5 GT/s x1 (SSC)
[    0.130249] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.130268] pci_bus 0000:00: root bus resource [bus 00-ff]
[    0.130286] pci_bus 0000:00: root bus resource [mem 0x600000000-0x603ffffff] (bus address [0xf8000000-0xfbffffff])
[    0.130326] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    0.130414] pci 0000:00:00.0: PME# supported from D0 D3hot
[    0.133756] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    0.133881] pci 0000:01:00.0: [10ec:8168] type 00 class 0x020000
[    0.133931] pci 0000:01:00.0: reg 0x10: [io  0x0000-0x00ff]
[    0.133969] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x00000fff 64bit]
[    0.133999] pci 0000:01:00.0: reg 0x20: [mem 0x00000000-0x00003fff 64bit]
[    0.134129] pci 0000:01:00.0: supports D1 D2
[    0.134143] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    0.137409] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    0.137446] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    0.137467] pci 0000:01:00.0: BAR 4: assigned [mem 0x600000000-0x600003fff 64bit]
[    0.137498] pci 0000:01:00.0: BAR 2: assigned [mem 0x600004000-0x600004fff 64bit]
[    0.137526] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0100]
[    0.137541] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0100]
[    0.137556] pci 0000:00:00.0: PCI bridge to [bus 01]
[    0.137572] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    0.137718] pcieport 0000:00:00.0: enabling device (0000 -> 0002)
[    0.137838] pcieport 0000:00:00.0: PME: Signaling with IRQ 38
[    0.138067] pcieport 0000:00:00.0: AER: enabled with IRQ 38
```

And here's the r8169 bringup:

```
root@OpenWrt:~# dmesg | grep r8169
[    6.700709] r8169 0000:01:00.0: enabling device (0000 -> 0002)
[    6.716237] r8169 0000:01:00.0: can't read MAC address, setting random one
[    6.727472] libphy: r8169: probed
[    6.731232] r8169 0000:01:00.0 eth1: RTL8168h/8111h, 22:f9:6b:8a:6f:ca, XID 541, IRQ 39
[    6.739271] r8169 0000:01:00.0 eth1: jumbo features [frames: 9200 bytes, tx checksumming: ko]
[   12.208327] Generic FE-GE Realtek PHY r8169-100:00: attached PHY driver [Generic FE-GE Realtek PHY] (mii_bus:phy_addr=r8169-100:00, irq=IGNORE)
[   12.329145] r8169 0000:01:00.0 eth1: Link is Down
[   15.494370] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control rx/tx
```

geerlingguy, 2021-05-19:
I enabled the USB 2.0 port on the side by running `vi /boot/config.txt` and adding the following line at the very end of the file:

```
dtoverlay=dwc2,dr_mode=host
```

(Note: `nano` is not installed in the default OpenWRT distro.)

geerlingguy, 2021-05-19:
I also wanted to see what the PCI bus saw, so I did the following:

```
# opkg update
# opkg install pcitools
# lspci -vvvv
...
01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
	Subsystem: Realtek Semiconductor Co., Ltd. RTL8111/8168 PCI Express Gigabit Ethernet controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 41
	Region 0: I/O ports at <unassigned> [disabled]
	Region 2: Memory at 600004000 (64-bit, non-prefetchable) [size=4K]
	Region 4: Memory at 600000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fffffffc  Data: 6540
	Capabilities: [70] Express (v2) Endpoint, MSI 01
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <512ns, L1 <64us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 4096 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s unlimited, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via message/WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [b0] MSI-X: Enable- Count=4 Masked-
		Vector table: BAR=4 offset=00000000
		PBA: BAR=4 offset=00000800
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [140 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [160 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [170 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [178 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=150us PortTPowerOnTime=150us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: r8169
```

geerlingguy, 2021-05-19:
Internet Speedtest.net tests give 931 Mbps down and 39 Mbps up (which is in line with my normal 2.5/10G network and my ASUS Router).

Local `iperf3` between mac on router and LAN gets 943 Mbps (which is basically as close to line speed as it gets).

Looking into what other benchmarks I could run on the router boards to test them out...

geerlingguy, 2021-05-19:
Some other benchmarking with [Flent](https://flent.org):

```
# On my Mac:
pip3 install flent
pip3 install matplotlib  # Needed for Python 3 support
pip3 install QtPy  # Needed for GUI support
flent-gui
```

geerlingguy, 2021-05-19:
I've tried bypassing OpenWRT's firewall following [these directions](https://blog.51sec.org/2021/01/configure-openwrt-allow-all-traffic.html). That still didn't seem to allow netperf to get through :(

geerlingguy, 2021-05-19:
Next test, is benchmarking packet throughput using `hping3`:

On my Mac:

  1. Install `hping3`: `brew install hping`
  2. `sudo time hping 10.0.100.10 -q -i u20 --icmp|tail -n10`

Initial result (about 10 seconds):

```
$ sudo time hping 10.0.100.10 -q -i u15 --icmp|tail -n10
^C
--- 10.0.100.10 hping statistic ---
386378 packets tramitted, 382130 packets received, 2% packet loss
round-trip min/avg/max = 0.2/0.4/10.2 ms
Command terminated abnormally.
        9.97 real         2.08 user         4.94 sys
```

`382,130` packets / `9.97` seconds ~= 38,328 pps (x2 = **76,656 pps**)

geerlingguy, 2021-05-19:
Testing with 64-byte packets (28 headers + 36 bytes data size):

```
$ sudo time hping 10.0.100.10 -q -d 36 -i u15 --icmp | tail -n10
^C
--- 10.0.100.10 hping statistic ---
177372 packets tramitted, 174405 packets received, 2% packet loss
round-trip min/avg/max = 0.2/0.3/9.2 ms
Command terminated abnormally.
        4.60 real         0.99 user         2.27 sys

# Result:
75,828 pps
```

And with 1500 bytes of data, 73,291 pps (and my mac actually started getting a little toasty at this point).

ripplecart, 2021-05-21:
Hi Jeff!

I really appreciate all the work you're doing with this router board and your YouTube channel in general!

Now I'm tinkering with this board as well but I'm having issues flashing the eMMC using rpiboot. Did you try flashing the eMMC of the CM4 using this board at all? I only get to the point where rpiboot says "Waiting for BCM2835/6/7/2711..." then nothing happens. Have tried on both Windows 10 and Ubuntu 20.04. Any ideas? I've reached out to Dfrobot support but haven't heard back yet.

Oh, and that white, plastic "pad" that is seen in the image above, did you 3D-print that yourself?

geerlingguy, 2021-05-24:
@ripplecart - The plastic 'pad' is the base of the 3D case design DFRobot has available for download on their website. It's not the best case in the world... but it's serviceable.

geerlingguy, 2021-05-24:
I also wanted to test WiFi performance as an AP; I noticed that `/boot/distroconfig.txt` contained:

```
[pi4]
dtoverlay=disable-bt
```

Which turns off WiFi as well. The reason given is to `Restore PL011 (ttyAMA0) to GPIOs 14 & 15, instead of Mini UART (ttyS0).` — If you want serial console access that way, I guess it's an option, but in my case I want to test WiFi, so I commented out that line and rebooted.

Following the [set up WiFi in OpenWRT guide](https://openwrt.org/docs/guide-quick-start/basic_wifi), it seems like the interface should be picked up automatically, but I get:

```
root@OpenWrt:/etc/config# uci show wireless
uci: Entry not found
```

So I'm wondering if the DFRobot compiled OpenWRT just doesn't have it? There's no `/etc/config/wireless` file at all (looking at the docs for [How do I enable WiFi?](https://openwrt.org/faq/how_do_i_enable_wifi).

geerlingguy, 2021-05-24:
For WiFi to work, I had to copy the Pi 4 model B config to the location the kernel expected—see [this comment](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/137#issuecomment-847276278). Now I can do WiFi too. Yay!

geerlingguy, 2021-05-26:
lol... with my earlier attempt with Flent (https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/114#issuecomment-844307125), it seems I may have hit my own barrier—I'm pretty sure I have a firewall set up on the Pi where I was trying to hit it with netperf, so the port netperf was using was of course blocked since it wasn't explicitly opened.

Oops. I'll have to try again at some point. I still have the data I need for now. But will want to get better data with Flent!

rrimc69, 2021-05-27:
Totally saved my bacon on the DFrobot DFR0767 route board !! Thanks a lot Jeff.

> For WiFi to work, I had to copy the Pi 4 model B config to the location the kernel expected—see [this comment](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/137#issuecomment-847276278). Now I can do WiFi too. Yay!

geerlingguy, 2021-05-28:
Video is up today: https://www.youtube.com/watch?v=w7teLVwi408

@rrimc69 - Glad I could help!

closed 2021-05-28 by geerlingguy

jupiler, 2021-05-31:
By any chance did you test the performance of a ipsec tunnel with this board?
Would be nice if i could use them to build a site 2 site vpn.

geerlingguy, 2021-05-31:
@jupiler - I have not, would like to at some point but right now I have the routerboard boxed up for future testing while I move onto a few other projects.

Bartvz, 2021-05-31:
@geerlingguy would it be possible to connect an LTE hat to the DFRobot CM4 Routerboard since it only has 26 GPIO pins? If so, it would make a nice LTE router. Ofcourse, you could use the USB 2.0 connector but USB 2.0 has a limited throughput.

aduewel, 2021-06-15:
I just tested this router board with Raspbian OS 64bit ... the realtek pci-e network card will not be recognized because of the missing driver. Installing the debian package `r8168-dkms` does not work, because of compile error with the newest kernel. (IIRC the same applies to Raspbian OS 32bit)

```
root@raspberrypi:~# apt install r8168-dkms
...
Setting up r8168-dkms (8.046.00-1) ...
Loading new r8168-8.046.00 DKMS files...
It is likely that 5.10.17-v7l+ belongs to a chroot's host
Building for 5.10.17+, 5.10.17-v7+, 5.10.17-v7l+ and 5.10.17-v8+
Building initial module for 5.10.17+
Error! Bad return status for module build on kernel: 5.10.17+ (armv7l)
Consult /var/lib/dkms/r8168/8.046.00/build/make.log for more information.
dpkg: error processing package r8168-dkms (--configure):
 installed r8168-dkms package post-installation script subprocess returned error exit status 10
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 r8168-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@raspberrypi:~# cat /var/lib/dkms/r8168/8.046.00/build/make.log
DKMS make.log for r8168-8.046.00 for kernel 5.10.17+ (armv7l)
Tue 15 Jun 2021 08:55:12 PM BST
make: Entering directory '/usr/src/linux-headers-5.10.17+'
  CC [M]  /var/lib/dkms/r8168/8.046.00/build/r8168_n.o
  CC [M]  /var/lib/dkms/r8168/8.046.00/build/r8168_asf.o
  CC [M]  /var/lib/dkms/r8168/8.046.00/build/rtl_eeprom.o
  CC [M]  /var/lib/dkms/r8168/8.046.00/build/rtltool.o
/var/lib/dkms/r8168/8.046.00/build/r8168_n.c:60:10: fatal error: linux/pci-aspm.h: No such file or directory
 #include <linux/pci-aspm.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/r8168/8.046.00/build/r8168_n.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1804: /var/lib/dkms/r8168/8.046.00/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.10.17+'
```

But I found this helpful post https://askubuntu.com/a/1334101/342872 which recommends to compile the kernel module from the source directly provided by Realtek. The version mentioned in the post has already been updated (to 8.049.01 as of today).

Hint: you need to extract the `/src` folder from the tar.bz2 into `/usr/src/r8168-8.049.01`

Anyhow it compiles flawlessly with dkms (to auto-recompile when newer kernels are available).

danielcbit, 2021-07-14:
Hey @geerlingguy. Have you seen this board "passing through" charging to your macbook when connected both data and power ports? I have a CM4 board with built in eMMC and when trying to flash it using this board, the command ./rpiboot just sits there doing nothing (nothing is detected) and my macbook detects as charging  when I connect the power port of the board, which is realy weird.

ripplecart, 2021-07-14:
@danielcbit : I'm also having trouble trying to flash the CM4 board with eMMC using the router board. See my comment earlier in the thread: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/114#issuecomment-846249278. 

Flashing the same CM4 board using the standard CM4 IO board is no problem. I've tried to sort it out with DFRobot support but were unsuccesfull. They are sending me a new one as RMA.

I'm starting to think that the board just doesn't support flashing a CM4 with onboard eMMC. Perhaps @geerlingguy could give it a try?

danielcbit, 2021-07-14:
Hi @ripplecart 

I was doing some testing and was able to get it working. The trick was to use a USB-A to USB-C cable instead of a USB-C to USC-C cable. When I did that, after toggling the switch for RPIBOOT to 1, the rpiboot tool was able to detect the CM4 and boot into mass-storage mode using the DFRobot Routerboard.

I used a OnePlus USB-A to USB-C cable I had laying around.

Hope that helps!

ripplecart, 2021-07-15:
Thanks for the tip @danielcbit and glad you got it working! Unfortunately I've already tried a bunch of different cables (both USB-C to USB-C and USB-C to USB-A and even an USB-C to USB-A adapter) with no success.

Would you mind detailing in what order you connect and power everything up and when exactly you flip the RPIBOOT switch and start ./rpiboot on you computer? The instructions from DFRobot differ somewhat from what I've read in other places. Thanks!

zipyinthenet, 2021-07-21:
> I just tested this router board with Raspbian OS 64bit ... the realtek pci-e network card will not be recognized because of the missing driver. Installing the debian package `r8168-dkms` does not [...]

I have the same problem when im trying to install de last raspi os lite 64bits on my dfrobot iot router + compute module , the problem was the same , but some searches in google i get that someone in this link(https://blog.zakkemble.net/rpi-compiling-a-module-for-the-64-bit-kernel/comment-page-3/) , they get to install the new kernel module and it works for me now.

I have openwrt , but i want to get a raspios system working with these two nics , because i think that openwrt its good to get a firewall rules and dhcp and no little more , but with raspios , we have the power to get the same and more with docker or any deployment , i want to put now mailtrail , and get with dfrobot module and this compute module raspi , get a firewall/router with more extras as an ids an openvpn server , its more efficient to get and install for me all in terminal than getting/installing modules on GUI with GUI as LUCI , but well.. 

here we go with raspi os ..

rrimc69, 2021-08-30:
Well it works . .  now to extend, I'd like to shoehorn either pFsense or ipFire (or suggest another Firewall with GUI that's FOSS . . ? ) on to this thing.  My first attempt with ipfire, I must have fumble fingered the power connections and fried the Onboard storage . :(    Rather bad news as replacement for my board looks to  be 6~9 months away??? geeze ..  got a 2GB Lite CM4 as a replacement today so ... yay. Can move forward.. it still took 2 months tho.

asquelt, 2021-09-06:
official 21.02.0 still doesn't have r8169 OOTB, so the same procedure applies :(

dergeberl, 2021-09-08:
> official 21.02.0 still doesn't have r8169 OOTB, so the same procedure applies :(

I had experienced the same problem today. I found a way to install the needed package without the need of a display by edit 2 files on the rootfs. Testet on sdcard only.

https://gist.github.com/dergeberl/56a4d29e62acd9aea96e220c64254317

nexxusty, 2021-09-24:
Well, I'm not doing that, so I guess I'm waiting...

jimbiggins, 2021-10-02:
I'm having an issue with the fact the RTL8111 chipset (eth1) on the DF Robot Router Board will not provide a valid MAC address.  It looks like Jeff experienced the same thing using OpenWRT.  From @geerlingguy 's early post:
> And here's the r8169 bringup:
> 
> ```
> root@OpenWrt:~# dmesg | grep r8169
> [    6.716237] r8169 0000:01:00.0: can't read MAC address, setting random one

And here is my dump using the R8168 driver directly from the Realtek site compiled as a module and loaded on Raspbian Buster w/64 bit kernel:
```
me@hostname:~ $ dmesg|grep r8168
[    5.129407] r8168: no symbol version for module_layout
[    5.129461] r8168: loading out-of-tree module taints kernel.
[    5.135927] r8168 Gigabit Ethernet driver 8.049.02-NAPI loaded
[    5.136074] r8168 0000:01:00.0: enabling device (0000 -> 0002)
[    5.166201] r8168 0000:01:00.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    5.166236] r8168 0000:01:00.0 (unnamed net_device) (uninitialized): Random ether addr 56:4b:80:d7:2a:57
```
A MAC address of all zeros is being returned which is invalid.  This causes the driver to generate a random MAC address on each boot.  The random MAC address makes it impossible to assign a connection in NetworkManager that persists after a reboot.  My issue is I assign a static IP to the connection associated with eth1, reboot, and NetworkManager has created a new connection with eth1 and defaulted back to DHCP.  The old connection configurations become stale because the MAC they are linked to will never exist again.  On every boot a new connection configuration is created which defaults to DHCP.

I need to use NetworkManager and not dhcpcd for my application.  I can temporarily change the MAC with ifconfig and that works until the next reboot.  When I try to change it permanently with ethtool, I get an error:
```
me@hostname:~ $ sudo ethtool -e eth1
Cannot get EEPROM data: Operation not supported
```
Either there is no EEPROM in this hardware design or the driver doesn't allow access to it:
```
me@hostname:~ $ sudo ethtool -i eth1
driver: r8168
version: 8.049.02-NAPI
firmware-version:
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no
```
Why is there no valid MAC being returned from the RTL8111 Ethernet Transceiver?
Any other ways to permanently change the MAC in a case where the driver is assigning a random MAC on boot up?

saket424, 2021-10-03:
did anyone succeed in using the usb port in non-gadget mode (without all that dwc2 stuff) ?

I followed https://openwrt.org/docs/guide-user/storage/usb-drives-quickstart without success

opkg update && opkg install block-mount e2fsprogs kmod-fs-ext4 kmod-usb-storage kmod-usb2 kmod-usb3

nexxusty, 2021-10-10:
Official DFRobot 21.02:

https://wiki.dfrobot.com/Compute_Module_4_IoT_Router_Board_Mini_SKU_DFR0767

Or (Direct Link): 

https://wiki.dfrobot.com/Compute_Module_4_IoT_Router_Board_Mini_SKU_DFR0767

Enjoy!

pelwell, 2021-10-12:
Oops.

nexxusty, 2021-10-12:
> Errm... [raspberrypi/linux#4625](https://github.com/raspberrypi/linux/pull/4625)

LOL, I stand corrected.

danielcbit, 2021-10-12:
> It can be added as a module. No issues in that way.
> … <#m_-6410043559419744035_>
> On Tue, Oct 12, 2021 at 1:14 PM Phil Elwell *@*.***> wrote: Errm...
> raspberrypi/linux#4625 <https://github.com/raspberrypi/linux/pull/4625> <
> raspberrypi/linux#4625 <https://github.com/raspberrypi/linux/pull/4625>>

The only issue is including this as a built in module to the kernel. That
would just increase the size of the RPI Default Kernel without any benefits.

Several other Linux Distros build kernel drivers as modules all at once so
you can keep a small kernel but still allow a device to be hotplugged and
work without having to re-compile the kernel for that.

jimbiggins, 2021-10-15:
> I'm having an issue with the fact the RTL8111 chipset (eth1) on the DF Robot Router Board will not provide a valid MAC address. 
> 
> Why is there no valid MAC being returned from the RTL8111 Ethernet Transceiver? Any other ways to permanently change the MAC in a case where the driver is assigning a random MAC on boot up?

I'll answer my own question.  I got this response from DFRobot Tech Support:
"_The chip supplier has not responded to how to manually burn for hardware Mac address._"
I assume that means they don't know how to burn a MAC into the chip because Realtek hasn't responded with instructions on how to do it.  Seems odd.

So I came up with a software workaround that fixes it by reading the "real" MAC on eth0, incrementing it by 1, and assigns it to eth1 using `macchanger`.

First install macchanger:
`sudo apt install macchanger -y`

Then create this script.  I put it in /etc.
`sudo nano /etc/changemac`
changemac script:
```
#!/bin/sh -e
# changemac

#Read the MAC of eth0, increment by 1, assign to eth1:
read realmac </sys/class/net/eth0/address
mac=$(echo $realmac | tr -d ':')
macadd=$(( 0x$mac + 1 ))
macnew=$(printf "%012x" $macadd | sed 's/../&:/g;s/:$//')
sudo macchanger --mac=$macnew eth1

exit 0
```
Make it executable:
`sudo chmod +x /etc/changemac`

Create a startup service to run it:
`sudo nano /etc/systemd/system/changemac@.service`

Paste the following:
```
[Unit]
Description=changes mac for %I
Wants=network.target
Before=network.target
BindsTo=sys-subsystem-net-devices-%i.device
After=sys-subsystem-net-devices-%i.device

[Service]
Type=oneshot
ExecStart=/etc/changemac
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

Enable the startup service:
`sudo systemctl enable changemac@eth1.service`

Reboot and use “ip a” to verify MAC of eth1 is one greater than MAC of eth0.

geerlingguy, 2021-10-29:
@asquelt - I used Fusion 360.

slmb3, 2021-11-06:
@asquelt I don't know if there's a better way or not, but in the FreeCAD Part workbench, there is an "Explode compound"  option which breaks the pieces into their components.

@geerlingguy are you able to upload the STL files that you made? I know they weren't perfect, but would still be useful. Thanks!

geerlingguy, 2021-11-06:
@slmb3 - I put the files up on Thingiverse here: https://www.thingiverse.com/thing:5100208

mzeesimba, 2021-11-10:
> Well it works . . now to extend, I'd like to shoehorn either pFsense or ipFire (or suggest another Firewall with GUI that's FOSS . . ? ) on to this thing. My first attempt with ipfire, I must have fumble fingered the power connections and fried the Onboard storage . :( Rather bad news as replacement for my board looks to be 6~9 months away??? geeze .. got a 2GB Lite CM4 as a replacement today so ... yay. Can move forward.. it still took 2 months tho.
> 
> with two full speed Eth Interfaces, I can maybe move back to a physical firewall and put that the router and the piehole on a UPS .. ;) .

Curious if you ever got pfSense to work on this board?  I'd like to switch to it based on form factor so wondering if anyone has actually proved it out on pfSense?

sdwilsh, 2021-12-13:
In the video, Jeff had mentioned VyOS testing would be happening soon (and showed [runborg/vyos-pi-builder](https://github.com/runborg/vyos-pi-builder) in the video), although also alluded to it not being a usable state yet.  It seems like [runborg/vyos-pi-builder](https://github.com/runborg/vyos-pi-builder) might be good to use now.  Has anybody tried to use VyOS with one of these yet?

moeller0, 2021-12-15:
> $ sudo time hping 10.0.100.10 -q -d 36 -i u15 --icmp | tail -n10
> ^C
> --- 10.0.100.10 hping statistic ---
> 177372 packets tramitted, 174405 packets received, 2% packet loss
> round-trip min/avg/max = [...]

Mmmh, silly question, these 64Byte frames will take up effectively 64+38 bytes (measured as byte equivalents on the wire), so `1000^3/((64+38)*8) = 1225490.196` pps is the theoretical maximum.
And under saturation each of these should take `(1 * 1000^2 micro/unity) / (1000^3 b/sec /((64B+38B)*8b/B)) = 0.816` microseconds each on the wire. Now you requested 15us inter packet gap for a theoretical maximum of `1000^2/15 = 66666.667` pps. IWell possible that I am totally wrong here, but IMHO this test really only shows that hping is not good at keeping time ;)

geerlingguy, 2022-01-04:
I just wanted to point out that @martin-niklasson posted a nice Gist with instructions for making an up-to-date OpenWRT image for this board: https://gist.github.com/martin-niklasson/6912a7e5ba49b92801d54766f1d7277a

Alternative-V, 2022-01-05:
@sdwilsh I compiled runborg/vyos-pi-builder on my M1 Macbook just to see what was in the image and it didn't look like the r8169 kernel module was included.

@mzeesimba pfSense CE is only available for AMD64 and I don't know if anyone has gotten it to work on any ARM device. Netgate has no plans on porting it to any ARM hardware other than their own either. OPNsense had a few ARM builds in the past but seems like the effort has since been abandoned. Vanilla FreeBSD should work out of the box with [RTL8111 driver](https://www.freebsd.org/cgi/man.cgi?query=if_re) included and would make a great firewall, you just gotta configure `pf.conf` yourself.

blizzardengle, 2022-01-28:
**Background**

I am thinking of using this as an OpenWRT Router + Firewall but I'm concerned about constant logging wearing out the SD card. I found that I could disable logging altogether (https://forum.openwrt.org/t/how-to-disable-all-logs-on-lede/8001) but I had two other ideas:

1. Configure logging to save on a USB device attached to the USB 2.0 port.
2. Add and boot from an NVMe hat of some kind to run OpenWRT from, this way logs are being written to the NVMe.

**Question**

Does anyone know if option 2 is possible with the DFRobot CM4 Routerboard? Do you have any other suggestions/ input?

danluckner, 2022-02-17:
Hi Guys.

I wanted to share a little bit of my findings using the DFRobot router with a CM4.

First of all: I was able to purchase a CM4102032 (Wifi, eMMC 32GB, 2GB of RAM) from seeed studio website (yes, they have it in their stock in China).

Then, I wanted to use the eMMC storage instead of a SD card. I was able to use it and change its partition size so OpenWRT use all space available.

How I achieved this was:

- Plugged CM4 in to the DFRobot router
- There's a physical switch on the DFRobot router called 'RPiBOOT'. Swithed it to 1 (default is 0)
- Plugged the USB to a Linux PC (I used Ubuntu 20.4)
- Powered the router
- Then installed the rpiboot - @geerlingguy tutorial helped a lot
- When the eMMC was recognized as a SD card, it was just a matter of using the Raspberry Pi (or Balena Etcher) to flash. I used the 21.04 DFRobot release from their Wiki - the standard OpenWRT release can also be used, but you have to do a lot of work to install the drivers. Trust me, go for the DFRobot release
- After flashing, I powered off then back on, with everything still connected to the PC
- When powered on, used rpiboot again to find the eMMC storage

Here comes the tricky part:

- Opened GParted and found (and then selected) the eMMC drive.
- Before making any modifications in GParted, you must unmount the drive. If you don't do that, GParted will get an error and then you have the reflash it to be able to do the work again
- After that, you can resize the ext4 partition where the OpenWRT was installed, without any problems.

Also, I bought DFRobot's case and using the `vcgencmd measure_temp` command, I'm seeing 54.5 °C of temperature. Not sure if that's high or low for the CM4, but I'm getting all the internet speed I have and with AdGuardHome to prevent ads on some websites.

You can also install `luci-app-statics` and `collectd` services with its variations (`collectd-mod-thermal` for instance) to monitor the temps, and etc.

I hope that helps some of the folks wondering in adventure yourself with this thing.

Best Regards to all.

blizzardengle, 2022-02-18:
> Stop being ridiculous.

A firewall router combo is a critical network appliance and I want to be proactive in reducing wear on the system. I would hate to be traveling for example and be unable to VPN into my home network because the SD card died from all the firewall logging.

>  Buy a SanDisk high endurance if this is an actual issue for you.

This is a great  idea. Looking up the datasheets for the High Endurance line shows for the 256gb @ ~$40 I could expect probably up to 1-2 years of logging before needing to replace the card to prevent burn out. I'm still concerned it would die at an inopportune time though and the write speed is capped at 40mbs. Not terrible, but it could at a bottle neck to the system/ network if I log large things, like deep packet inspection (DPI).

> SD Cards are incredibly cheap.

I don't disagree, but the overall cost of building a firewall on a DFRobot CM4 Routerboard adds up when everything is accounted for. The 256gb SD Card cost roughly the same as the board itself.

> OpenWRT does not use a lot of space

Yes, but this will be a firewall as well. I have a lot of additional packages to install and plan to use the entire file system space to accommodate this comfortably.

---

I think going the NVMe route and booting from an NVMe hat of some kind is the best solution for my use case. It's robust, I can log directly to the same drive as OpenWRT, and I can clone the NVMe if I ever wanted to quickly replace the drive with a new one after several years of service. I just don't know if the  DFRobot CM4 Routerboard supports/ plays well with any NVMe hats and would hate to invest what little money I have in trying a bunch out on my own.

Since the DFRobot CM4 Routerboard build would have cost me near ~$200 I decided to get an old HP Elite Desktop 600 or 800 G2 from Ebay for $180. Way more power than OpenWRT needs, but prefect for a firewall build.

nexxusty, 2022-02-20:
> > Stop being ridiculous.
> 
> A firewall router combo is a critical network appliance and I want to be proactive in reducing wear on the system. I would hate to be traveling for example and be unable to VPN into my home network because the SD card died from all the firewall logging.
>  DF
> > Buy a SanDisk high endurance if this is an actual issue for you.
> 
> This is a great idea. Looking up the datasheets for the High Endurance line shows for the 256gb @ ~$40 I could expect probably up to 1-2 years of logging before needing to replace the card to prevent burn out. I'm still concerned it would die at an inopportune time though and the write speed is capped at 40mbs. Not terrible, but it could at a bottle neck to the system/ network if I log large things, like deep packet inspection (DPI).
> 
> > SD Cards are incredibly cheap.
> 
> I don't disagree, but the overall cost of building a firewall on a DFRobot CM4 Routerboard adds up when everything is accounted for. The 256gb SD Card cost roughly the same as the board itself.
> 
> > OpenWRT does not use a lot of space
> 
> Yes, but this will be a firewall as well. I have a lot of additional packages to install and plan to use the entire file system space to accommodate this comfortably.
> 
> I think going the NVMe route and booting from an NVMe hat of some kind is the best solution for my use case. It's robust, I can log directly to the same drive as OpenWRT, and I can clone the NVMe if I ever wanted to quickly replace the drive with a new one after several years of service. I just don't know if the DFRobot CM4 Routerboard supports/ plays well with any NVMe hats and would hate to invest what little money I have in trying a bunch out on my own.
> 
> Since the DFRobot CM4 Routerboard build would have cost me near ~$200 ### **I decided to get an old HP Elite Desktop 600 or 800 G2 from Ebay for $180**. Way more power than OpenWRT needs, but prefect for a firewall build.

Really bad decision, IMO. Will use literally 55x more wattage to accomplish the same thing, just did the math. My DF Robot router board uses 1.5w on average. CM4 with eMMC should have been your choice. My CM4 will easily last a decade running OpenWRT if I needed it to.

paulwratt, 2022-02-21:
@nexxusty 

> Stop being rediculous

Not helpful, please dont post "opinions" like this is the future (I dont appreciate having to wade through the 250-500 word Essay replys).

@blizzardengle 

> I found that I could disable logging altogether, but I had two other ideas

Did you consider using tmpfs and/or flushing the log when it get over a certain size/age? (sorry I cant remember where I saw this - StackExchange? RPi Forum?)

blizzardengle, 2022-02-21:
@paulwratt tmpfs is actually a great idea! I had not thought of that. I was more preoccupied with all the parts and cost adding up for the firewall build. Like I said in my previous post I decided to go a different route:

- OpenWRT Router Firewall combo on an Awow AK34 for < $200 on Amazon.
- NAS System on an old HP Elite Desktop 800 G2 from Ebay for $180.
    -  Adding an NVMe to 5 sata port adapter card in the NVMe slot.

If anyone wants to build a firewall on the DFRobot CM4 Routerboard still without adding a hat for NVMe (which I still don't know if it would work), a SanDisk High Endurance using tmpfs would work (theoretically) perfectly! You may have to write your own scripts to manage tmpfs though.

paulwratt, 2022-02-21:
If they are "off the shelf" log files, then I believe there is a way to move them at rotatation time.

> theoretically

Nah, it works, really well on cheap "xrappy" sd-cards and usb thumb drives (according to some old RPi Forum I read in about 2015).

phitur, 2022-03-03:
@paulwratt 
> Did you consider using tmpfs and/or flushing the log when it get over a certain size/age?

I could be wrong, but doesn't tmpfs use ram? And if it uses ram, then wouldn't that prevent the logs from being persistent on a power loss? All the logs would go away each time the unit is rebooted. If that's ok for the use case, then it would be a great option. However, if logs are needed across power cycles, then it might not be the best approach. He could be back to writing on the card.

@blizzardengle I believe SanDisk's 'max endurance' card has the highest endurance rating if you wanted more write cycles. If you got the 256GB version, it would last a very long time if only capturing log files, depending on how verbose your logs are. It's rated for 6x more write cycles than the 'high endurance' card.

Another great option could be a combination of eMMC and an SD card. You would keep your OS memory relatively clean on the eMMC and it would last forever if you wrote all the logs and any other files outside of the system files to the SD card. Win-win!

paulwratt, 2022-03-03:
_tmpfs and/or flushing_ implied "write to permanant location". This technique has been used for 10 years by various people to extend the life of an SD-Card, so its well tested.

For example, the logs on my Alpine installation are rotated at 200K, so using `tmpfs` (even if the log is full) only uses 200K of memory.

evgeni, 2022-03-10:
Out of curiosity, did anyone measure NAT performance of this setup? I'm getting ~760MBit/s on a barebones FreeBSD 13 with iperf3. Wondered if anyone had Linux numbers for comparison ;)

punnypenguins, 2022-04-11:
Live site page for this board: https://pipci.jeffgeerling.com/boards_cm/dfrobot-cm4-iot-router.html

satmandu, 2022-04-27:
I've been using OpenWRT snapshots with this hardware for the last several months with no problems.

Set eth0 to be WAN, eth1 to be LAN, and run a script at boot to make sure that the kernel modules for the eth1 interface get loaded at first boot after snapshot install thus. I also use hurricane electric's setup for IPv6, so your setup might differ.

```
#!/bin/sh                                                                                                                                                                              
# Put your custom commands here that should be executed once                                                                                                                           
# the system init finished. By default this file does nothing.                                                                                                                                                                                                                             
opkg update || (sleep 30 ; opkg update )                                                                                                                                               
opkg install curl libcurl luci-app-acme acme luci-theme-openwrt-2020 6in4 jq                                                                                                           
opkg install kmod-r8169 && opkg install kmod-usb-dwc2 kmod-usb-net-rndis usb-modeswitch kmod-nls-base kmod-usb-core kmod-usb-net kmod-usb-net-cdc-ether kmod-usb2

uci set network.wan6=interface                                                                                
uci set network.wan6.proto=6in4        
# Tunnelbroker info is used for this.                                                                                                                                             
# uci set network.wan6.peeraddr=                                                                                                                                      
# uci set network.wan6.ip6addr=                                                                                                                               
# uci set network.wan6.ip6prefix=                                                                                                                                
# uci set network.wan6.tunnelid=                                                                                                                                               
uci commit network                                                                                                                                                                    
                                                                                                                                                                                      
uci set firewall.@zone[1].network='wan wan6'                                                                                                                                          
uci commit firewall   

if [ -e /etc/backup/installed_packages.txt ]; then                                                                                                                                    
  echo $opkginstalled | cut -f 1 -d ' ' | sort -u > /tmp/currentpkg && \                                                                                                              
    cat /etc/backup/installed_packages.txt | cut -f 1 | \                                                                                                                             
    sort -u > /tmp/oldpkg && grep -v -F -x -f /tmp/currentpkg /tmp/oldpkg > /tmp/inst                                                                                                 
   if [ -s "/tmp/inst" ]; then                                                                                                                                                        
    opkg install $(cat /tmp/inst | sort -u) && \                                                                                                                                      
      rm /tmp/currentpkg /tmp/oldpkg /tmp/inst /etc/backup/installed_packages.txt                                                                                                     
    echo "installs done"                                                                                                                                                              
   else                                                                                                                                                                               
    echo "no installs needed" && \                                                                                                                                                    
      rm /tmp/currentpkg /tmp/oldpkg /tmp/inst /etc/backup/installed_packages.txt                                                                                                     
   fi                                                                                                                                                                                 
fi                                                                                                                                                                                    
[... cut ...]
  luci-proto-wireguard rpcd-mod-luci luci-app-firewall luci-app-opkg luci-compat \
  libiwinfo libiwinfo-lua irqbalance htop luci-theme-bootstrap diffutils bind-dig 6in4 haproxy luci-app-ttyd \
  nmap bash kmod-usb-net-rndis usb-modeswitch luci-app-mwan3 tmux collectd collectd-mod-thermal \
  luci-app-statistics                                                                                                                   
ip link set 6in4-wan6 type sit local $(ip -o -4  -j address show dev eth0 | jq -r '.[].addr_info[].local') 
```

[31 more replies]