geerlingguy/raspberry-pi-pcie-devices #116: Test SATA card (IOCrest, 4-ports, ASM1064 chipset)

ici0255, 2021-04-19:
Yesterday I did receive this card:
https://www.sybausa.com/index.php?route=product/product&manufacturer_id=13&product_id=1062
Did stick it into my CM4 I/O board, and it is working fine with AHCI drivers (I have compiled custom kernel with SATA support, as suggested by Jeff).
--------------------------
pi@rasp-4b-cm:~ $ lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02)
pi@rasp-4b-cm:~ $ lspci -k
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02)
Subsystem: ZyDAS Technology Corp. Device 2116
Kernel driver in use: ahci
Kernel modules: ahci
--------------------------
Seems that system can handle it pretty well:
--------------------------
pi@rasp-4b-cm:~ $ dmesg | grep pci
[    1.261065] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.261169] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.261331] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.261493] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x007fffffff -> 0x0400000000
[    1.343978] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.344447] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.344536] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.344614] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.344808] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.345119] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.348974] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.352743] pci 0000:01:00.0: [1b21:1064] type 00 class 0x010601
[    1.356214] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00001fff]
[    1.359691] pci 0000:01:00.0: reg 0x24: [mem 0x00000000-0x00001fff]
[    1.363058] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0007ffff pref]
[    1.366390] pci 0000:01:00.0: enabling Extended Tags
[    1.369814] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    1.373101] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 7.876 Gb/s with 8.0 GT/s PCIe x1 link)
[    1.389195] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.392573] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.395949] pci 0000:01:00.0: BAR 6: assigned [mem 0x600000000-0x60007ffff pref]
[    1.399310] pci 0000:01:00.0: BAR 0: assigned [mem 0x600080000-0x600081fff]
[    1.402652] pci 0000:01:00.0: BAR 5: assigned [mem 0x600082000-0x600083fff]
[    1.406001] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.409312] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    4.268176] pci 0000:00:00.0: enabling device (0000 -> 0002)
--------------------------
I have only HDD-s available (no SSD at hand), so the HDPARM in test with TOSHIBA 4TB drive got this:
--------------------------
hdparm -Ttv /dev/sda

/dev/sda:
 multcount     = 16 (on)
 IO_support    =  1 (32-bit)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 486401/255/63, sectors = 7814037168, start = 0
 Timing cached reads:   2078 MB in  2.00 seconds = 1040.90 MB/sec
 Timing buffered disk reads: 590 MB in  3.00 seconds = 196.61 MB/sec
--------------------------

geerlingguy, 2021-04-19:
Just adding a few more details on the card here for easier searching:

```
4 Port SATA III PCI-e 3.0 x1 Card Non-Raid with Low Profile Bracket - SI-PEX40156
ASMedia 1064
```

Looks like a nice little card for 4-drive solutions.

geerlingguy, 2021-06-23:
Going to leave this open until I get time to add to the site ;)

closed 2021-07-02 by geerlingguy

asperkins, 2021-08-01:
Should this card be recognized by the latest version of raspbery pi os without need for a recompile? It doesn't appear to be recognized when looking at dmesg and lspci. lspci shows no output and dmesg just notes the identification of PCI, but ends by saying link is down.

geerlingguy, 2021-08-02:
@asperkins - If it says `link is down` then the hardware is not being recognized at all. It could be a faulty card, a power issue, or something else. How are you powering the board and the card?

asperkins, 2021-08-02:
I am just plugging the card directly into the cm4 i/o board. I believe that is how you were doing it with the other 4 port card. Maybe this one has higher power requirements? I'm thinking of ordering the board you used in your video and trying that one. I chose this particular one because of it's small size.

geerlingguy, 2021-08-02:
@asperkins - What kind of power supply do you have? Sometimes if you use a 2A 12V supply, I've noticed, cards don't initialize correctly. I'd make sure it's at least 5A @ 12V.

asperkins, 2021-08-02:
Darn, I got all excited when I realized it IS a 2A wall wart. But, I grabbed a 10A unit and it does the same thing. I did order the other card you used in the video, so will see if I just have a bad card. I appreciate you help.

asperkins, 2021-08-02:
Ok, so now I'm just embarrassed. I decided to go back to square one. I decided to take everything apart, checked for anything out of the ordinary. When I unplugged power, pci card and went to check the cm4 module.... You probably see where this is going. It was seated well enough for the PI to work fine, but not seated far enough for the PCI to make connections. Thanks again for the pointers, hopefully I'll have my NAS working soon.

geerlingguy, 2021-08-04:
@asperkins heh, the ol' "Did you try plugging it in again" problem strikes again!

Srverde90, 2022-01-06:
This card use the same chip, what do you think, it's working with the raspberry?

https://es.aliexpress.com/item/1005003164217007.html

itiberiu, 2023-02-19:
i have the same card on a proxmox server (debian) and i freeze my sistem on hi trafic data. Any help with driver update ar firmware? ty

ici0255, 2023-02-22:
The card with 12 SAT ports is using some I/O multiplexer, as it is discussed here:
https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/116
AFAIR, ASM1064 chipset itself cannot handle so many SATA ports.
So, most likely the issue is the HW on-top of the ASM chipset.
P.S ASM1064 PCI SATA card, which I was testing and reported in this thread above, currently is not used. I have no tested it with the latest Raspberry OS versions, so I can not report any performance tests with it.

ezaul, 2024-02-04:
Good afternoon!

I have Rasp 5 8gb + HatDrive! Bottom (NVMe 2230, 2242, 2280 GEN 3) for Raspberry Pi 5 + Expansion Board SFF-8087 M.2 NVME to Mini SAS Support 4 Ports SATA3.0 6Gbps HDD SSD SATA Controller SFF8087 to M2 NVME Adapter

https://pineberrypi.com/products/hatdrive-bottom-2230-2242-2280-for-rpi5?pr_prod_strat=jac&pr_rec_id=aa41a0aa3&pr_rec_pid=8729962250573&pr_ref_pid=8729958023501&pr_seq=uniform

https://es.aliexpress.com/item/1005006050617325.html?spm=a2g0o.order_list.order_list_main.5.4f40194dhsM2q1&gatewayAdapt=glo2esp

lspci -vvvk -s 0000:01:00.0

0000:01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02) (prog-if 01 [AHCI 1.0])
        Subsystem: ASMedia Technology Inc. Device 2116
        !!! Unknown header type 7f
        Interrupt: pin ? routed to IRQ 169
        Region 0: Memory at 1b00080000 (32-bit, non-prefetchable) [size=8K]
        Region 5: Memory at 1b00082000 (32-bit, non-prefetchable) [size=8K]
        Expansion ROM at 1b00000000 [disabled] [size=512K]
        Kernel driver in use: ahci
        Kernel modules: ahci
        SATA controller: ASMedia Technology Inc. Device 1064 (rev 02) (prog-if 01 [AHCI 1.0])

0000:01:00.0 SATA controller [0106]: ASMedia Technology Inc. Device [1b21:1064] (rev 02)

0000:01:00.0 SATA controller: ASMedia Technology Inc. Device 1064 (rev 02) (prog-if 01 [AHCI 1.0])
        Subsystem: ASMedia Technology Inc. Device 2116
        Flags: bus master, fast devsel, latency 0, IRQ 166
        Memory at 1b00080000 (32-bit, non-prefetchable) [size=8K]
        Memory at 1b00082000 (32-bit, non-prefetchable) [size=8K]
        Expansion ROM at 1b00000000 [virtual] [disabled] [size=512K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [80] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [130] Secondary PCI Express
        Capabilities: [170] Latency Tolerance Reporting
        Capabilities: [200] L1 PM Substates
        Kernel driver in use: ahci
        Kernel modules: ahci

hwinfo --pci

02: PCI 100.0: 0106 SATA controller (AHCI 1.0)
  [Created at pci.386]
  Unique ID: VCu0.cl04_Rcc_0E
  Parent ID: qLht.sFJvPhlHmw3
  SysFS ID: /devices/platform/axi/1000110000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0
  SysFS BusID: 0000:01:00.0
  Hardware Class: storage
  Model: "ASMedia SATA controller"
  Vendor: pci 0x1b21 "ASMedia Technology Inc."
  Device: pci 0x1064
  SubVendor: pci 0x1b21 "ASMedia Technology Inc."
  SubDevice: pci 0x2116
  Revision: 0xff
  Driver: "ahci"
  Driver Modules: "ahci"
  Memory Range: 0x1b00080000-0x1b00081fff (rw,non-prefetchable)
  Memory Range: 0x1b00082000-0x1b00083fff (rw,non-prefetchable)
  Memory Range: 0x1b00000000-0x1b0007ffff (ro,non-prefetchable)
  IRQ: 169 (no events)
  Module Alias: "pci:v00001B21d00001064sv00001B21sd00002116bc01sc06i01"
  Driver Info #0:
    Driver Status: ahci is active
    Driver Activation Cmd: "modprobe ahci"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #3 (PCI bridge)

I'm trying to run it on Rasp 5 without HD, it returns this to me in dmesg

[    3.142703] ahci 0000:01:00.0: version 3.0
[... 261 lines cut ...]
[   59.836823] ahci 0000:01:00.0: AER: can't recover (no error_detected callback)
[   59.849775] pcieport 0000:00:00.0: AER: device recovery failed
[   59.855676] pcieport 0000:00:00.0: AER: Multiple Uncorrected (Non-Fatal) error received: 0000:00:00.0
[   59.864986] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[   59.876192] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00044000/00400000
[   59.930810] pcieport 0000:00:00.0:    [14] CmpltTO
[   59.936942] pcieport 0000:00:00.0:    [18] MalfTLP                (First)
[   59.943758] pcieport 0000:00:00.0: AER:   TLP Header: 60000001 0100000f 000000ff ffffe000
[   59.951975] ahci 0000:01:00.0: AER: can't recover (no error_detected callback)
[   59.959240] pcieport 0000:00:00.0: AER: device recovery failed
[   59.965096] pcieport 0000:00:00.0: AER: Uncorrected (Non-Fatal) error received: 0000:00:00.0
[   59.973573] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[   59.984753] pcieport 0000:00:00.0:   device [14e4:2712] error status/mask=00044000/00400000
[   59.993139] pcieport 0000:00:00.0:    [14] CmpltTO
[   59.999258] pcieport 0000:00:00.0:    [18] MalfTLP                (First)
[   60.006072] pcieport 0000:00:00.0: AER:   TLP Header: 60000001 0100000f 000000ff ffffe000
[   60.014287] ahci 0000:01:00.0: AER: can't recover (no error_detected callback)

I'm using
Linux pi-nas 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

Can anyone tell me what this could be? Do you have any secrets?

Thank you very much in advance
Thanks

commit 6bcfb482 (2021-07-02): Fixes #116: Add ASM1064 Syba or IO Crest 4-port SATA card.