geerlingguy/raspberry-pi-pcie-devices #117: Test SATA card (no-name, 2-port, Marvell 9125)

ici0255, 2021-04-19:
I have tested also this no-name card:
https://www.amazon.com/PCE2SAT-M01-PCI-2-Port-Expansion-Adapter/dp/B085LXSXDS

Card is accepted by custom kernel with SATA and Marvell support:
-----------------------------------
pi@raspberrypi:~ $ lspci
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
01:00.0 SATA controller: Marvell Technology Group Ltd. 88SE9125 PCIe SATA 6.0 Gb/s controller (rev 11)
-----------------------------------
Although the details does not look so nice:
-----------------------------------
pi@raspberrypi:~ $ dmesg | grep pci
[    1.265029] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.267781] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.270567] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.273253] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x007fffffff -> 0x0400000000
[    1.292362] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.295389] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.297991] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.300652] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.303404] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.306226] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.312668] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.315637] pci 0000:01:00.0: [1b4b:9125] type 00 class 0x010601
[    1.318393] pci 0000:01:00.0: reg 0x10: [io  0x8000-0x8007]
[    1.321080] pci 0000:01:00.0: reg 0x14: [io  0x8040-0x8043]
[    1.323657] pci 0000:01:00.0: reg 0x18: [io  0x8100-0x8107]
[    1.326268] pci 0000:01:00.0: reg 0x1c: [io  0x8140-0x8143]
[    1.328840] pci 0000:01:00.0: reg 0x20: [io  0x800000-0x80000f]
[    1.331440] pci 0000:01:00.0: reg 0x24: [mem 0x00900000-0x009007ff]
[    1.333995] pci 0000:01:00.0: reg 0x30: [mem 0x610000000-0x61000ffff pref]
[    1.336658] pci 0000:01:00.0: PME# supported from D3hot
[    1.342803] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.345318] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.347808] pci 0000:01:00.0: BAR 6: assigned [mem 0x600000000-0x60000ffff pref]
[    1.350262] pci 0000:01:00.0: BAR 5: assigned [mem 0x600010000-0x6000107ff]
[    1.352655] pci 0000:01:00.0: BAR 4: no space for [io  size 0x0010]
[    1.355027] pci 0000:01:00.0: BAR 4: failed to assign [io  size 0x0010]
[    1.357250] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0008]
[    1.359552] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0008]
[    1.361833] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0008]
[    1.364109] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0008]
[    1.366375] pci 0000:01:00.0: BAR 1: no space for [io  size 0x0004]
[    1.368598] pci 0000:01:00.0: BAR 1: failed to assign [io  size 0x0004]
[    1.370835] pci 0000:01:00.0: BAR 3: no space for [io  size 0x0004]
[    1.372998] pci 0000:01:00.0: BAR 3: failed to assign [io  size 0x0004]
[    1.375205] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.377450] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    3.980710] pci 0000:00:00.0: enabling device (0000 -> 0002)
-----------------------------------
The output is quite similar to Jeff's report about IOCrest 4-port Marvell 9215 card (https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/1).
Card is working, performance results are fine (TOSHIBA 4TB HDD):
-----------------------------------
pi@rasp-4b-cm:~ $ sudo hdparm -Ttv /dev/sda
/dev/sda:
 multcount     = 16 (on)
 IO_support    =  1 (32-bit)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 486401/255/63, sectors = 7814037168, start = 0
 Timing cached reads:   2140 MB in  2.00 seconds = 1071.32 MB/sec
 Timing buffered disk reads: 580 MB in  3.00 seconds = 193.31 MB/sec
-----------------------------------
Downside with this card is Kernel panic randomly happening upon system restart with this card inserted. It does not happen often (1case in 5 attempts), but is is not nice behavior. Error is reported (if I do correctly understand system output) at module "pci_generic_config_read+0x44/0xe8", "pci_generic_config_read+0x2c/0xe8".
Because of this instability I did order another card, which is ASM1064 chipset based. Test report of this card testing is also here on this GitHub thread.

geerlingguy, 2021-04-19:
> Downside with this card is Kernel panic randomly happening upon system restart with this card inserted.

Strange, I wonder if it could be in any way related to power input? What kind of power supply are you using with the IO Board?

ici0255, 2021-04-19:
The CM4 on I/O board currently are powered by PC power supply (+12 V output is used). But I have tested it also with another 12 V supply - the same results there.
So far no such issues with another SATA card (ASM1064). Let us see, how it will show itself on longer run.

Le-Div, 2021-04-26:
I have encountered the same kernel panic during boot with the cards I am currently testing. The cards are the Qualcom Atheros AR9462 / AR5B22 and Intel 5300 NIC. Both are Wifi-Network cards . For the first one I also created an issue on this GitHub thread. Do you have any Ideas on how to get more debug information and do you investigate the problem further ?

ici0255, 2021-04-27:
@Le-Div
Unfortunately I am total new-be in Raspberry Pi programing and debugging. I did try to find some info about SError in Google, and some discussions came up, but no much info about the investigation of such cases. One thing was - add UART console to GPIO pins and log all the messages related to SError and debugging, but currently I have no equipment to do this.
There was discussion about similar SError and debugging in this thread: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/26, but it was caused by another module while testing video card. May be Jeff or someone from Raspberry team can enlighten us what to do and where to start?

Le-Div, 2021-05-01:
@ici0255 
Thanks for all the information. I was able to get boot logs via UART. Unfortunately I only obtained them for a normal boot without any PCIe card attached. After inserting a card the UART terminal remained empty. I assume that the kernel panic happens so early that the UART hasn't even started. So I couldn't get any further information from that.

ici0255, 2021-05-06:
Look this article:
https://kiljan.org/2021/02/20/64-bit-arch-linux-arm-on-a-raspberry-pi-4-model-b-with-8-gb-of-ram/
According to this, additional configuration steps must be applied to get boot messages to UART.
Also Raspberry Pi documents do speak about "earlycon" console support:
https://www.raspberrypi.org/documentation/configuration/uart.md
which must be enabled in "cmdline.txt".
But in general, also this info we can see on our screenshots, give us following picture:

SError do occur upon executing the module "pci_generic_config_read", which is called by following modules - "pci_bus_generic_read_dev_vendor_id" ->"pci_bus_read_config_dword" -> "pci_generic_config_read".
So, kernel is attempting to read the inserted card Vendor ID, and the result is this SError.
The big question is - why this is happening? What illegal action do happen during this read operation?
In my test this happen just randomly - in majority of restarts, the operation is successful (DMESG do report inserted PCI card ID). In your case all the attempts do fail.

So, there is something, why this reading is causing kernel panic, but to dig deeper, special kernel version with debug support have to be compiled.
I am not so skilled enough to do this myself, so, maybe we should address it to Raspberry Pi forum, where developers can help?

ici0255, 2021-05-08:
@Le-Div
I did receive UART module yesterday and was able to get standard kernel logs via UART for both successful and failing CM4 boot.
As I did write in my previous note, there is no anything new beside the obvious - system attempt to read PCI Vendor ID is causing the Serror.
Searching over the Inet about ARM64 and Serror I did find one really deep (to my mind) discussion about PCI and Serror, but this is about another ARM64 chip:
https://lore.kernel.org/linux-pci/CAMdYzYoTwjKz4EN8PtD5pZfu3+SX+68JL+dfvmCrSnLL=K6Few@mail.gmail.com/
According to this thread, it seems that ARM64 PCI drivers are not fully developed to handle all the responses, that can be received from inserted PCI card. There is also some patch offered to bypass the Serror for debugging purposes, but this is too complicated to me.
I you feel experienced enough, you can try to build similar environment for Raspberry CM4 (kernel with debug patches allowed + similar patch for SError bypass as described in that discussion) to let your CM4 to register PCI devices you are testing.
For my case I am not sure, what direction to seek, because in some cases it is working just fine, but sometimes PCI card initialization just fails. This might be some timing issue on the I/O module bus, and probably need some careful debugging, but I am not quite ready yet (just learning).

paulwratt, 2021-05-08:
> I have tested also this no-name card:
> https://www.amazon.com/PCE2SAT-M01-PCI-2-Port-Expansion-Adapter/dp/B085LXSXDS

That look very similar to the card offered for the RockPro64.

RE kernel build:
It is not as big a task as you may think, there are tutorials on RPI.org forum, and there should instuctions on the landing page of the RPi-linux-kernal page (the README.md).

there are no extra libraries for the kernel, you pass the ".rpi4-cm.config" (<= not the real file name) file to the menu configuration (the same one RPi Foundation use to build `kernelv8.img` - it'll say `armv8` or `aarch64` in the name) with your first `make` command, and tick "DEBUG", then exit which will resave the .config file with the DEBUG defines enabled (the .config is _hidden_ by default, use `ls -a`, but is visible on the GitHub repo page), then you run the `make` command again

That is over simplifying, but it is that easy. The _big_ things to note about kernel building are filespace for the repo, and compile times (2-4 hours) - an 8G RPi4 (or CM4) should get through it fairly quickly, a 1Gb RPi2B+ will take about 12 hours without zram enabled - cross-compiling means installing the compiler tool chain, which can sometimes be painful, but many documentations (and tutorials) are available

It might be time to ask @geerlingguy for a video on how to compile debug kernel for RPi CM4 when you have to enable extra things (like is needed for `io bar` for graphics cards)

ici0255, 2021-05-08:
@paulwratt
Thank you for suggestions. I shall try to build kernel with DEBUG enabled.
FYI, I have done this testing with custom kernel, cross-compiled on my Kubuntu PC, so I do know how to do this and compilation time would not be the problem.
The main issue for me would be the understanding of all the DEBUG info I shall get and what else can be done, if the debug info do not show enough details about the failure.

geerlingguy, 2021-05-10:
@paulwratt - I will be doing a livestream this Wednesday trying a GeForce GTX 1080 graphics card, and in part of it, I plan on recompiling (cross-compiling) the kernel live, so hopefully that can be used as a good reference point ;)

But I agree with @ici0255 - debug can be a little... weird sometimes. Especially if you also try debugging through a serial port.

ici0255, 2021-05-10:
@geerlingguy
Totally agree about weirdness of debugging. Today I did cross compile the 5.10.35-v8+ with a lot of debugging options set, but the result was non-booting system.
On my current custom-built kernel (5.10.31-v8+) I am able to get all the messages via UART to my PC, even the bootlog, but with this debug kernel I got only bootlog via UART, afterwards the output was switched to HDMI display (tty1 I suppose) and system did hung waiting for eMMC second partition (root) to become available (?).
So, with USBBOOT I was able to set my current 5.10.31 kernel as active and get it working again, but I am not sure if I shall continue to walk this weird debugging path :-(.

ici0255, 2023-02-22:
Some late info update:
I am using this card in my SAMBA server for last two years, and with the latest Raspberry OS it seems quite stable (no kernel panic occured during last year reboots). Here are some recent details:

`Linux rasp-4b-cm 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64

pi@rasp-4b-cm:~ $ lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 SATA controller: Marvell Technology Group Ltd. 88SE9125 PCIe SATA 6.0 Gb/s controller (rev 11)
pi@rasp-4b-cm:~ $ lspci -k
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 SATA controller: Marvell Technology Group Ltd. 88SE9125 PCIe SATA 6.0 Gb/s controller (rev 11)
        Subsystem: Marvell Technology Group Ltd. 88SE9125 PCIe SATA 6.0 Gb/s controller
        Kernel driver in use: ahci
        Kernel modules: ahci
pi@rasp-4b-cm:~ $ sudo hdparm -Ttv /dev/sda
/dev/sda:
 multcount     =  0 (off)
 IO_support    =  1 (32-bit)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 972801/255/63, sectors = 15628053168, start = 0
 Timing cached reads:   2180 MB in  2.00 seconds = 1091.60 MB/sec
 Timing buffered disk reads: 752 MB in  3.01 seconds = 250.03 MB/sec
`
To my mind this issue can be closed.