geerlingguy/raspberry-pi-pcie-devices #118: First (AFAIK) relatively successful PCI Express sound card test  : Creative SB1040

Darry77, 2021-04-23:
I managed to get a Creative SB1040 (CA0110 based) sound card to work when connected directly to a Pi 4 CMIO PCI Express slot . It does not work when connected through an ASM1184E PCI Express switch (same sound card and PCIE switch combo work fine on an x86-64 setup). "dtoverlay=pcie-32bit-dma" is required .

I only tested recording, which works fine at least at 44.1KHz 16-bit over S/PDIF (I tested recording quality using RightMark Audio Analyzer). >16-bit recording displays weird distortion, which is not present on an x86-64 platform . 

I am waiting on a Pericom-based PCIE switch to test whether that changes anything . 

See some details starting at this post : 
https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=306055&p=1855518#p1854603

geerlingguy, 2021-04-23:
@Darry77 - Nice work! Can you take a picture of the card (preferably in the style of other cards in the database) and upload it here (and release any rights so I can put it in the OSS repo)? That way I can add it to the database and people will more easily find it. I may add a new category like 'sound and AV cards' or something like that (including TV tuners and capture cards as well).

Darry77, 2021-04-24:
@geerlingguy I hope the picture is good enough (quick cell phone shot in harsh lighting) . I hereby relinquish all rights to the attach picture and release to the public domain . 

On a different note, my issues with the sound card when using a PCI Express switch may be related to my use of an RT patched kernel . When testing on an x86-64 platform, I get the same errors as on a Pi 4, but only when using an RT patched kernel and a PCI Express switch. I have never tested with a stock Raspbian kernel on the Pi 4 (I have using an RT one practically since the beginning of my project), but I intend to test this in a short while . 

[image]

geerlingguy, 2021-04-30:
Thanks so much for sending in this data; I added the card to the site, and put links back to this issue and your Pi Forum post, so hopefully others looking to do sound on the Pi will have some good references! Please confirm that all you needed to do (besides make sure the card was plugged into the Pi directly, at least for now) was to compile a custom kernel with `CONFIG_SND_HDA_INTEL` and add `dtoverlay=pcie-32bit-dma` to your boot config?

commit e8e2033e (2021-04-30): Fixes #118: Add Creative SB1040 card to site.
closed 2021-04-30 by geerlingguy