geerlingguy/raspberry-pi-pcie-devices #119: Scalenode - Baseboard for CM4 optimized for clustering

geerlingguy, 2021-04-23:
Similar to uptime.lab's CM4 blade concept, it looks like antmicro's [Scalenode](https://github.com/antmicro/scalenode) is made for high-density Pi rackmount 'blade' setups:

[image]

  - Raspberry Pi CM4
  - Gigabit Ethernet with integrated PoE circuitry
  - On-board M.2 (key-M) slot supporting NVMe SSDs
  - Slim PCB outline for 1U chassis
  - Expansion connector for peripherals connected over USB
  - Expansion connector for HDMI adapters

geerlingguy, 2021-04-30:
Added to the site; if they reach out and want to send one to me for testing, I'd be glad, could also see how it compares to the CM4 Blade being built by Uptime.Lab.

warmwaffles, 2021-05-02:
I'll open a PR soon to add some links and information that I've been looking at.

The board is opensource and can be viewed here https://github.com/antmicro/scalenode note it is still "in progress" but at the moment it looks like they are doing some more updates as I pointed out in this issue [with unplaced components](https://github.com/antmicro/scalenode/issues/1).