geerlingguy/raspberry-pi-pcie-devices #120: Test Intel AX210NGW WiFi 6E M.2 card

geerlingguy, 2021-04-26:
I just ordered a [WISE TIGER AX210NGW WiFi Card, Wi-Fi 6E 11AX](https://amzn.to/3aIBJK9) card, and I'm going to see if I can get it working using a similar layout as what I used for the AX200.

TODO: Picture here.

The other thing I'd like to see is if I can set it up as an access point and how well it handles that. Apparently the AX200 didn't handle AP mode that well (though I didn't test it at all), and this chipset should handle it better... according to random commenters :)

paulwratt, 2021-05-17:
make sure you get the latest kernel fixes for (_all_) wifi 6 devices

bcliu430, 2021-08-05:
I have tried the raspberry pi 5.10.y kernel. I can get it working with the commit (6495fa3) I tried on 7/28. I just enabled iwlwifi and copied the latest ucode to /lib/firmware.

geerlingguy, 2021-11-18:
It looks like the AX210 firmware requires kernel 5.10+ (which we have), and is available for download here: https://www.intel.com/content/www/us/en/support/articles/000005511/wireless.html

I downloaded `iwlwifi-ty-59.601f3a66.0.tgz` and ran:

```
$ tar -xvzf iwlwifi-ty-59.601f3a66.0.tgz
$ cd iwlwifi-ty-59.601f3a66.0/
$ sudo cp iwlwifi-*.ucode /lib/firmware
$ sudo reboot
```

For the driver, it looks like it'll need a recompile.

bcliu430, 2021-11-18:
> It looks like the AX210 firmware requires kernel 5.10+ (which we have), and is available for download here: https://www.intel.com/content/www/us/en/support/articles/000005511/wireless.html
>
> I [...]

Check and see if you have **iwlwifi** and **mvm** checked in kernel config. When I did some testing before, not all functionalities are fully implemented, i.e. wifi scanning is working but it does not always scan 6GHz channels. It is not fully supported until kernel 5.13.

geerlingguy, 2021-11-18:
Using my [cross-compile environment](https://github.com/geerlingguy/raspberry-pi-pcie-devices/tree/master/extras/cross-compile), I recompiled enabling iwlwifi in menuconfig:

```
Device Drivers
  > Network device support
    > Wireless LAN
      > Intel Wireless WiFi Next Gen AGN - Wireless-N/Advanced-N/Ultimate-N (iwlwifi)
        > Intel Wireless WiFi MVM Firmware support
```

(I had to check the iwlwifi support to reveal the MVM option).

Also, regarding AP mode for 5/6 Ghz: https://bugzilla.kernel.org/show_bug.cgi?id=206469#c2

geerlingguy, 2021-11-18:
Before custom kernel:

```
pi@seaberry:~ $ uname -a
Linux seaberry 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
```

After custom kernel:

```
pi@seaberry:~ $ uname -a
Linux seaberry 5.10.79-v8+ #1 SMP PREEMPT Thu Nov 18 21:53:24 UTC 2021 aarch64 GNU/Linux
```

And driver loads:

```
[   10.878198] iwlwifi 0000:12:00.0: enabling device (0000 -> 0002)
[   10.909737] iwlwifi 0000:10:00.0: api flags index 2 larger than supported by driver
[   10.909878] iwlwifi 0000:10:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 93.8.63.28
[   10.913151] iwlwifi 0000:10:00.0: loaded firmware version 59.601f3a66.0 ty-a0-gf-a0-59.ucode op_mode iwlmvm
[   11.176677] iwlwifi 0000:10:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[   11.494958] iwlwifi 0000:10:00.0: base HW address: 84:5c:f3:f6:e9:29
```

geerlingguy, 2021-11-18:
Wrote up a guide since I do this a lot and confuse myself often: [Working with multiple WiFi interfaces on a Raspberry Pi](https://www.jeffgeerling.com/blog/2021/working-multiple-wifi-interfaces-on-raspberry-pi).

And dang... I just realized I don't have any external antennae with the plug size for the AX210. I only have the very-slightly-larger size suitable for things like the U.FL antenna connector on the CM4 itself, or larger WiFi cards.

geerlingguy, 2021-11-18:
Just ordered a [pair of NGFF / IPEX4 / MHF4 antennae](https://amzn.to/30FmmzJ) for next-day shipping. Hope they come quickly!

angelia-gt, 2021-11-19:
Hi friend
Can you tell us which order number?thank you

龚婷

Fenvi Technology Co., Ltd/国际部

15820354009

geerlingguy, 2021-11-19:
@angelia-gt - See the link — that's the exact part I ordered from Amazon (mostly because it should ship next day :).

angelia-gt, 2021-11-19:
龚婷

Fenvi Technology Co., Ltd/国际部

15820354009

geerlingguy, 2021-11-22:
```
$ iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          
wlan1     IEEE 802.11  ESSID:"geerling-acn"  
          Mode:Managed  Frequency:5.18 GHz  Access Point: 3C:7C:3F:6A:FA:C4   
          Bit Rate=1.7296 Gb/s   Tx-Power=22 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=67/70  Signal level=-43 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:77   Missed beacon:0
```

And iperf in both directions—Mac (10G wired) to seaberry (AX210):

```
$ iperf3 -c 10.0.100.27
Connecting to host 10.0.100.27, port 5201
[  5] local 10.0.100.100 port 64950 connected to 10.0.100.27 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  29.3 MBytes   246 Mbits/sec                  
[  5]   1.00-2.00   sec  30.0 MBytes   251 Mbits/sec                  
[  5]   2.00-3.00   sec  18.3 MBytes   153 Mbits/sec                  
[  5]   3.00-4.00   sec  29.4 MBytes   247 Mbits/sec                  
[  5]   4.00-5.00   sec  30.1 MBytes   252 Mbits/sec                  
[  5]   5.00-6.00   sec  31.7 MBytes   266 Mbits/sec                  
[  5]   6.00-7.00   sec  31.5 MBytes   264 Mbits/sec                  
[  5]   7.00-8.00   sec  32.2 MBytes   270 Mbits/sec                  
[  5]   8.00-9.00   sec  32.5 MBytes   273 Mbits/sec                  
[  5]   9.00-10.00  sec  30.4 MBytes   255 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   295 MBytes   248 Mbits/sec                  sender
[  5]   0.00-10.00  sec   295 MBytes   248 Mbits/sec                  receiver

iperf Done.
```

And seaberry (AX210) to Mac (10G wired):

```
$ iperf3 -c 10.0.100.100
Connecting to host 10.0.100.100, port 5201
[  5] local 10.0.100.27 port 39026 connected to 10.0.100.100 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   120 MBytes  1.01 Gbits/sec    0   3.29 MBytes       
[  5]   1.00-2.00   sec   148 MBytes  1.23 Gbits/sec    0   3.51 MBytes       
[  5]   2.00-3.00   sec   144 MBytes  1.21 Gbits/sec    0   3.90 MBytes       
[  5]   3.00-4.00   sec   144 MBytes  1.21 Gbits/sec    0   3.90 MBytes       
[  5]   4.00-5.00   sec   145 MBytes  1.22 Gbits/sec    0   3.90 MBytes       
[  5]   5.00-6.00   sec   144 MBytes  1.21 Gbits/sec    0   3.90 MBytes       
[  5]   6.00-7.00   sec   139 MBytes  1.16 Gbits/sec    0   3.90 MBytes       
[  5]   7.00-8.00   sec   136 MBytes  1.14 Gbits/sec    0   3.90 MBytes       
[  5]   8.00-9.00   sec   138 MBytes  1.15 Gbits/sec    0   3.90 MBytes       
[  5]   9.00-10.00  sec   138 MBytes  1.15 Gbits/sec    0   3.90 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.36 GBytes  1.17 Gbits/sec    0             sender
[  5]   0.00-10.01  sec  1.36 GBytes  1.17 Gbits/sec                  receiver

iperf Done.
```

I was hoping it would've stayed at 1.21 Gbps, since that sounds similar to 1.21 gigawatts, which would've made a great scifi reference.

Anyways, so it seems I can get a lot better TX performance than RX, with a couple mobile antenna and 67/70 link quality.

geerlingguy, 2021-11-22:
(Also noting this is through my [ASUS RT-AX86U](https://amzn.to/3HGW29A) router, which is WiFi 6, but not 6e.)

geerlingguy, 2021-11-22:
Need to summarize my findings on the page on the site before closing out this issue.

lealog, 2021-11-23:
What is the hardware that you are using to connect ax210 into cm4? Can you please show me the full list of hardware that you are using? I'm assuming you have several pieces (cm4, PCI adapter, etc)?

angelia-gt, 2021-11-23:
Can you tell us what is your order number?from which platform?
Thanks in advance

龚婷

Fenvi Technology Co., Ltd/国际部

15820354009

geerlingguy, 2021-11-23:
@angelia-gt - I'm not sure what you're asking... I order most of these things from Amazon, and you seem to be asking a lot about order numbers...?

geerlingguy, 2021-11-23:
@lealog - I've been testing through Alftel's [Seaberry board](https://pipci.jeffgeerling.com/boards_cm/seaberry.html), and through their [12-slot M.2 A+E key carrier](https://pipci.jeffgeerling.com/cards_m2/alftel-12x-pcie-m2-carrier-board.html) on a CM4 IO Board as well.

lu-zero, 2021-12-30:
Did you try with smaller carrier board as well? (do a A+E M.2 carrier exists?)

aromond, 2022-01-25:
Just wanted to say I've been trying to get multiple of these cards working on the seaberry. I was finally able to get 4 ax210 cards working in it after compiling the 5.16 kernel for raspbian 32 bit. had to make sure I included the iwlwifi and iwlmvm drivers in the kernel.

jackloomen, 2022-01-27:
> Just wanted to say I've been trying to get multiple of these cards working on the seaberry. I was finally able to get 4 ax210 cards working in it after compiling the 5.16 kernel for raspbian 32 bit. had to make sure I included the iwlwifi and iwlmvm drivers in the kernel.

That seems interesting, are you trying a bonded 4gbit mesh across 4 channels? ax210 shouldn't be able to do 6ghz or 5ghz software AP.

Can you share the iw list of ax210 and whether or not it can operate 6ghz with 5ghz or 2.4ghz simultaneously?

aromond, 2022-01-27:
> That seems interesting, are you trying a bonded 4gbit mesh across 4 channels? ax210 shouldn't be able to do 6ghz or 5ghz software AP.
>
> Can you share the iw list of ax210 and whether or not it can [...]

Sorry did not intend to mislead here, for my application I was more interested in monitor mode and capturing. for iw list I see the following:

                Frequencies:
                        * 5955 MHz [1] (22.0 dBm) (no IR)
                        * 5975 MHz [5] (22.0 dBm) (no IR)
                        * 5995 MHz [9] (22.0 dBm) (no IR)
                        * 6015 MHz [13] (22.0 dBm) (no IR)
                        * 6035 MHz [17] (22.0 dBm) (no IR)
                        * 6055 MHz [21] (22.0 dBm) (no IR)
                        * 6075 MHz [25] (22.0 dBm) (no IR)
                        * 6095 MHz [29] (22.0 dBm) (no IR)
                        * 6115 MHz [33] (22.0 dBm) (no IR)
                        * 6135 MHz [37] (22.0 dBm) (no IR)
                        * 6155 MHz [41] (22.0 dBm) (no IR)
                        * 6175 MHz [45] (22.0 dBm) (no IR)
                        * 6195 MHz [49] (22.0 dBm) (no IR)
                        * 6215 MHz [53] (22.0 dBm) (no IR)
                        * 6235 MHz [57] (22.0 dBm) (no IR)
                        * 6255 MHz [61] (22.0 dBm) (no IR)
                        * 6275 MHz [65] (22.0 dBm) (no IR)
                        * 6295 MHz [69] (22.0 dBm) (no IR)
                        * 6315 MHz [73] (22.0 dBm) (no IR)
                        * 6335 MHz [77] (22.0 dBm) (no IR)
                        * 6355 MHz [81] (22.0 dBm) (no IR)
                        * 6375 MHz [85] (22.0 dBm) (no IR)
                        * 6395 MHz [89] (22.0 dBm) (no IR)
                        * 6415 MHz [93] (22.0 dBm) (no IR)
                        * 6435 MHz [97] (22.0 dBm) (no IR)
                        * 6455 MHz [101] (22.0 dBm) (no IR)
                        * 6475 MHz [105] (22.0 dBm) (no IR)
                        * 6495 MHz [109] (22.0 dBm) (no IR)
                        * 6515 MHz [113] (22.0 dBm) (no IR)
                        * 6535 MHz [117] (22.0 dBm) (no IR)
                        * 6555 MHz [121] (22.0 dBm) (no IR)
                        * 6575 MHz [125] (22.0 dBm) (no IR)
                        * 6595 MHz [129] (22.0 dBm) (no IR)
                        * 6615 MHz [133] (22.0 dBm) (no IR)
                        * 6635 MHz [137] (22.0 dBm) (no IR)
                        * 6655 MHz [141] (22.0 dBm) (no IR)
                        * 6675 MHz [145] (22.0 dBm) (no IR)
                        * 6695 MHz [149] (22.0 dBm) (no IR)
                        * 6715 MHz [153] (22.0 dBm) (no IR)
                        * 6735 MHz [157] (22.0 dBm) (no IR)
                        * 6755 MHz [161] (22.0 dBm) (no IR)
                        * 6775 MHz [165] (22.0 dBm) (no IR)
                        * 6795 MHz [169] (22.0 dBm) (no IR)
                        * 6815 MHz [173] (22.0 dBm) (no IR)
                        * 6835 MHz [177] (22.0 dBm) (no IR)
                        * 6855 MHz [181] (22.0 dBm) (no IR)
                        * 6875 MHz [185] (22.0 dBm) (no IR)
                        * 6895 MHz [189] (22.0 dBm) (no IR)
                        * 6915 MHz [193] (22.0 dBm) (no IR)
                        * 6935 MHz [197] (22.0 dBm) (no IR)
                        * 6955 MHz [201] (22.0 dBm) (no IR)
                        * 6975 MHz [205] (22.0 dBm) (no IR)
                        * 6995 MHz [209] (22.0 dBm) (no IR)
                        * 7015 MHz [213] (22.0 dBm) (no IR)
                        * 7035 MHz [217] (22.0 dBm) (no IR)
                        * 7055 MHz [221] (22.0 dBm) (no IR)
                        * 7075 MHz [225] (22.0 dBm) (no IR)
                        * 7095 MHz [229] (22.0 dBm) (no IR)
                        * 7115 MHz [233] (22.0 dBm) (no IR)
If my understanding is correct this means I cannot initiate Radiation, meaning It can't be used as an access point. I think it should still work in monitor mode listening however. unfortunately I don't have a wifi 6e router to test with. My goal is using this with kismet wireless for wardriving and capturing primarily.

jackloomen, 2022-01-27:
> If my understanding is correct this means I cannot initiate Radiation

That's correct. No AP mode unfortunately.

Regarding iw list can you also show valid interface combinations?

example:
```
valid interface combinations:
		* #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
		  total <= 2, #channels <= 1, STA/AP BI must match
```

Thanks.

scrappylabs, 2022-04-02:
@geerlingguy love the YouTube channel! Our university research lab is trying to do some 6e stuff. Per your guidance we got the AX210 recognized, the RPI sees the 6e SSID from our 6e router, but it is grayed out. Can’t connect to it. No errors in dmesg, just lots of warnings about some iwl calls going across PCIe bus. I’m not sure what is going on any idea? Using the intel 59 firmware. 66/67/68 won’t work. 63 works, but still can’t connect over 6e band. 2.4, and wifi 6 works great.

scrappylabs, 2022-04-02:
> @geerlingguy love the YouTube channel! Our university research lab is trying to do some 6e stuff. Per your guidance we got the AX210 recognized, the RPI sees the 6e SSID from our 6e router, but it [...]

Hey ya'll praise the LORD Jesus Christ! I just wanted to say that my issue is solved. The issue is the version of wpa_supplicant.

I followed this tutorial: [here](https://medium.com/@barry_wu.me/how-to-connect-wi-fi-6e-ap-over-the-6-ghz-band-with-ubuntu-21-04-2a0b4f071ac2)

Here is where I deviated:

When installing the new wpa_supplicant, the latest is now v2.11:
`pi@raspberrypi1:~ $ wpa_supplicant -v` 
`wpa_supplicant v2.11-devel-hostap_2_10-170-gc5d9f9064`
`Copyright (c) 2003-2022, Jouni Malinen <j@w1.fi> and contributors`

Be sure you use the exact wpa_supplicant, from the tutorial.
I did not have my supplicant know about the 2.4GHz or 5GHz, just the 6GHz
Be sure to feed the wpa_supplicant config to wpa_supplicant: 

`sudo wpa_supplicant -B -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlan0`

And it connects!
On the desktop the wifi network will still be grayed out, but it will be connected.

Boom:
`pi@raspberrypi1:~ $ iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:"NETGEAR25-6G"
          Mode:Managed  Frequency:**6.135 GHz**  Access Point: 80:CC:9C:2C:7A:7D
          Bit Rate=490 Mb/s   Tx-Power=22 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=70/70  Signal level=-40 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:12   Missed beacon:0`

geerlingguy, 2022-04-04:
@scrappylabs - Now that you have it working... it'd be interesting to see if you could run some speedtests to some more-than-1 Gbps targets and see how fast you can get it. I see you have 70/70 signal at -40 dBm, though bit rate seems low for that kind of signal if it's a newer 6e AP.

scrappylabs, 2022-04-04:
@geerlingguy Would really appreciate your help with that. I was looking through your blog but couldn't find the configurations you use for iperf3 for your tests. I can try to change the bit rate if you know the iwconfig commands, I think the card is just selecting a lower bitrate, I may be able to force it. The AP says the link is 1738Mbps, we know uplink can be slower than downlink.  If it would be easier to discuss I'm open to a zoom call.

Also we didn't need to apply the iwlwifi.patch. Apparently that has been fixed as of kernel 5.15, at least for the AX210. I do get some memory warnings regarding the iwl driver in dmesg.

geerlingguy, 2022-04-04:
@scrappylabs - For iperf3 the ideal would be to run it on the AP itself but that's not usually possible. Failing that, on another computer on the network (ideally connected through 2.5+ Gbps connection) run `iperf3 -s`. Then on the Pi that's connected to 6e, run `iperf3 -c [ip of main computer running iperf3 -s here]`.

That will get TCP bandwidth from Pi to the main computer. Then to get the reverse, run separately `iperf3 -c [ip of main computer running iperf3 -s here] -R`. There's an option to run `--bidir` to run both tests at once, too, though sometimes that can behave strangely.

ayyyuki1, 2022-09-11:
@geerlingguy AP mode can be made to work

https://github.com/coolsnowwolf/lede/issues/9976#issuecomment-1242657036

credits to https://tildearrow.org/?p=post&month=7&year=2022&item=lar

paulwratt, 2022-09-11:
nice, and presetting the country code gets around those issues the original author had ..

geerlingguy, 2022-09-11:
Now I have extra incentive to get the new 6E AP Netgear sent me working, so I can do some more testing...

ayyyuki1, 2022-09-11:
> Now I have extra incentive to get the new 6E AP Netgear sent me working, so I can do some more testing...

you can make one now using MT7921k, MT7922, and MT7921u
https://github.com/morrownr/USB-WiFi/issues/104#issuecomment-1235173344

CN-Goic, 2022-09-17:
[我刚买了AX210网卡，目前内核5.15，PVE驱动都已更新，Openwrt无法加载无线模块，向你们学习下怎么弄

paulwratt, 2022-09-18:
> 我刚买了AX210网卡，目前内核5.15，PVE驱动都已更新，Openwrt无法加载无线模块，向你们学习下怎么弄

according to translate.google.com ..

> I just bought the AX210 network card, the current kernel 5.15, the PVE driver has been updated, Openwrt cannot load the wireless module, I will learn from you how to do it

CN-Goic, 2022-09-19:
问题解决了，我的PVE有驱动，但是虚拟化的openwrt里缺少驱动，将ucode放入/lib/fireware/，
/etc/pve/local/qemu-server/100.conf 在hostpci0 下面一行加入 args: -set device.hostpci0.x-msix-relocation=bar2可以避免openwrt死机
今天测试开启了AP模式 但只有WIFI6E+2.4GHz可以用，如果选择WIFI6E+5Ghz有发射信号但设备无法识别出来，不知道是不是手机的问题；
需要手动选择信道，开启MU-MIMO，关闭WMM模式和强制40M频宽选项，网络选择lan

CN-Goic, 2022-09-19:
问题解决了，我的PVE有驱动，但是虚拟化的openwrt里缺少驱动，先将ucode放入/lib/fireware/
PVE直通AX210会导致openwrt死机，解决办法是：
vi /etc/pve/local/qemu-server/100.conf 
在hostpci0 下面一行加入 args: -set device.hostpci0.x-msix-relocation=bar2

今天测试开启了AP模式 但只有WIFI6E+2.4GHz可以用，如果选择WIFI6E+5Ghz有发射信号但设备无法识别出来，不知道是不是手机的问题；
需要手动选择信道，开启MU-MIMO，关闭WMM模式和强制40M频宽选项，网络选择lan

paulwratt, 2022-09-19:
this time providing help (translate.google.com)

> The problem is solved, my PVE has a driver, but the virtualized openwrt lacks the driver,
> first put the ucode into /lib/fireware/PVE through AX210 will cause openwrt to crash, the solution is:
> `vi /etc/pve/local/qemu-server/100.conf`
> Add args to the line below
> `hostpci0: -set device.hostpci0.x-msix-relocation=bar2`
> 
> Today's test turned on the AP mode, but only WIFI6E+2.4GHz can be used. If you choose WIFI6E+5Ghz, there is a 
> transmitting signal but the device cannot recognize it. I don't know if it is a problem with the mobile phone;
> Need to manually select the channel, turn on MU-MIMO, turn off WMM mode and forced 40M bandwidth option,
> choose lan for network

paulwratt, 2022-09-19:
> Today's test turned on the AP mode, but only WIFI6E+2.4GHz can be used.

A couple of days ago I red a post somewhere about why this happens. The Intel device _driver_ no longer allows setting the Country code before startup. If you do, it will automatically turn off 5Ghz.

The solutions was to turn the device on within range of another WiFi AP and let the _drivers_ startup auto-scan _detect_ the Country Code from the _other device_ . You _do not_ need to connect to the AP/device, just be with range of it when the Intel device does the auto-scan, after which _you can_ set the Country Code.

Hope that helps (yay for Intel being nice to users and programmers - bah)

ayyyuki1, 2022-09-19:
@paulwratt with tildearrow's patch, it does do the scan and if it finds an AP around, it uses that country code
heres the kicker, when theres no 5GHz AP around, it will accept the hint as long as it did the scan. Hope it helps.

samwiseg0, 2022-09-22:
> What is the hardware that you are using to connect ax210 into cm4? Can you please show me the full list of hardware that you are using? I'm assuming you have several pieces (cm4, PCI adapter, etc)?

If you would like a more compact solution there is [this board from MCUZone](https://www.aliexpress.com/item/3256803637327862.html). They have a PoE and non PoE variant. 

p.s. Thanks Jeff for the detailed information and your cross compile platform. It made the process of getting this to work much easier.

paulwratt, 2022-09-22:
> with tildearrow's patch

I am pretty sure that the same guy that wrote the article, he wrote a patch for something specific (wpa_supplicant?). The imprtant thing to not tho, it that _if there is no 5G devices around when the driver scans, 5G will be disabled_ .

ayyyuki1, 2022-09-23:
@paulwratt 
5Ghz AP will work as it will accept the 802.11d hint if theres no 5GHz AP around. but you really have to do that scan. intel's lar is weird like that

geerlingguy, 2022-11-09:
Back to doing some testing on this over the next week, now that I have a proper 6E AP at my house over 2.5G wired Ethernet.

geerlingguy, 2022-11-10:
New round of tests. Using built-in WiFi:

```
pi@wifitest:~ $ iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"nebula"  
          Mode:Managed  Frequency:2.412 GHz  Access Point: C8:9E:43:13:4A:41   
          Bit Rate=24 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=70/70  Signal level=-35 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:40  Invalid misc:0   Missed beacon:0

pi@wifitest:~ $ iperf3 -c 10.0.100.15
Connecting to host 10.0.100.15, port 5201
[  5] local 10.0.100.163 port 33512 connected to 10.0.100.15 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  6.41 MBytes  53.8 Mbits/sec    0    348 KBytes       
[  5]   1.00-2.00   sec  6.21 MBytes  52.1 Mbits/sec    0    506 KBytes       
[  5]   2.00-3.00   sec  5.85 MBytes  49.1 Mbits/sec    0    653 KBytes       
[  5]   3.00-4.00   sec  6.25 MBytes  52.4 Mbits/sec    0    724 KBytes       
[  5]   4.00-5.00   sec  1.25 MBytes  10.5 Mbits/sec    0    762 KBytes       
[  5]   5.00-6.00   sec  2.50 MBytes  21.0 Mbits/sec    2    553 KBytes       
[  5]   6.00-7.00   sec  1.25 MBytes  10.5 Mbits/sec    2    392 KBytes       
[  5]   7.00-8.00   sec  2.50 MBytes  21.0 Mbits/sec    0    428 KBytes       
[  5]   8.00-9.00   sec  5.00 MBytes  41.9 Mbits/sec    0    455 KBytes       
[  5]   9.00-10.00  sec  5.00 MBytes  41.9 Mbits/sec    0    471 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  42.2 MBytes  35.4 Mbits/sec    4             sender
[  5]   0.00-10.02  sec  39.5 MBytes  33.1 Mbits/sec                  receiver

iperf Done.
```

I made sure to disable the built-in Ethernet during the test with `sudo ifconfig eth0 down`.

And after [forcing wlan1 instead of wlan0](https://www.jeffgeerling.com/blog/2021/working-multiple-wifi-interfaces-on-raspberry-pi), here are the results with the AX210 from about 15' away through two drywall walls:

```
TODO
```

geerlingguy, 2022-11-10:
Testing is complete and I updated [its page on the CM4 DB site](https://pipci.jeffgeerling.com/cards_network/intel-ax210-wifi-6e.html), so I'm going to close this issue—but feel free to continue discussion here, especially concerning AP performance!

closed 2022-11-10 by geerlingguy

dcousens, 2022-11-10:
@geerlingguy did you try the AX200/AX210's on >40MHz bandwidth?

I have a few of each, and neither of them can hold-up without packet loss and drop outs.

geerlingguy, 2022-11-10:
@dcousens - Hmm... I just realized this thing is connecting on the 5 GHz network too... Apparently I need to upgrade to the latest wpa_supplicant?

```
pi@wifitest:~ $ iwconfig wlan1
wlan1     IEEE 802.11  ESSID:"nebula"  
          Mode:Managed  Frequency:5.785 GHz  Access Point: C8:9E:43:13:4A:61   
          Bit Rate=117 Mb/s   Tx-Power=22 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=70/70  Signal level=-19 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:29   Missed beacon:0

pi@wifitest:~ $ wpa_supplicant -v
wpa_supplicant v2.9
Copyright (c) 2003-2019, Jouni Malinen <j@w1.fi> and contributors
```

dcousens, 2022-11-10:
`iw dev wlan1 info` might be easier for retrieving that information

geerlingguy, 2022-11-10:
Indeed:

```
pi@wifitest:~ $ iw dev wlan1 info
Interface wlan1
	ifindex 4
	wdev 0x1
	addr 84:5c:f3:f6:e9:29
	ssid nebula
	type managed
	wiphy 0
	channel 157 (5785 MHz), width: 40 MHz, center1: 5795 MHz
	txpower 22.00 dBm
	multicast TXQ:
		qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
		0	0	0	0	0	0	0	0		0
```

geerlingguy, 2022-11-10:
I tried running `wpa_supplicant` in debug mode:

```
pi@wifitest:~ $ sudo pkill wpa_supplicant
pi@wifitest:~ $ sudo wpa_supplicant -dd -i wlan1 -c /etc/wpa_supplicant/wpa_supplicant-wlan1.conf 
wpa_supplicant v2.10
random: getrandom() support available
Successfully initialized wpa_supplicant
Initializing interface 'wlan1' conf '/etc/wpa_supplicant/wpa_supplicant-wlan1.conf' driver 'default' ctrl_interface 'N/A' bridge 'N/A'
...
wlan1: Selecting BSS from priority group 0
wlan1: 0: c8:9e:43:13:4a:81 ssid='nebula-6g' wpa_ie_len=0 rsn_ie_len=20 caps=0x511 level=-27 freq=6135 
wlan1:    skip RSN IE - key mgmt mismatch
wlan1:    reject due to mismatch with WPA/WPA2
wlan1: No suitable network found
wlan1: Already sched scanning
p2p-dev-wlan1: Updating scan results from sibling
```

(Note: I couldn't just `sudo systemctl stop wpa_supplicant` because the process was still running in the background. I had to `pkill` it outright.)

geerlingguy, 2022-11-10:
Ah, [ArchLinux Wiki on WPA3 wpa_supplicant configuration](https://wiki.archlinux.org/title/wpa_supplicant#Connections_to_pure_WPA3-SAE_access_points) to the rescue! I had to update my wpa_supplicant.conf block to:

```
network={
        ssid="SSID_HERE"
        psk="PASSWORD_HERE"
        key_mgmt=SAE
        ieee80211w=2
}
```

Note the addition of the latter two options. With that:

```
pi@wifitest:~ $ iw dev wlan1 info
TODO
```

dcousens, 2022-11-10:
@geerlingguy did that genuinely print `TODO`? haha

geerlingguy, 2022-11-10:
@dcousens - Heh... I often throw in a TODO if I'm writing a long comment and want to make sure part of it is saved so I don't lose formatting or any thoughts. I've been burned too many times by GitHub's "You can't comment at this time" feature :D

brcisna, 2023-01-02:
@geerlingguy , Hi Jeff,,,First off,,Hope you are getting recouped from your major surgery,,,Being at hospital is never a fun ordeal.
Have looked high and low,,,but never could find an example of your hostapd.conf file, Searching for this everything is all over the map for trying to get card to work in 5GHz mode.
trying to get an old intel 7265 to work in 5GHz just to see if it will work,Of course all the 5GHz frequencies are blocked,,Tried the patch from tildearrow many hours on it,,still no frequencies are available,, waiting on an Intel AX200 to arrive here,,,This is just my practice card.

geerlingguy, 2023-01-05:
I just posted a video and blog post on the AX210 with the CM4:

  - https://www.youtube.com/watch?v=TdUxWEyafUg
  - https://www.jeffgeerling.com/blog/2023/getting-15-gbps-wifi-6e-on-raspberry-pi-cm4

brcisna, 2023-01-05:
Thanks Jeff for links,,,

Hope you are getting recouped from your hospital stay,
Ill check out the Youtube and blog post for info .
Have a good day,

Barry Cisna

me-peng, 2023-03-01:
I bought the same NETGEAR device as Jeff Geerling. Raspberry pie is also installed with AX210, and can normally connect to 5G and 2.4G WIFI. When the 6G WIFI is turned on, raspberry pie cannot search for WIFI, which is also the 5G band of WPA3

Excuse me, what's the problem

There are different versions of the AX210. Will different versions directly affect it?

Is it related to regional settings?

chris78123, 2023-10-20:
Hi Jeff and everyone else,

I am trying to use the AX210 on a CM4 (rpi 64bit). The CM4 has only eMMC (32G) no SD card. 
I want to use the kernel rpi-5.15.y and I compiled the iwlwifi modules on the pi (quick test), directly into the kernel (not as a module) and load the firmware from _https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git_ to _/lib/firmware/iwlwifi-xxx_ .

The new kernel will boot without errors even with the iwlwifi. 

BUT the iwlwifi does not find any firmware:

_iwlwifi: .... Direct loading of the firmware for xxx fails with error -2_ .

 this happens with all iwlwifi-ty-xxxx

but the firmware is absolutely copied to /lib/firmware/ iwlwify-ty-xxxx, and also the permissions are standard.

Any idea why the kernel can't find the firmware?

Thanks for any help!

BR Chris

RomanKubicaNXP, 2024-04-09:
> Hi Jeff and everyone else,
> 
> I am trying to use the AX210 on a CM4 (rpi 64bit). The CM4 has only eMMC (32G) no SD card. I want to use the kernel rpi-5.15.y and I compiled the iwlwifi modules on the pi (quick test), directly into the kernel (not as a module) .....

Hi Chris and everyone,

did you resolve this issue?
I have same behavior with 5.15.. kernel

Thanks, Roman

Roman124K5, 2024-04-09:
> Hi Jeff and everyone else,
> 
> I am trying to use the AX210 on a CM4 (rpi 64bit). The CM4 has only eMMC (32G) no SD card. I want to use the kernel rpi-5.15.y ....

Hi Chris and everyone, 

Did you resolve this issue??
I have same problem with 5.15 kernel.

iwlwifi: .... Direct loading of the firmware for xxx fails with error -2 
both IWLWIFI=y and IWLMVM=y are enabled

Thanks, 
Roman

[24 more replies]