geerlingguy/raspberry-pi-pcie-devices #123: Test Nvidia GeForce GTX 1080 Ti

geerlingguy, 2021-04-30:
A good friend (@CajuCLC) graciously offered to loan me his [Nvidia GeForce GTX 1080 Ti](https://amzn.to/3t5j6q2) (not sure exactly what version it is... I'll get it soon and update this post).

TODO: Picture here.

I'll try Nvidia's proprietary ARM driver first, if that fails, I'll try recompiling with Nouveau. And I might need to expand the BAR space if the card has a ton of RAM, not sure.

CajuCLC, 2021-04-30:
Hey!
It's a GTX 1080 and this is the one: [Newegg](https://www.newegg.com/msi-geforce-gtx-1080-gtx-1080-gaming-x-8g/p/N82E16814127943).
I did NOT pay $2k for such card. Just realized how expensive it is now. Wanna buy it? 😆

geerlingguy, 2021-05-13:
`lspci -vvvv` gives:

```
01:00.0 VGA compatible controller: NVIDIA Corporation GP104 [GeForce GTX 1080] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: Micro-Star International Co., Ltd. [MSI] GP104 [GeForce GTX 1080]
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at 618000000 (32-bit, non-prefetchable) [disabled] [size=16M]
	Region 1: Memory at 600000000 (64-bit, prefetchable) [disabled] [size=256M]
	Region 3: Memory at 610000000 (64-bit, prefetchable) [disabled] [size=32M]
	Region 5: I/O ports at <unassigned> [disabled]
	[virtual] Expansion ROM at 619000000 [disabled] [size=512K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [78] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 <64us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <512ns, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range AB, TimeoutDis+, LTR+, OBFF Via message
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [250 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [128 v1] Power Budgeting <?>
	Capabilities: [420 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [600 v1] Vendor Specific Information: ID=0001 Rev=1 Len=024 <?>
	Capabilities: [900 v1] #19

01:00.1 Audio device: NVIDIA Corporation GP104 High Definition Audio Controller (rev a1)
	Subsystem: Micro-Star International Co., Ltd. [MSI] GP104 High Definition Audio Controller
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin B routed to IRQ 0
	Region 0: Memory at 619080000 (32-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable- Count=1/1 Maskable- 64bit+
[... 2 lines cut ...]
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 <64us
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x16, ASPM L0s L1, Exit Latency L0s <512ns, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range AB, TimeoutDis+, LTR+, OBFF Via message
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
```

(Note: Had to power the card through an external 700W PSU so I could put power through the 8-pin and 6-pin connectors.)

geerlingguy, 2021-05-13:
`dmesg` shows:

```
[    1.238621] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.238721] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.238876] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.239029] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00ffffffff -> 0x0400000000
[    1.290024] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.293155] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.295889] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.298700] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.301560] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.304596] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.310918] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.313920] pci 0000:01:00.0: [10de:1b80] type 00 class 0x030000
[    1.316690] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00ffffff]
[    1.319409] pci 0000:01:00.0: reg 0x14: [mem 0x00000000-0x0fffffff 64bit pref]
[    1.322139] pci 0000:01:00.0: reg 0x1c: [mem 0x00000000-0x01ffffff 64bit pref]
[    1.324827] pci 0000:01:00.0: reg 0x24: [io  0x0000-0x007f]
[    1.327508] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0007ffff pref]
[    1.330498] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 126.016 Gb/s with 8.0 GT/s PCIe x16 link)
[    1.333456] pci 0000:01:00.0: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[    1.336352] pci 0000:01:00.1: [10de:10f0] type 00 class 0x040300
[    1.339169] pci 0000:01:00.1: reg 0x10: [mem 0x00000000-0x00003fff]
[    1.345756] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.348580] pci 0000:00:00.0: BAR 9: assigned [mem 0x600000000-0x617ffffff 64bit pref]
[    1.351398] pci 0000:00:00.0: BAR 8: assigned [mem 0x618000000-0x6197fffff]
[    1.354204] pci 0000:01:00.0: BAR 1: assigned [mem 0x600000000-0x60fffffff 64bit pref]
[    1.357001] pci 0000:01:00.0: BAR 3: assigned [mem 0x610000000-0x611ffffff 64bit pref]
[    1.359734] pci 0000:01:00.0: BAR 0: assigned [mem 0x618000000-0x618ffffff]
[    1.362422] pci 0000:01:00.0: BAR 6: assigned [mem 0x619000000-0x61907ffff pref]
[    1.365073] pci 0000:01:00.1: BAR 0: assigned [mem 0x619080000-0x619083fff]
[    1.367675] pci 0000:01:00.0: BAR 5: no space for [io  size 0x0080]
[    1.370266] pci 0000:01:00.0: BAR 5: failed to assign [io  size 0x0080]
[    1.372830] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.375350] pci 0000:00:00.0:   bridge window [mem 0x618000000-0x6197fffff]
[    1.377888] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x617ffffff 64bit pref]
[    1.380585] pci 0000:01:00.1: D0 power state depends on 0000:01:00.0
```

geerlingguy, 2021-05-13:
Driver install attempt number 1:

  1. Downloaded [460.80 Driver](https://www.nvidia.com/Download/driverResults.aspx/175204/en-us) from Nvidia's website.
  2. Installed `raspberrypi-kernel-headers`
  3. Ran the driver `.run` file.

geerlingguy, 2021-05-13:
Nvidia driver resulted in the following output:

```
[   38.399643] NVRM: GPU 0000:01:00.0: RmInitAdapter failed! (0x31:0xffff:1026)
[   38.399748] NVRM: GPU 0000:01:00.0: rm_init_adapter failed, device minor number 0
[   38.523167] Unable to handle kernel write to read-only memory at virtual address ffffffc0096dc3e1
[   38.532214] Mem abort info:
[   38.535043]   ESR = 0x9600004f
[   38.538154]   EC = 0x25: DABT (current EL), IL = 32 bits
[   38.543548]   SET = 0, FnV = 0
[   38.546643]   EA = 0, S1PTW = 0
[   38.549832] Data abort info:
[   38.552757]   ISV = 0, ISS = 0x0000004f
[   38.556654]   CM = 0, WnR = 1
[   38.559670] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000001096000
[   38.566476] [ffffffc0096dc3e1] pgd=00000000fbfff003, p4d=00000000fbfff003, pud=00000000fbfff003, pmd=000000004da2f003, pte=0040000049700783
[   38.579212] Internal error: Oops: 9600004f [#1] PREEMPT SMP
[   38.584860] Modules linked in: bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc nvidia_drm(PO) nvidia_modeset(PO) nvidia(PO) brcmfmac bcm2835_isp(C) bcm2835_codec(C) brcmutil bcm2835_v4l2(C) vc4 v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig v3d cec videobuf2_memops cfg80211 videobuf2_v4l2 gpu_sched drm_kms_helper rfkill drm videobuf2_common drm_panel_orientation_quirks raspberrypi_hwmon videodev snd_soc_core snd_bcm2835(C) mc snd_compress vc_sm_cma(C) snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight rpivid_mem uio_pdrv_genirq uio aes_neon_bs sha256_generic aes_neon_blk crypto_simd cryptd ip_tables x_tables ipv6
[   38.648031] CPU: 3 PID: 584 Comm: nvidia-smi Tainted: P         C O      5.10.17-v8+ #1414
[   38.656412] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   38.662943] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
[   38.670643] pc : _nv008121rm+0x30/0x348 [nvidia]
[   38.676972] lr : _nv031679rm+0x124/0x918 [nvidia]
[   38.681741] sp : ffffffc011f03630
[   38.685097] x29: ffffffc011f03630 x28: ffffff8045f6d800 
[   38.690484] x27: ffffff804e3c8008 x26: 0000000000000000 
[   38.695870] x25: 0000000000000068 x24: 0000000000000001 
[   38.701256] x23: ffffff8045fa4808 x22: 0000000000000000 
[   38.706641] x21: ffffff804e0e4008 x20: ffffff8045e9c808 
[   38.712027] x19: ffffffc0096db608 x18: ffffffc0099127c8 
[   38.717412] x17: ffffffc0099127d8 x16: ffffffc0099127f8 
[   38.722798] x15: ffffffc009912808 x14: ffffffc00aee8000 
[   38.728184] x13: ffffffc00aee8000 x12: ffffffc00913d650 
[   38.733570] x11: 00000000000193e0 x10: 0000000000000000 
[   38.738956] x9 : ffffffc009178eb4 x8 : 0000000000000000 
[   38.744341] x7 : 0000000000000001 x6 : ffffffc011f035f0 
[   38.749727] x5 : 0000000000000000 x4 : ffffffc00951aa08 
[   38.755112] x3 : ffffffc009515168 x2 : 0000000000000001 
[   38.760497] x1 : ffffffc0096db608 x0 : ffffff8045e9c808 
[   38.765884] Call trace:
[   38.770012]  _nv008121rm+0x30/0x348 [nvidia]
[   38.775981]  _nv031679rm+0x124/0x918 [nvidia]
[   38.782035]  _nv031712rm+0xd4/0x268 [nvidia]
[   38.788002]  _nv031678rm+0x5c/0x258 [nvidia]
[   38.793980]  _nv000670rm+0xa78/0x1b48 [nvidia]
[   38.800085]  rm_init_adapter+0xb0/0xc0 [nvidia]
[   38.806255]  nv_open_device+0x420/0x6f8 [nvidia]
[   38.812515]  nvidia_open+0xec/0x398 [nvidia]
[   38.818420]  nvidia_frontend_open+0x74/0xc0 [nvidia]
[   38.823462]  chrdev_open+0xb0/0x1a8
[   38.826997]  do_dentry_open+0x134/0x398
[   38.830883]  vfs_open+0x34/0x40
[   38.834064]  path_openat+0xa24/0xe20
[   38.837684]  do_filp_open+0x84/0x100
[   38.841304]  do_sys_openat2+0x1f8/0x2a8
[   38.845189]  do_sys_open+0x60/0xa8
[   38.848633]  __arm64_sys_openat+0x2c/0x38
[   38.852696]  el0_svc_common.constprop.2+0xac/0x1d0
[   38.857550]  do_el0_svc+0x2c/0x98
[   38.860909]  el0_svc+0x20/0x30
[   38.864000]  el0_sync_handler+0x90/0xb8
[   38.867886]  el0_sync+0x174/0x180
[   38.871245] Code: 12001c79 f90013f5 2a0203f8 f90037bf (39376439) 
[   38.877424] ---[ end trace c398f5dd6e0c32ac ]---
```

(Thanks to alex and a bunch of other people for suggesting `nvidia-smi` to get that debug output.)

uruha-komachin, 2021-05-13:
You can edit the `nvidia-linux.h` in the official driver and comment out the `NVCPU_X86`, `NVCPU_X86_64` part

```c
static inline int nv_kmem_cache_alloc_stack(nvidia_stack_t **stack)
{
    nvidia_stack_t *sp = NULL;
//#if defined(NVCPU_X86) || defined(NVCPU_X86_64)
    sp = NV_KMEM_CACHE_ALLOC(nvidia_stack_t_cache);
    if (sp == NULL)
        return -ENOMEM;
    sp->size = sizeof(sp->stack);
    sp->top = sp->stack + sp->size;
//#endif
    *stack = sp;
    return 0;
}
```

virusz4274, 2021-05-13:
The nvidia driver you downloaded from nvidia website is aarch64 .  Isnt the raspbian os based on armhf. can that be a issue ?

Moldytzu, 2021-05-13:
> The nvidia driver you downloaded from nvidia website is aarch64 . Isnt the raspbian os (pi os)based on armhf. can that be a issue ?

he's using the 64-bit version of the os

geerlingguy, 2021-05-13:
With the nouveau driver compiled into a custom kernel build, I tried blacklisting and manually loading the nouveau module and dmesg showed this before a total system lockup:

```
[   62.330803] pci 0000:00:00.0: enabling device (0000 -> 0002)
[   62.330837] nouveau 0000:01:00.0: enabling device (0000 -> 0002)
[   62.330952] nouveau 0000:01:00.0: NVIDIA GP104 (134000a1)
[   62.450685] nouveau 0000:01:00.0: bios: version 86.04.60.00.5b
[   62.470739] nouveau 0000:01:00.0: pmu: firmware unavailable
[   62.515105] nouveau 0000:01:00.0: fb: 8192 MiB GDDR5X
client_loop: send disconnect: Broken pipe
```

masip85, 2021-05-13:
I think this driver fixing is never going to happen. Great effort against high walls, @geerlingguy . Your luck with this is outr happiness.

geerlingguy, 2021-05-13:
@uruha-komachin - I may try that, didn't know you can actually get at the source, good to know!

uruha-komachin, 2021-05-13:
@geerlingguy Hope you got some luck with the GTX 1080! I only have a GTX 3090 and the nouveau driver for this card is at pretty early stage [(see the NV170 column in the nouveau feature matrix)](https://nouveau.freedesktop.org/FeatureMatrix.html).

As for the official driver, the sad thing is that the problematic part, the `rm_*` functions (related to the NVIDIA Resource Manager), were distributed in binary form (at least I wasn't able to locate any actual source code in the embedded xz file). These `rm_*` functions can be found in the `nvidia.o` file (or the `nvidia-drm.ko` file which basically contains a copy of the `nvidia.o`) while compiling.

I copied the `nvidia.o` to another directory, did some reverse engineering, and [this (discussed in #26)](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/26#issuecomment-826120793) was what I ended up with.

At this stage, I honest think that there isn't much we can do, even if we can (to some extent) get our hands on some of the official source code. I really hope that someone in NVIDIA can give us some hint or push this forward.

geerlingguy, 2021-05-13:
@uruha-komachin - Good catch, I do wonder if the Pi Foundation could find a way to work around this in the kernel... if not it could be the major limitation for higher-end cards on the BMC2711.

paulwratt, 2021-05-14:
@uruha-komachin the following post contains:

> The BCM2711 has other DMA types called "DMA4" (40bits) that can access the full 35bit BCM2711 address range.
The PCIe bars are placed at: 0x600000000 - 0x7ffffffff and would therefore permit full 8Gig BAR access.

I had heard/read somewhere else that there was a limit, ie the 12Gb+ on some newer card was not usable over 8G (or 4Gb?)

avipars, 2021-05-14:
wait, are you allowed to put amazon affiliate links in github?

geerlingguy, 2021-05-14:
@avipars - There's no restriction on it that I'm aware of. It's not much, but using the affiliate links on this project does net enough money in that account that I can buy some of the supporting gear I need to test the cards on the project :)

Key word being 'some'.

CajuCLC, 2021-05-14:
Hey @geerlingguy , is there some PCIe for the pi that runs on x8+ mode?

geerlingguy, 2021-05-14:
@CajuCLC - No, the Pi just has a x1 lane and the IO Board has a x1 slot. But as with crypto miners, you can use expanders/adapters/switches (or just slice the x1 slot's edge) and plug in any size card, and for _most_ x4/x8/x16 cards, they'll just work as they are supposed to be able to drop down to slower speeds.

That's what the initialization section in the kernel output shows: `[    1.330498] pci 0000:01:00.0: 4.000 Gb/s available PCIe bandwidth, limited by 5.0 GT/s PCIe x1 link at 0000:00:00.0 (capable of 126.016 Gb/s with 8.0 GT/s PCIe x16 link)`

Basically the card sees the limit and _should_ react accordingly. For the network and storage cards I've tried, there haven't been any issues there. And some graphics cards seem to work fine on ARM on other platforms with non-x8/x16 slots, so in theory it can work here. I think it's the memory allocations that's the main issue.

CajuCLC, 2021-05-14:
@geerlingguy - Yeah, it will technically work, but if you need to install drivers I have seen that running on x4 or lower mode the driver installation doesn't even recognize the card as Nvidia. Windows and Ubuntu are able to output video because they have their own generic drivers that make it run. But if I open GPU-Z for example, it doesn't show as a Nvidia card at all. Only when running x8 or x16. So I wonders that's the problem with the Pi.

geerlingguy, 2021-05-14:
I just tried installing the NVIDIA driver on armv7l, but got:

```
$ sudo ./NVIDIA-Linux-aarch64-460.80.run 

ERROR: this .run file is intended for the
Linux-aarch64 platform, but you appear to be
running on Linux-armv7l.  Aborting installation.
```

Looking at their site, if I try choosing 'All Operating Systems' and then arm32, it redirects to arm64... so I don't think they maintain a 32-bit proprietary driver. Going to try the kernel recompile next on 32-bit with Nouveau and see if it makes a difference. You never know.

geerlingguy, 2021-05-14:
Nope. Same thing: `sudo modprobe nouveau` and `dmesg` shows:

```
[  189.321855] nouveau 0000:01:00.0: enabling device (0140 -> 0142)
[  189.322269] nouveau 0000:01:00.0: NVIDIA GP104 (134000a1)
[  189.444049] nouveau 0000:01:00.0: bios: version 86.04.60.00.5b
[  189.455856] nouveau 0000:01:00.0: pmu: firmware unavailable
[  189.495719] nouveau 0000:01:00.0: fb: 8192 MiB GDDR5X
```

uruha-komachin, 2021-05-14:
> @geerlingguy - Yeah, it will technically work, but if you need to install drivers I have seen that running on x4 or lower mode the driver installation doesn't even recognize the card as Nvidia. [...]

The official driver (as well as the nouveau one) did recognise the card and this can be confirmed by `ls /proc/driver/nvidia/gpus/*/information`. But any attempt to read the related kernel information (under `/proc/driver/nvidia/*`) will trigger an exception/kernel crash/panic because the official driver cannot initialise the adapter (will crash at `rm_init_adapter`).

In short, recognise the card? Yes. Can initialise the card? Not yet...

uruha-komachin, 2021-05-14:
@paulwratt I think it's safe to say any card that has VRAM <= 4GB will have a better chance with current CM4. I don't know whether we can initialise and use a graphics card with only first 4GB of its VRAM or not... But everything on CM4 is at least 32bit AFAIK, perhaps we can focus on GPU cards with VRAM <= 4GB.

Arjdroid, 2021-05-17:
One thing that is a quirk with the newer generation GPUs (estimating Nvidia 900 series and above) is that they only work on computers with UEFI. I learnt this because I've been doing some experimenting with GPU passthrough to VM's and the only way they work is with UEFI VM's not BIOS ones...

There is also something strange with Nvidia 1000 series and older cards where your VM _might_ have to supply a "patched" VBIOS / ROM to the memory in order for the GPU to work properly.

I don't know whether these same quirks apply to the Raspberry Pi but it could be worth reading.

There is a very good repository / guide which goes into detail about GPU Passthrough to VM's -

https://github.com/bryansteiner/gpu-passthrough-tutorial

Arjdroid, 2021-05-17:
> I just tried installing the NVIDIA driver on armv7l, but got:
>
> ```
> $ sudo ./NVIDIA-Linux-aarch64-460.80.run
>
> ERROR: this .run file is intended for the
> Linux-aarch64 platform, but you appear to be [...]

I don't know if you have already tried this before but...

If you're okay with running an operating system on your Raspberry Pi Compute Module 4 that is _NOT_ Raspberry Pi OS, you could try running Ubuntu 21.04 which is a 64-Bit OS and has compatability for the Raspberry Pi Compute Module 4.

It is available at https://ubuntu.com/download/raspberry-pi and since it is 64 bit, it will utilise the aarch64 platform of the CM4 rather than ARMv7 so hopefully the official nvidia drivers could work on it and you don't have to keep trying to recompile Nouveau on it either. (Maybe nouveau would also actually work because it is 64-Bit?) 

The CM4 has a 64-Bit processor it's just that Raspberry Pi OS is still 32-Bit to maintain compatibility with Pi's older than the Raspberry Pi 3 and they haven't fully moved to porting Raspberry Pi OS to 64-Bit. But Ubuntu has already made their OS for aarch64 on the Raspberry Pi.

paulwratt, 2021-05-17:
@Arjdroid  all of the Geerling PCIe card tests, kernel fixes, and driver patching is done through `aarch64` version of RPiOS - sometimes, out of pure curiosity, 32bit RPiOS will be tested, and with some older PCIe hardware, where the drivers have been in 32bit Linux Kernel for a while, they _may_ work, especially if the card is _not_ dependant on BAR space or memoy huge alloction blocks (often required by video cards).

Part of the reason for _only_ using 64bit OS is that 99% of ARM server & ARM desktop (not ARM SBC) development, was started when `armv8` appeared, which allowed 64bit coding, both of which are used heavily in the Server and Workstation environments.

FYI with standard "32bit RPiOS", if you just change the "config.txt" to boot with the **`arm_64bit=1`**, your kernel will be 64bit, even though the rest of the OS is all 32bit.
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
pi@raspberrypi:~ $ ls -l $(which gcc) $(which gcc-8)
lrwxrwxrwx 1 root root  5 Mar 19  2019 /usr/bin/gcc -> gcc-8
lrwxrwxrwx 1 root root 25 Apr 16  2019 /usr/bin/gcc-8 -> arm-linux-gnueabihf-gcc-8
pi@raspberrypi:~ $ file /usr/bin/arm-linux-gnueabihf-gcc-8
/usr/bin/arm-linux-gnueabihf-gcc-8: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=b507462b5ae8ff3ec55a4d432b7f2787e08d54b5, stripped
```

Arjdroid, 2021-05-17:
@paulwratt This is very interesting, so the kernel would be running 64 bit while all the userland programs would be 32 bit running in backwards compatibility mode...

So assuming the 64 bit driver isn't the problem, the only other problem I could see is UEFI booting because almost all the modern GPUs require UEFI to function and this issue is also pointed out in another issue, #132 which talks about possible solutions for the problem.

Has that also been covered? I'm actually pretty curious about UEFI compatibility with the Pi and what that means for future boards and operating systems.

paulwratt, 2021-05-17:
> so the kernel would be running 64 bit while all the userland programs would be 32 bit running in backwards compatibility mode...

the kernel has complete access to all address space, the way modern OS work, they get passed a _virtual address_, so it does not matter if the application can only addess 4Gb, the kernel slices up _all_ memory ad servers it to the OS (mapped through the `mmu` to the OS componants)

coolman6eorge, 2021-06-02:
> I just tried installing the NVIDIA driver on armv7l, but got:
>
> ```
> $ sudo ./NVIDIA-Linux-aarch64-460.80.run
>
> ERROR: this .run file is intended for the
> Linux-aarch64 platform, but you appear to be [...]

You can try to install [this](https://www.nvidia.com/Download/driverResults.aspx/173109/en-us) driver. It might be compatible with armv7l.

adamprice2, 2021-07-16:
Just watched your livestream on this (it was great BTW).
At the end of the stream you note this:-
 
> ```
> [  189.321855] nouveau 0000:01:00.0: enabling device (0140 -> 0142)
> [  189.322269] nouveau 0000:01:00.0: NVIDIA GP104 (134000a1)
> [  189.444049] nouveau 0000:01:00.0: bios: version 86.04.60.00.5b [...]

I wonder if the **pmu: firmware unavailable** might be part of the problem? There seems to have been some 'recentish' changes in the kernel which relate to this and perhaps they have not been incorporated in the custom pi kernel?
Note this is just uninformed speculation, so I might be entirely wrong, but the message stood out to me as something to investigate.

geerlingguy, 2021-07-16:
@adamprice2 - It seems like that might not be the problem; a lot of other people seem to have the same issue but different problems (or a working video card), like https://www.reddit.com/r/archlinux/comments/k1ns8e/nouveau_pmu_firmware_unavailable_error_while/

geerlingguy, 2021-07-16:
Marking as testing complete, because I needed to ship the card back to my friend, and unlike with AMD cards, it seems like there's little to no support from anyone internal at Nvidia who could really help push things forward to the effort :(

closed 2021-07-16 by geerlingguy

masip85, 2021-07-21:
Oh! Game over?