geerlingguy/raspberry-pi-pcie-devices #125: Test ZS ZHISHANG PCI-E X1 to 4 PCI-E X16 Expansion Riser (ASMedia ASM1184e)

geerlingguy, 2021-04-30:
I just received the [ZS ZHISHANG PCI-E X1 to 4PCI-E X16 Expansion Kit](https://amzn.to/3u8oqdJ) in the mail after a four-month roller coaster of ordering, then having orders cancelled, then ordering from other vendors... etc. until _finally_, this thing arrived.

It adapts one x1 PCIe port to 4 x16 slots, with an independent molex 4-pin power supply for each port.

It has the ASM1184e switch chip on it, which _seems_ like a safe bet in terms of most card functionality—though there are a number of cards which seem to have issues on Pi OS through any kind of switch.

Anyways, I have plans... want to test a variety of setups with this, along with the new much smaller 230W PC power supply I bought so I don't have to lug the 700W Redragon around :D

paulwratt, 2021-05-17:
a comparison of Expansion Risers would be good. Especially where you got one working with anothers USB-A cable. You might try scouring ETA-Prime videos where he tests GPU's on x86 SBC (although some of them are going to be NVME expansion risers).

I hope this one works, it would be nice to see someone send in a photo of a it working in a case.

mahtin, 2024-02-21:
The info page https://pipci.jeffgeerling.com//cards_adapter/zs-zhishang-pcie-x1-to-4-pcie-x16-expansion-riser.html has a link to a dead product purchase page.

I did find this product on Amazon that looks the same. https://a.co/d/gFsZK0A

nejlontacsko, 2025-06-11:
Hi. Did you managed to make it work?
I checked with two NICs, the PCIe on my CM4 IO board is properly working, but if I connect the ASM1184e board with the USB riser, and applied PATA power, the `lspci` command doesn not show anything, but based on the TPU video of Jeff, I supposed to see at least the switch IC listed on it.
What i figured out, the ASM got the supply voltage from the 2nd PATA connector, but there is an unpopulated footprint (U7) on that line, which i believe supposed to be a regulator IC. Based on the photo which Jeff posted here, the U7 also not populated on his board.