geerlingguy/raspberry-pi-pcie-devices #126: Waveshare CM4 IO Base double sided SSD support

lukechilds, 2021-05-01:
Thanks for this absolutely brilliant resource @geerlingguy. Sorry if this is not the appropriate place to post this question. I tried to get in touch with the manufacturer but got no response.

I'm trying to find out if the [Waveshare CM4 IO Base](https://pipci.jeffgeerling.com/boards_cm/waveshare-cm4-io-base.html) supports a double sided SSD like in the image below.

[image]

If anyone has any knowledge of this I'd greatly appreciate some information. Thanks!

lukechilds, 2021-05-01:
My gut tells me the 2230 notch on the board will get in the way:

[image]

geerlingguy, 2021-05-01:
I would say likely not—too many things on the board there that would get in the way. A double-sided SSD needs a pretty much flat board underneath.

paulwratt, 2021-05-02:
an nvme slot extender cable.

or maybe forget the drive, and throw one of these in there instead:
https://www.youtube.com/watch?v=Su3WUNyYklE

(MOTILE 14 NVME M.2 to 16x PCIe adapter)

warmwaffles, 2021-05-02:
I _just_ ran into this issue today.

lukechilds, 2021-05-03:
Cheers guys.

>I just ran into this issue today. It does not fit the double sided nvme well. I may try to desolder the shorter screw hole to get it to flush mount.

Thanks for letting us know @warmwaffles. I'll close this issue as it seems like that's a definite no. But if you could update us with how removing the 2230 notch goes I'd really appreciate it!

closed 2021-05-03 by lukechilds

warmwaffles, 2021-05-03:
@lukechilds after looking at this a little longer, removing the 2230 notch would gain me a little bit of clearance but there are capacitors and stuff near there that I think will prevent me from having it sit perfectly flush. For now I've just resorted to using kapton tape to hold it down.

lukechilds, 2021-05-03:
Aww that sucks :/

Thanks for letting me know!

warmwaffles, 2021-05-03:
@lukechilds or @geerlingguy maybe add a note about double sided nvme use for the waveshare board?

geerlingguy, 2021-05-03:
@warmwaffles - I don't think it's necessarily important to note—double-sided NVMe are non-standard and (IIRC) out of spec, so there are many devices where they would not physically fit. That's why those SSDs often come with an extra warning (e.g. 'check for fit before purchase!').

warmwaffles, 2021-05-03:
I definitely did not see that warning / did not pay enough attention to it. 😛

piexil, 2021-05-03:
Double sided M.2s are in-spec 
https://en.wikipedia.org/wiki/M.2

Under form factors and keyring, S1-S5 for single sided thinkness and D1-D5 for double sided.

warmwaffles, 2021-05-03:
If the components were not underneath the m.2 slot, it would fit just fine. Although thinking about it now, cooling the memory module against the board may be difficult.

geerlingguy, 2021-05-04:
@piexil - Ah, well TIL. I will reopen the issue until I or someone else adds a PR to update the docs on that board in particular (maybe with a note like below):

> **Note**: This board's NVMe slot may not work with double-sided M.2 NVMe drives due to clearance issues.

I still think it's risky using double-sided cards unless you have the room for adequate cooling (or blast a fan over that area).

dans98, 2021-08-19:
An update for everyone.

I purchased the [Waveshare Mini Base Board (B)](https://www.waveshare.com/cm4-io-base-b.htm) and the Sabrent 512gb m.2 from amazon.

The double sided m.2 will not work, as the  2230 screw mount will get in the way. 👎 

Hopefully I can return the drive and find a decent sizes single sided drive.