geerlingguy/raspberry-pi-pcie-devices #129: Test KIOXIA CM6 NVMe and PM6 SAS SSD SFF drives

geerlingguy, 2021-05-04:
As the title says... here are some pretty pictures:

TODO: Add pretty pictures.

Good summaries of each drive can be found on anandtech:

  - [Kioxia Launches PM6: First 24G SAS SSD, up to 30.72 TB](https://www.anandtech.com/show/15855/kioxia-launches-pm6-first-24g-sas-ssd-up-to-3072-tb)
  - [Kioxia Releases First PCIe 4.0 SSDs: CD6 & CM6](https://www.anandtech.com/show/15537/kioxia-releases-first-pcie-40-ssds-cm6-cd6)

I have on loan 3x CM6-R KCM61RUL960G 960 GB 24G SAS Enterprise PCIe 4.0 NVMe drives, and 1x 1x PM6-R KPM61RUG960G 960 GB 24G SAS Enterprise SSD, and I will be testing them inside the SFF-TA-1001-compliant enclsure with the Tri-Mode MegaRAID card Broadcom sent and graciously helped me get going.

geerlingguy, 2021-05-07:
Set in JBOD mode:

```
$ sudo storcli64 /c0 set jbod=on
```

Formatted one PM6 and one CM6 as EXT4 then waited for lazy init to finish (took < 10 min):

```
sudo fdisk /dev/sdc  # follow prompts, and write with `w`
sudo mkfs.ext4 /dev/sdc1
sudo mkdir /mnt/pm6
sudo mount /dev/sdc1 /mnt/pm6
```

Ran benchmarks:

```
$ wget https://raw.githubusercontent.com/geerlingguy/raspberry-pi-dramble/master/setup/benchmarks/disk-benchmark.sh
$ chmod +x disk-benchmark.sh

$ sudo DEVICE_UNDER_TEST=/dev/sdc1 DEVICE_MOUNT_PATH=/mnt/pm6 ./disk-benchmark.sh
```

| Test | PM6 | CM6 |
| --- | --- | --- |
| fio seq read | 397MiB/s | 398MiB/s |
| iozone 1M rand read | 367MiB/s | 368MiB/s |
| iozone 1M rand write | 304MiB/s | 384MiB/s |
| iozone 4K rand read | 33MiB/s | 34MiB/s |
| iozone 4K rand write | 65MiB/s | 73MiB/s |

Test was run 3x with average taken in above table. The results were incredibly consistent, like ±.05%.

geerlingguy, 2021-05-07:
For comparison, in a PCIe Gen 4-capable system, ServeTheHome had [these results](https://www.servethehome.com/kioxia-cm6-review-pcie-gen4-ssds-for-the-data-center/2/):

128K sequential read: 6900MiB/s
128K sequential write: 4100MiB/s

So... just a _tad_ faster than the Pi. Like 17x faster 😂

One note, though—to get the maximum performance out of these drives, you need the latest high-end server architecture... even a couple-year old enterprise server that's still otherwise in great shape will not be able to eke out the performance these drives _can_ deliver.

geerlingguy, 2021-05-12:
CM6 has been added to the site. I will be talking about the PM6 in my next video, but I am not adding it to the site as it's not NVMe / not PCI-based storage.

paulwratt, 2021-05-14:
(thanks for documenting these, incl. PM6) Wendal of LevelOneTech would be proud (he uses enterprise KIOXIA SSD's)
:)

geerlingguy, 2021-05-14:
Video is up today: https://www.youtube.com/watch?v=jOU-NDDyyuY

And accompanying blog post: https://www.jeffgeerling.com/blog/2021/trying-kioxia-cm6-and-pm6-enterprise-ssds-on-raspberry-pi

closed 2021-05-14 by geerlingguy

geerlingguy, 2021-06-28:
I wanted to add some clarifications here, since it was pointed out to me that some of my graphs were slightly misleading (due to, basically, the marketing around speeds for different drives and interfaces changing a bit):

There was no SAS 6G or 12G — those were "6 Gb/s SAS" and "12 Gb/s SAS", respectively.

SAS 24G is technically "24G SAS" (note: not _24 Gb/s_, just 24G). That's because 24G SAS is actually 22.5 Gbps. And 48G SAS is actually 45 Gbps.

commit dbd83285 (2021-05-14): Fixes #129: Add KIOXIA CM6 video to CM6 page and MegaRAID page.