geerlingguy/raspberry-pi-pcie-devices #130: Add Seaberry Mini ITX Motherboard with tons of PCIe connectivity

geerlingguy, 2021-05-06:
The [Seaberry Pi CM4 Carrier Board](https://www.crowdsupply.com/alftel-systems/seaberry-pi-cm4-carrier-board) is a Mini ITX form factor motherboard for the Pi with tons of PCI Express slots (including M.2 M-key for NVMe, A/E-key for WiFi/Google Coral TPUs, Mini PCIe, x16 PCIe, x1 PCIe... need I go on? It also has a number of other interesting IO and features built-in to make it a very full-featured motherboard for a Pi-based PC!

[image]

closed 2021-05-06 by geerlingguy

sfxworks, 2021-12-02:
The super micro riser I got from a friend fits so n I c e l y oh man. This is gonna make a great router. 
[image]

geerlingguy, 2021-12-02:
@sfxworks - looks great! What's that HAT?

TheGuyDanish, 2021-12-02:
Seems to be the Adafruit Ultimate GPS HAT.

https://www.adafruit.com/product/2324

sfxworks, 2021-12-03:
Yep, that's the one @TheGuyDanish. I don't have an atomic pcie clock card like you @geerlingguy but I did have this lying around. I want to see about doing local ntp things using that.

sfxworks, 2021-12-06:
Regarding PoE, has anyone figured out how to get it to work? It says it's poe capable, and I have two PoE switches but they don't power on the board. I also have some LoveRPi poe hats but they don't fit on the board, nor do I think they would support 802.3bt. So I am a bit anxious as to whether or not I should take some cables to connect the pins to see what would happen. 

Also what size is the barrel? The ones I use for my quartz/rockpro64's don't work.

sfxworks, 2021-12-29:
Got it, though am I already pushing the board to its limit or is this somehow user error?
I have a 6 port tr-link with igp compiled, an adafruit GPS hat, an nvme, and an intel AX210. Last two with antennas. Sometimes all devices show up. At first I thought it was a fluke when my nvme was having trouble mounting on a restart, but then the AX210 started to freak and I just noticed the trlink didn't show up on this last restart. 

Two dmsg on a restart: https://gist.github.com/sfxworks/1c6cacafffddc7da6b49f68b4b00dca5
Last known pcie with trlink missing:
```
-[0000:00]---00.0-[01-11]--+-00.0-[02-11]--+-01.0-[03]--
                           |               +-02.0-[04]--
                           |               +-03.0-[05]--
                           |               +-04.0-[06]--
                           |               +-05.0-[07]--
                           |               +-06.0-[08]--
                           |               +-07.0-[09]----00.0  Silicon Motion, Inc. SM2263EN/SM2263XT SSD Controller
                           |               +-08.0-[0a]--
                           |               +-09.0-[0b]--
                           |               +-0a.0-[0c]--
                           |               +-0b.0-[0d]--
                           |               +-0c.0-[0e]--
                           |               +-0d.0-[0f]----00.0  Intel Corporation Device 2725
                           |               +-0e.0-[10]--
                           |               \-0f.0-[11]--
                           \-00.1  PLX Technology, Inc. PEX 8619 16-lane, 16-Port PCI Express Gen 2 (5.0 GT/s) Switch with DMA
```

geerlingguy, 2021-12-29:
@sfxworks - What kind of power supply are you using? I was using an 8A 12V supply in my testing and didn't seem to have any issues. I should note that maybe once every 10 or 20 boots, I would see the entire PCIe bus get 'link down' on a regular CM4 IO Board in some types of tests, but I didn't see that with the Seaberry in my own testing—only some NVMe issues when I had more than 3-4 drives loaded.

sfxworks, 2021-12-29:
Oh whoops, I that would be on me then. I only paid attention to the barrel size and threw it a 12v3a one. Should have known better, especially with the previous mentions of the poe++ capabilities 
https://www.amazon.com/gp/product/B08C513N9P/

geerlingguy, 2021-12-29:
@sfxworks - Yeah, that'll do it! 3A is barely enough for one or two low-power PCIe devices. I was using this 8A adapter: https://www.amazon.com/gp/product/B07HTFGM6R/

commit 8727f645 (2021-05-06): Fixes #130: Add Seaberry Mini ITX motherboard.