geerlingguy/raspberry-pi-pcie-devices #131: Idea for Nvidia GPU on CM4

iAmInActions, 2021-05-13:
**How I got this Idea**
I have used other arm-based machines before and most had problems with proprietary drivers under debian and arch.

**What I did to solve it**
I installed a copy of Ubuntu Server 20.04 aarch64 and installed gdm3 and lxde to get a grafical user interface. Then I installed software-properties-gtk, started it and went to the advanced driver section. There I installed all drivers that were not working before. Then I closed it, updated the initramfs and rebooted. Usually, it worked after that. Sometimes I had to compile a newer kernel to get it to work.

**Why can't I try it on my own pi?**
I don't own a CM4 or a modded pi 4 so it would be nice if someone who ones one could try this for me. All feedback is appreciated.

paulwratt, 2021-05-14:
sounds fesible (in theory), but some of the issue that have com up so far require tweeking source:
- **MSI Mode** compatibility is not 100%
- there are issues with "plain" 64bit-write-combined functions, and reworking is needed of read-combined functions
- most memio functions are broken, mostly due to the above
- as a result, RPi CM4 PCIe is not 110% out-of-the-box, more like 95-98% (97% ?) (at least for graphics cards)

there is lots of technical details in #4 (HD5450) when @Coreforge start his work.
outside of graphics cards, it seems to depend on how the drivers were written, as a lot seem to work with only a kernel option needing to be enabled (I can see a `kernel8-cm4.img` being a default in future releases at some point, with all useful aarch64 PCIe related kernel settings enabled - see the Intel 2.5Gb network card and the 16 drive NAS videos for references)

FreddyFeuerstein, 2021-05-15:
Maybe compiling buildroot for the pi but with the aarch64 server flags set could configure the bus correctly. It would be a ton of work bc you would have to cross-compile a small but still complex operating system. And seeing as it took me 2 whole days to get buildroot working on a usb drive to rickroll friends with it, it would probably take around a week to get it set up.

paulwratt, 2021-05-16:
there are some distros for RPi that use `buildroot`, there should be (by now) one for RPi4, if so, a delve into their `.config' would proove worthwhile.

For anyone going as far to get a (clean) `aarch64` buildroot for RPi4/3 you might also (or alternatively) try [mkroot](https://github.com/landley/mkroot) and include the work done from [qemu-rpi-kernel](https://github.com/paulwratt/qemu-rpi-kernel) to get everything working on host OS first (presuming there is RPi3/4 emu - see **native-emulation** directory)

If nothing else it will get a extremely small (memory footprint) environment running on RPi.

Cheers
Paul
PS. both of these are on my TODO list, as soon as I get stable power, and replace my failing 4G RPi4 (mHDMI port).

FreddyFeuerstein, 2021-05-17:
> [...]
> For anyone going as far to get a (clean) `aarch64` buildroot for RPi
> [...]
I already have a working config for buildroot with X11, twm and rpm. Would just have to recompile it for aarch64 with uefi, put the boot loader and the os on a usb drive and it should work.

FreddyFeuerstein, 2021-05-20:
Ill buy a cm4 once they are available again in my country. I want to try both methods.

geerlingguy, 2021-09-09:
Going to move this to a discussion, since it's a little more of a conversation and less actionable.

closed 2021-09-09 by geerlingguy