geerlingguy/raspberry-pi-pcie-devices #132: Idea: use UEFI to boot and initialize hardware

dns53, 2021-05-14:
To use some hardware such as graphics cards you most likely need to change the way you are booting your raspberry pi.
From what I can see you are compiling and booting the kernel directly.
What you most likely need to do is to load a uefi and have this initialize the pci connected hardware in to a known state.
You then can boot a linux kernel and talk to a video card and the video card firmware is ready to talk back.

From what I can see from your video's you are booting directly to a linux kernel so the process is:
uboot (chip) > uboot (sd card) > linux kernel

The proposed boot would be:
uboot(chip) > uefi firmware > linux kernel (compiled as an uefi binary)

Take a look at the efi project for raspberry pi 4, the change log suggests there is some initial support for compute module.
https://github.com/pftf/RPi4/tree/v1.26

Trying this out would involve using a sd card with a fat 32 partition and thair firmware images and a linux kernel compiled with efi support.
You should be able to boot a generic server installer such as debian, suse, ubuntu built for aarch64 instead of an os build specifically for the raspberry pi.

paulwratt, 2021-05-14:
the WOR windows 10 arm on RPi4 uses an UEFI bootloader, as does the VMware ESXi-Arm for RPi4.
if someone is interested in persuing UEFI, it maybe worth checking out the following, a demonstration of flashing the AMD aarch64 UEFI bios to a Saphire RX270 (UEFI enabled) PCIe graphics card:
https://www.youtube.com/watch?v=3cYJdw3Y9_Y

Arjdroid, 2021-05-17:
This is a good idea, in addition to UEFI, using aarch64 that will eliminate the other driver problems as well. UEFI and 64-Bit are the two things that I currently see to be missing from the tests Jeff has been doing and I think those two would help make the card work...

paulwratt, 2021-05-17:
UEFI stems from x86, and _all_ cards are sold with _x86 BIOS_, weather that are UEFI enable cards or not. The reason `uboot` is not "useful" is because (like most SBC linux kernels) it uses **devicetree**, mapping UEFI to the device tree is "fraught with challanges".

mi-hol, 2022-01-03:
> hardware raid ... had to configure the disks on a x86 board

@rtreffer did the RAID controller you tried have a UEFI BIOS? 
I'm currently investigating the benefits of "ARM System Ready" certification and found so far only one SATA HW RAID controller that mentions a UEFI BIOS, but have not had access to test it yet.
I've started a conversation on this topic on openmediavault forum [here](https://forum.openmediavault.org/index.php?thread/41409-anyone-tried-sbc-with-system-ready-uefi-firmware-and-sata-hw-controller-with-uef/)

rtreffer, 2022-01-03:
@mi-hol ok, from the top of my head.....

- You can flash an UEFI option rom to the raid controller (see e.g. [this post](https://forums.serverbuilds.net/t/guide-updating-your-lsi-sas-controller-with-a-uefi-motherboard/131) - you will most likely need a x86 machine for this) - old LSI SAS cards are the gold standard for home re-use.
- This will provide the UEFI with a driver to use the RAID as a boot device - on x86!
- You will need an UEFI compiled with `DEFINE X64EMU_ENABLE           = TRUE` (add it to the [rpi dsc file](https://github.com/tianocore/edk2-platforms/blob/master/Platform/RaspberryPi/RPi4/RPi4.dsc))
- This will add a Qemu to your UEFI. Loading these drivers will be slow, borderline slow, but it will work
- You will need a x86 machine with the controller and all disks to initialize the RAID. I didn't get the classic POST to run
- Plug the flashed + configured RAID controller + disks into an CM4 carrier, boot with UEFI and choose the RAID boot device.

I would call this an academic exercise as using a x86 for flashing and raid setup is cheating at best. The only advantage I can see is support for backup batteries. Everything else should be betterin IT mode with a modern filesystem (read: btrfs, zfs).

mi-hol, 2022-01-03:
> using a x86 for flashing and raid setup is cheating at best.

I fully support this view, hence the question is: will using UEFI on aach64 avoid the need for x86 to configure RAID setup?

rtreffer, 2022-01-03:
@mi-hol no, because that tool runs as part of the POST, not as a UEFI driver (unless I am mixing something up). AFAIK some Dell machines have full UEFI support, but that's very custom. I couldn't get the common raid tools to run on aarch64 either (no megacli).

mi-hol, 2022-01-27:
> will using UEFI on aarch64 avoid the need for x86 to configure RAID setup?

Still searching for answers and [got links](https://github.com/pftf/RPi4/issues/203) to these blog post:
[Revolutionizing ARM Technology: x86_64 Option ROM on AArch64](https://www.suse.com/c/revolutionizing-arm-technology-x86_64-option-rom-aarch64/)
[AArch64 option ROMs for AMD GPUs](https://www.workofard.com/2020/12/aarch64-option-roms-for-amd-gpus/)

Preliminary conclusion from reading is: it should be feasible.
Next step is to get my hands on hardware with a UEFI-BIOS to test it.
My [Broadcom / LSI SAS2008 card ](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/357) doesn't seem to have that :(

mi-hol, 2022-03-08:
finally I had some progress with compiling RPi4 UEFI-BIOS (see https://github.com/pftf/RPi4/issues/209 for the detailed instructions). Next goal is to enable x86 emulation in this RPi4 UEFI-BIOS

mi-hol, 2022-03-08:
> * You will need an UEFI compiled with `DEFINE X64EMU_ENABLE           = TRUE`

build with X64EMU_ENABLE was successful. Next step boot a CM4 with it.

gCommandLineDefines: {'X64EMU_ENABLE': 'TRUE', 'NETWORK_ISCSI_ENABLE': 'TRUE', 'SMC_PCI_SUPPORT': '1', 'ARCH': 'AARCH64'}
BuildOptionPcd: [('gEfiMdeModulePkgTokenSpaceGuid', 'PcdFirmwareVersionString', '', 'L"RPi4 UEFI 1.33 Test Firmware (mi-hol)"', ('build command options', 1))]