geerlingguy/raspberry-pi-pcie-devices #136: PIzza4 Micro Server (CM4 Edge Computer)

geerlingguy, 2021-05-18:
I just noticed the PIzza4 from a post on Twitter: https://www.openembed.com/products-list/13

[image]

[image]

[image]

[image]

[image]

[image]

It looks like it has:

  1. Dual Gigabit Ethernet (looks like it uses a Realtek PCIe chip for the 2nd slot)
  2. Built-in 3-way PCIe Switch ([PI7C9X2G404SV](https://ar.mouser.com/datasheet/2/115/PI7C9X2G404SV-Product-Brief-1536988.pdf))
  2. Dual full-size HDMI
  3. 1x M.2 M Key slot (full size)
  4. 1x M.2 A+E Key slot
  5. 2x USB (2.0? 3.0?)
  6. Custom designed enclosure
  7. USB Type-C for Power? OTG?

This is... a rather full-featured tiny board! I like it.

closed 2021-07-02 by geerlingguy

YumingChang02, 2021-07-30:
I've received my unit!!
5. ~1x M.2 A+E Key slot~ the slot is a miniPCIe port (with PCIe gen2x1 and USB2)
6. 2x USB (2.0? 3.0?), its USB2 (seems to be powered off Microchip USB2514B, located between second Ethernet port and the stacked USB ports)
8. USB Type-C can both provide power and flash EMMC (haven't tested OTG yet), To flash onboard EMMC, there is a jumper around where the power button is on the PCB (the yellow jumper is also included in the box from seeed studio)
[image]
I got my unit with a black cover, which seems to be aluminum heatsink

* Issues found now
    * using openwrt with custom configurations [config.buildinfo](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/6906812/config.buildinfo.txt)
    * routing speed is around 8xx Mbps, however direct connection from PIzza to pfsense can go up to 9xx Mbps
        * PIzza <--[ switch ]--> pfsense : Wireguard speed can reach around 700Mbps via iperf3 single side
    * the MAC address of realtek 8111H seems odd ( showing 00:00:00:00:00:05 )

mthird, 2021-10-29:
I finally got the 2nd NIC to work by updating to the latest nightly firmware using: `sudo rpi-update`!

mthird, 2021-10-29:
Also, now that I realized there is no USB3 chip on board, putting the LTE modem into USB2 mode solved my remaining problem.

AllizomFoxfire, 2021-10-31:
@mthird Well, I have two of these units on my desk now after thinking one was faulty. I had them send me another in exchange as I could only get it into RPI boot once. Immediately following that one instance a package arrived in which I had bought some new USB-C super-speed cables which absolutely do NOT work as OTG cables. I had no idea this was the case. It took me days to figure this out by pure trial and error.

Now I have properly changed the boot order to 0xf34516 to reflect the NVME  > emmc > USB storage > RPIBOOT failsafe > loop. The 4 and 5 are really interchangeable as there are no USB3 ports, and as you said, that is the one major shortcoming of this device as the bus exists since the second gigabit ethernet sits on it.

Otherwise, this is a fantastic little device that ticks every other box on the list of what I could want in a micro server.

duckgogo, 2022-03-10:
@AllizomFoxfire I think Pizza 4 cannot boot from NVME, as it uses PCIe switch

abin444, 2025-01-10:
Hey does anyone has the schematics for this board? Almost all details about this board seems to  have vanished. I'm more interested in the Pericom PCIe switch and the I210 ethernet controller implementation as I use this combinations for a CM5 based board. The intel IPS a/c creation has been dragging me for a few days.

commit faccdead (2021-07-02): Fixes #136: Add Pizza4 Micro Server by OpenEmbed.