geerlingguy/raspberry-pi-pcie-devices #142: Add new board

NitinJSanket, 2021-05-22:
https://grabcad.com/library/tpu-cam-with-cm4-1?utm_source=tr.im&utm_medium=no_referer&utm_campaign=tr.im%2Ftpucam&utm_content=direct_input

This is an amazing board.

geerlingguy, 2021-05-24:
Wow, very neat board! Some more background—here are a couple pictures:

[image]
[image]

From the description:

> Combination of Coral Edge TPU, Raspberry Pi Compute Module 4 and imx219 camera sensors.

In the [YouTube video about the board](https://www.youtube.com/watch?v=6aQB4FLPs9k), Jakub mentioned Coral was working through a 'PCI Express bus', but I wonder if he's actually got it going through USB 3.0 to PCIe or if he found a way to get it working directly?

See: https://github.com/google-coral/edgetpu/issues/280

geerlingguy, 2021-07-02:
And here's the project page: https://grabcad.com/library/tpu-cam-with-cm4-1

commit 631fe965 (2021-07-02): Fixes #142: Add Jakub Gal's TPU cam.
closed 2021-07-02 by geerlingguy