geerlingguy/raspberry-pi-pcie-devices #149: cannot find cross-compile

mschemerii, 2021-05-31:
Just tried this out and got (on Catalina 10.15.7):

[mschemerii@MacBook-Pro:~]$ docker build -t cross-compile .
[+] Building 0.1s (1/2)                                                         
 => ERROR [internal] load build definition from Dockerfile                 0.0s
 => => transferring dockerfile: 40B                                        0.0s
------
 > [internal] load build definition from Dockerfile:
------
failed to solve with frontend dockerfile.v0: failed to read dockerfile: error from sender: open .Trash: operation not permitted

geerlingguy, 2021-05-31:
@mschemerii - You need to make sure you're inside the `extras/cross-compile` directory of this repository. From your prompt (`~`), it looks like you're actually in your home directory.

mschemerii, 2021-05-31:
So I need to download the whole cross-compile repository? Sorry. Still a n00b

geerlingguy, 2021-05-31:
@mschemerii - Yes; you'll need to either download or `git clone` this project to your computer somewhere, then in the Terminal go into this folder, then into `extras/cross-compile`, then run the commands there :)

mschemerii, 2021-06-01:
@geerlingguy worked like a charm. I was just downloading the single dockerfile.

geerlingguy, 2021-06-01:
Great!

closed 2021-06-01 by geerlingguy