geerlingguy/raspberry-pi-pcie-devices #150: Benchmark Pi kernel compilation on M1 Mac mini and MacBook Air

geerlingguy, 2021-06-01:
Just for funsies... I'm trading in my Intel i9 2.4 GHz 2019 16" MacBook Pro for both an M1 Mac mini (with 10Gbps and 16 GB RAM) and an M1 MacBook Air (with 16 GB of RAM) — notably, those two computers cost the same as the amount I can sell the 2019 MBP for _used_!

The 2019 MBP will do about 12.5 minutes cross-compile using the following command on the 5.10.x kernel:

```
time make -j10 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
```

I want to see how the Air and mini perform for the same exact command (under qemu) and see what kind of things I can do to optimize build times since the M1 is native ARM/`aarch64`...

geerlingguy, 2021-06-01:
| Scenario | Time (hh:mm:ss) |
| --- | --- |
| 2019 i9 MacBook Pro - cross-compile | 00:12:30 |
| 2020 M1 MacBook Air - cross-compile | 00:10:23 |
| 2020 M1 Mac mini - cross-compile | TODO |

geerlingguy, 2021-06-01:
Also wondering if `-j10` might be too much for the optimum mix on the M1 (with it's 4 'performance' + 4 'efficiency' cores).

geerlingguy, 2021-06-01:
Seems like `-j8` might be optimal on the M1.

geerlingguy, 2021-06-01:
With `-j8` I'm getting 8:58 on the M1 mini, so yeah, looks like that'll be my new recommendation.

geerlingguy, 2021-06-01:
[image]

closed 2021-06-01 by geerlingguy

geerlingguy, 2021-06-01:
Blog post about it: [The Apple M1 compiles Linux 30% faster than my Intel i9](https://www.jeffgeerling.com/blog/2021/apple-m1-compiles-linux-30-faster-my-intel-i9).

mwpastore, 2021-11-13:
M1 Pro (10-core) 👀 

```
ubuntu@preferable-planthopper:~/linux$ time make -j8 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs
.
.
.
real	6m1.945s
user	41m10.009s
sys	5m36.866s
```

I used [multipass](https://multipass.run/) to create a quick Impish VM with eight cores. It appears to be using the P-cores only. It wouldn't let me create a VM with ten cores.

geerlingguy, 2021-11-13:
Dang... can you also try with maybe -j10 or -j12 and see if you still get a boost? Usually you can go a little beyond the strict core count, but it depends on the architecture.

Now I want to see how long it takes on some M1 Max Macs (that'll never get old).

mwpastore, 2021-11-13:
`-j10`:

```
real	5m15.496s
user	36m10.306s
sys	4m47.460s
```

`-j12`:

```
real	5m27.698s
user	37m44.642s
sys	4m53.749s
```

So yeah, 5-ish minutes. Nice.

> Now I want to see how long it takes on some M1 Max Macs (that'll never get old).

It might be a bit faster due to the increased memory bandwidth (and capacity) but probably not that much faster.

geerlingguy, 2021-11-13:
True, CPU cores are the same.

ericcurtin, 2021-11-18:
I always go with -j number of cores even on x86. Even if it's faster with more parallelism it's nice to spare a few CPU cycles for other stuff

ericcurtin, 2021-11-18:
Would someone mind running a quick test for me building with multipass vs docker vs podman on the M1? I might be buying a Mac mini in the next few days for work. I want to see what the performance is like between these tools. The slowness of my raspberry pi is becoming a bottleneck for me?

mwpastore, 2021-11-18:
> multipass vs docker vs podman on the M1

Docker on macOS runs in a VM, so don't expect too much performance difference between these options.

paulwratt, 2021-11-20:
> The slowness of my raspberry pi is becoming a bottleneck for me

get more Raspberry Pi's

commit 9ead5312 (2021-06-01): Fixes #150: Optimized for my new M1 MacBook Air and mini.