geerlingguy/raspberry-pi-pcie-devices #151: CM4-BoxPro - CM4 I/O with NVMe housed in an industrial enclosure

ALANLOW-LB, 2021-06-01:
The CM4-BoxPro is an I/O carrier board with an NVMe slot, passive cooling and made to be housed in one of our standard enclosures. The PCB has passive cooling and we have tested the enclosure with the processor running at 100% with the internal temperature of the enclosure being measured at 64°C. The attached graph shows the results of 2 tests – the board without any heatsinking fully enclosed and the board with the passive heating fully enclosed.
We have designed this board with the intension of it being housed in an enclosure. As you are aware most SBC have connectors on all sides which make it very difficult to house. Our board has connectors on the front and rear only to negate this issue.
The enclosure has many mounting options including, surface, DIN Rail, VESA Mount and 19” rack mount. The Board will also be available to fit in a Eurorack case.
[image]
[image]
[CM4-BoxPro.pdf](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/6575586/CM4-BoxPro.pdf)

We believe that there is a sector of businesses that want to develop a new product but due to the either the small product run or budget, they are not in a position to develop a new board and housing. We can customise the enclosure in terms of colour and artwork to match corporate branding.

geerlingguy, 2021-06-01:
Very nice looking minimal enclosure! Do you have any link for this or any pricing available?

ALANLOW-LB, 2021-06-02:
Hi Jeff

Thank you for your reply.
I am afraid that you have been bombarded by us the last few days.

My MD Martyn sent you an email yesterday

We have a holding page on our website for this product while it is in development
https://lincolnbinns.com/shop/internet-of-things-iot/compute-module-4-enclosures.html

I will have the pricing together for the various options tomorrow and will send it over to you

Regards
Alan Low
Sales Manager
***@***.***D757C2.2D174290]<http://www.lincolnbinns.com/>
Tel: 01403 865486<tel:01403865486>
E-mail: ***@***.******@***.***>
Web: www.lincolnbinns.com<http://www.lincolnbinns.com/>
Zoom: ***@***.***<https://us04web.zoom.us/j/7852335780?pwd=Wkt6aFZ6SlVodEdoejduOGlLVjY4UT09>
***@***.******@***.******@***.******@***.******@***.***D757C2.2D174290]<https://us04web.zoom.us/j/7852335780?pwd=Wkt6aFZ6SlVodEdoejduOGlLVjY4UT09>
Lincoln Binns Ltd, Oakendene Estate, Cowfold, West Sussex, RH13 8AZ
Company Registration Number - 1904817 | VAT Number - GB 423 2273 85

stephenvalente, 2021-06-02:
Looks interesting, and being as you’re a UK supplier, I wouldn’t get hit for a load of import taxes and extortionate shipping costs with this product 😀

ALANLOW-LB, 2021-06-04:
We should have our first production batch of boards here in 5-6 weeks.

Ultimately we are looking to release various models based around the enclosure.
A Desktop version which will be housed in a smooth, curved edged clam shell enclosure
An Industrial version with many mounting options including surface, DIN Rail, VESA mount and 19" rack
The 19" rack will hold up to 3 units.
We are currently designing a Eurorack version

The industrial version can be fully customised with different coloured anodised case, company logos and face plate markings.

We are aiming for a starting price of £90-95 for the industrial unit

https://lincolnbinns.com/shop/internet-of-things-iot/compute-module-4-enclosures.html

ALANLOW-LB, 2021-06-04:
> Looks interesting, and being as you’re a UK supplier, I wouldn’t get hit for a load of import taxes and extortionate shipping costs with this product 😀
> […](#)
> Sent from my iPhone

Hi Stephen

As you say, we are a UK based manufacturer so carriage will be reasonable.

geerlingguy, 2021-07-02:
This is great! Please feel free to keep me in the loop if you have a unit you'd like me to test out; but for now, I've added the board to the Pi PCIe site and am closing this tracking issue: https://pipci.jeffgeerling.com/boards_cm/lincoln-binns-cm4box-pro.html

commit 0fe0047f (2021-07-02): Fixes #151: Add Lincoln Binns CM4-BoxPro to the site.
closed 2021-07-02 by geerlingguy