geerlingguy/raspberry-pi-pcie-devices #152: Lite Berry

TroopaShells, 2021-06-13:
Lite berry is a cm4 board to turn a 3rd party  Lite housing  into a retro handheld. Currently just a kickstarter.  I have been working on this for some time recently got a 3rd party to design and test pcb. 
Will be making a github for it soon.

https://www.kickstarter.com/projects/troopashells/lite-berry/posts.

TroopaShells, 2021-06-13:
[image]

geerlingguy, 2021-07-02:
@TroopaShells - I would be happy to put up something on the site once there's more content including a PCB design rendering or full prototype—right now there's not enough to go on to give people the information they'd need to learn from the build or much assurance the full product will exist once the Kickstarter is funded.

Please feel free to reopen or open a new issue once you have more information or renders to add.

closed 2021-07-02 by geerlingguy

TroopaShells, 2021-07-08:
Alright . I understand completely.