geerlingguy/raspberry-pi-pcie-devices #157: Test Apple 12+16 mSATA SSD + no name cheapo adapter

htruong, 2021-06-26:
- Start from the [Apple Samsung 12+16 mSATA SSD](https://beetstech.com/blog/apple-proprietary-ssd-ultimate-guide-to-specs-and-upgrades#hdr-6) (extracted from 2015 Macbook Pro)
- Put the SSD on a [no-name adapter](https://www.ebay.com/itm/263200771877).
- Put the whole thing on another no-name M.2 PCI-e to PCI-e 3.0 x4 adapter ($14 from brick and mortar Central Computer).
- Saw the PCI-e x1 socket on the mainboard to make the card fit. Note: This slot is pretty delicate, I don't recommend using a Dremel to saw it all the way through, better use an x-acto knife. I almost messed mine up.

`dmesg`:
```
[    5.527894] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[    5.528387] ata1.00: ATA-8: APPLE SSD SM0128G, BXZ13A0Q, max UDMA/133
[    5.528409] ata1.00: 236978176 sectors, multi 16: LBA48 NCQ (depth 32), AA
[    5.529012] ata1.00: configured for UDMA/133
[    5.529614] scsi 0:0:0:0: Direct-Access     ATA      APPLE SSD SM0128 3A0Q PQ: 0 ANSI: 5
[    5.531879] sd 0:0:0:0: [sda] 236978176 512-byte logical blocks: (121 GB/113 GiB)
[    5.531952] sd 0:0:0:0: [sda] 4096-byte physical blocks
[    5.532147] sd 0:0:0:0: [sda] Write Protect is off
[    5.532167] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    5.532472] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    5.583275]  sda: sda1 sda2
[    5.586143] sd 0:0:0:0: [sda] Attached SCSI disk
```

`lsblk`:
```
lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0   113G  0 disk
|-sda1        8:1    0   200M  0 part
`-sda2        8:2    0 112.8G  0 part
```

`hdparm -tT /dev/sda`:
```
/dev/sda:
 Timing cached reads:   1568 MB in  2.00 seconds = 784.21 MB/sec
 Timing buffered disk reads: 882 MB in  3.00 seconds = 293.92 MB/sec
```

geerlingguy, 2021-09-09:
Added!

closed 2021-09-09 by geerlingguy

cTurtle98, 2021-12-02:
the adapter linked to on amazon goes the wrong way. the one linked to is an adapter for installing a normal m.2 drive into a mac instead of a mac drive