geerlingguy/raspberry-pi-pcie-devices #161: Figure out easier way to build single kernel modules

geerlingguy, 2021-06-28:
@lhartmann suggested trying to build modules without a full kernel recompile: https://twitter.com/lcsvh/status/1403147011491311617

That links over to [this comment](https://github.com/MichaIng/DietPi/issues/3435#issuecomment-603537815), with a script to build a touchscreen driver:

```sh
#! /bin/sh -e
version="$1"
TMPDIR="/tmp/silead-build-$version-$$"
SRCDIR="/lib/modules/$version/build"
OUTDIR="/lib/modules/$version/kernel/drivers/input/touchscreen/"
URL="https://raw.githubusercontent.com/torvalds/linux/master/drivers/input/touchscreen/silead.c"
SUM="698396a0b070a93b16536e01fd7dc3c4d9bdfeda632822867b65d068b07cbd49"

# Sanity check:
[ -d "$SRCDIR" ] || exit 0
command -v gcc  >/dev/null 2>/dev/null || exit 0
command -v make >/dev/null 2>/dev/null || exit 0

# Create a temporary build directory (about 1MB)
mkdir "$TMPDIR"

# Get module source and validate
wget -q "$URL" -O "$TMPDIR/silead.c"
echo "$SUM  $TMPDIR/silead.c" | sha256sum -c --quiet

# Build
echo "obj-m += silead.o" > "$TMPDIR/Makefile"
make -C "$SRCDIR" M="$TMPDIR" modules

# Install
mkdir -p "$OUTDIR"
cp "$TMPDIR/silead.ko" "$OUTDIR"

# Clean
rm -rf "$TMPDIR"
```

It would be nice if I could build some drivers (even with patches hosted separately) without requiring a full kernel recompile. The script could even be hosted in this repo so people could use a curl one-liner.

samveen, 2021-06-29:
DKMS (Dynamic Kernel Module Support) allows for out -of-tree modules to be compiled easily against the installed and vendor packaged kernels if the kernel's associated `kernel-headers` package is also installed (DKMS has been around and part of most distributions for a long time). 

Most driver packages usually come packaged for DKMS (and/or use it internally). It should be available on raspberry Pi OS /raspbian too (it is available fordebian on pi and ubuntu for Pi, I use ubuntu 20.04.2)

6by9, 2021-06-29:
If you're just changing part of the kernel, then you can do ```make drivers/media``` or whichever subsystem you're rebuilding.

The niggle is ensuring that all the required dependencies are found, eg if I try ```make drivers/media/i2c/imx219.ko``` I get errors of
```
ERROR: modpost: "v4l2_event_subdev_unsubscribe" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_ctrl_subdev_subscribe_event" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "__v4l2_ctrl_s_ctrl" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "__v4l2_find_nearest_size" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "__v4l2_ctrl_handler_setup" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "__v4l2_ctrl_grab" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_async_register_subdev_sensor_common" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "media_entity_pads_init" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_ctrl_new_fwnode_properties" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_fwnode_device_parse" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_ctrl_new_std_menu_items" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_ctrl_new_std" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_ctrl_handler_init_class" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_fwnode_endpoint_free" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_fwnode_endpoint_alloc_parse" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_i2c_subdev_init" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_ctrl_handler_free" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "v4l2_async_unregister_subdev" [drivers/media/i2c/imx219.ko] undefined!
ERROR: modpost: "__v4l2_ctrl_modify_range" [drivers/media/i2c/imx219.ko] undefined!
```
as those are all provided by supporting modules.

If you look at the media tree for Linux then they provide the media_build system to rebuild all of the media tree and replace it in your live system. https://www.linuxtv.org/wiki/index.php/How_to_Obtain,_Build_and_Install_V4L-DVB_Device_Drivers

DKMS is an option, but is another layer of stuff to deal with.

Generally if you have a compiled kernel tree, then adding one or two modules is less than a 1 minute recompile, so I'm a little confused as to why you're having to rebuild everything. Yes if you do a significant rebase to your upstream tree then it recompiles nearly everything, but that's necessary.

geerlingguy, 2021-06-29:
@6by9 - For myself, I don't care much, since I have the recompile down to about 30s or less including the copy to the Pi. But for others who want to replicate my tests for a specific device, and have never compiled the kernel before, if I can get them a faster mechanism by which to compile one module, that's the goal.