geerlingguy/raspberry-pi-pcie-devices #163: uCM4 (Minimal CM4 carrier board)

just-grizzle, 2021-06-28:
I stumbled upon the uCM4 carrier board on [Hackaday](https://hackaday.io/project/179793-ucm4)  recently. It is intended as the "smallest possible headless server with help of Compute Module 4 unit."

[image]

Features:
- exactly same size as CM4
- support for both Lite and eMMC variants of CM4
- Gigabit Ethernet
- microSD slot (optional)
- Micro USB for power and USB 2.0 data transfer
- USB-OTG support

See it in action [here](https://www.youtube.com/watch?v=mftScuGE_xw) .

geerlingguy, 2021-06-28:
Very neat! I also liked your little micro USB connector flip about 20 seconds in on the video. Should definitely get this added to the site. Do you have any open source design files for this, or is it planned to be sold anywhere or offered as a kit?

just-grizzle, 2021-06-29:
Hey Jeff, I'm not the inventor of this board. Just came across it and thought it could be nice on your list.

But I found his [repo on Github](https://github.com/v3l0c1r4pt0r/ucm4). Can't watch the ZIP on mobile, but the name indicates there are pcb files inside.

commit 8b8e0c5a (2021-07-02): Fixes #163: Add uCM4 tiny network carrier board.
closed 2021-07-02 by geerlingguy