geerlingguy/raspberry-pi-pcie-devices #164: Add Hailo-8™ M.2 AI Acceleration Module

RezaSaadat1, 2021-07-02:
Hi,
there is another AI Accelerator out there, please check the compatibility in raspberry pi pcie devices
https://hailo.ai/product-hailo/hailo-8-m2-module/

geerlingguy, 2021-07-02:
@RezaSaadat1 - Do you know if there are any Linux drivers for this board? Any guides for it? I didn't do too much digging, but this is the first I've heard of it.

More11o, 2021-12-05:
So I've been looking this up since seeing this and it seems most of the information is behind a 'business enquiry' wall.

I did find a link to a data sheet from March 2021 that seems to suggest that the Linux driver is pre-complied and only for 4.15 and 5.0 kernels which makes me think its proprietary. It does list it as being compatible with  ARM64 though.

[Link](https://hailo.ai/wp-content/uploads/2021/04/Hailo-8-M.2-key-AE-Data-Sheet-_01_APR_2021.pdf)

geerlingguy, 2021-12-05:
I've sent in an email to Hailo, to see about getting one to test. Right now it _seems_ they mostly sell to integrators directly, it's not as much a direct-to-consumer product available in small quantities. But I'm asking about that and future plans, since supposedly it _should_ work with the CM4.

pritamghanghas, 2023-09-28:
I can see that the product is available from up-board.org as retail units and they claim that it works with their boards. But again those are intel atom boards that can feed it. I doubt Pi will even be able to feed the data to it considering only 1 lane pcie that too is shared with usb3 devices I think.

geerlingguy, 2023-09-28:
I may be able to get a device to test! Finally got in touch, we'll see.

cyclux, 2024-01-31:
@geerlingguy Any update on this topic? I am also very interested in the compatibility of the Pi5 with those Hailo-8 modules. It is hard to find any references on the internet.

cyclux, 2024-01-31:
Nice! Looking forward to your test results 👍

geerlingguy, 2024-02-06:
This is on the site now, at least: https://pipci.jeffgeerling.com/cards_m2/hailo-8-ai-module.html

And one other report from a user in the wild: https://twitter.com/jonlech/status/1754921750083596664

cyclux, 2024-02-06:
Funny, got the notification while watching a video of yours :)
I am very surprised there is no test report from anybody yet, although those Hailo chips have been out there since 2021. Maybe that is because of the relatively late adoption of M.2 with SBCs..
Soon I'll get my "Pineberry HAT AI". If there is success in running the M.2 Hailo, will try to order one as well using your link on [https://pipci.jeffgeerling.com](https://pipci.jeffgeerling.com/cards_m2/hailo-8-ai-module.html). BTW: Are you sure it's a working referral link? Cannot see any URL params, or will they recognize it via HTTP referer?

geerlingguy, 2024-02-06:
@cyclux - Not all of the links are referral links, usually only ones to Amazon product listings. I get a small amount of revenue from referral links but I mostly want to just share the neat things I find, referral or not!

cyclux, 2024-02-06:
Ah, I see, so `If you'd like to purchase this card, it helps me out if you use the following product link:` is a template sentence, and the link is not guaranteed to be an actual referral. Too bad, just wanted to support you, will find another way, sooner or later!

PS: Are you aware of any other "affordable" M.2 NPUs / TPUs / AI Accelerators that are known to be compatible with Pi / Pi OS?

geerlingguy, 2024-02-06:
@cyclux - It looks like Arducam has their PiNSIGHT setup: https://www.arducam.com/arducam-introduces-pinsight-your-vision-ai-mate-for-raspberry-pi-5/ (I have one to test, haven't gotten to test it yet) based on Intel Movidius Myriad X (4 TOPS). There's also Sony's AITRIOS lineup, but I haven't heard anything about that recently.

cyclux, 2024-02-06:
Thanks. I also stumbled upon Myriad X, but have not found a dedicated M.2 module yet (with reported support for Pi). Just some mysterious ones like https://www.aaeon.com/en/p/ai-edge-computing-board-ai-core-xm-2280 . Hopefully, with the Pi M.2 HAT releases, AI modules will be more common this year.

JonLech, 2024-03-14:
The Hailo-8 M.2 [works fine](https://twitter.com/jonlech/status/1768413755644219495) with the Pi5 (and pciex1_gen=3) using the Pineberry Pi HatDrive Bottom.

[image]

```
# dmesg | grep hailo
[    5.169204] hailo: Init module. driver version 4.16.2
[    5.186413] hailo 0000:01:00.0: Probing on: 1e60:2864...
[    5.186424] hailo 0000:01:00.0: Probing: Allocate memory for device extension, 11592
[    5.186444] hailo 0000:01:00.0: enabling device (0000 -> 0002)
[    5.186451] hailo 0000:01:00.0: Probing: Device enabled
[    5.186471] hailo 0000:01:00.0: Probing: mapped bar 0 - 00000000128eb0c1 16384
[    5.186477] hailo 0000:01:00.0: Probing: mapped bar 2 - 000000003516a586 4096
[    5.186482] hailo 0000:01:00.0: Probing: mapped bar 4 - 00000000cf4c1ecb 16384
[    5.186487] hailo 0000:01:00.0: Probing: Setting max_desc_page_size to 16384, (page_size=16384)
[    5.186494] hailo 0000:01:00.0: Probing: Enabled 64 bit dma
[    5.186497] hailo 0000:01:00.0: Probing: Using userspace allocated vdma buffers
[    5.186501] hailo 0000:01:00.0: Disabling ASPM L0s 
[    5.186505] hailo 0000:01:00.0: Successfully disabled ASPM L0s 
[    5.384172] hailo 0000:01:00.0: Firmware was loaded successfully
[    5.397205] hailo 0000:01:00.0: Probing: Added board 1e60-2864, /dev/hailo0
```

Before building _libhailort_ you'll need to change MAX_DESC_PAGE_SIZE to 16384 in [descriptor_list.hpp](https://github.com/hailo-ai/hailort/blob/master/hailort/libhailort/src/vdma/memory/descriptor_list.hpp#L46).

```
# hailortcli parse-hef yolov8n.hef 
Architecture HEF was compiled for: HAILO8
Network group name: yolov8n, Multi Context - Number of contexts: 2
    Network name: yolov8n/yolov8n
        VStream infos:
            Input  yolov8n/input_layer1 UINT8, NHWC(640x640x3)
            Output yolov8n/conv41 UINT8, FCR(80x80x64)
            Output yolov8n/conv42 UINT8, FCR(80x80x80)
            Output yolov8n/conv52 UINT8, FCR(40x40x64)
            Output yolov8n/conv53 UINT8, FCR(40x40x80)
            Output yolov8n/conv62 UINT8, FCR(20x20x64)
            Output yolov8n/conv63 UINT8, FCR(20x20x80)

# hailortcli benchmark yolov8n.hef 
Starting Measurements...
Measuring FPS in hw_only mode
Network yolov8n/yolov8n: 100% | 1710 | FPS: 113.99 | ETA: 00:00:00
Measuring FPS and Power in streaming mode
[HailoRT] [warning] Using the overcurrent protection dvm for power measurement will disable the overcurrent protection.
If only taking one measurement, the protection will resume automatically.
If doing continuous measurement, to enable overcurrent protection again you have to stop the power measurement on this dvm.
Network yolov8n/yolov8n: 100% | 1710 | FPS: 113.95 | ETA: 00:00:00
Measuring HW Latency
Network yolov8n/yolov8n: 100% | 1707 | HW Latency: 7.97 ms | ETA: 00:00:00

=======
Summary
=======
FPS     (hw_only)                 = 113.988
        (streaming)               = 113.955
Latency (hw)                      = 7.97256 ms
Device 0000:01:00.0:
  Power in streaming mode (average) = 1.21517 W
                          (max)     = 1.22032 W
```

cyclux, 2024-03-15:
Oh wow, awesome! Thank you very much :)
Is this the 2280 B+M key version? Which can be ordered at https://up-shop.org/hailo-m2-key.html ?

JonLech, 2024-03-15:
I've got the Key M model. I wanted M because I may end up using the Hailo-8 on a platform where I can take advantage of 4 PCIe lanes (i.e. not the Pi5). The B+M model only supports 2 lanes.

```
# hailortcli fw-control identify
Executing on device: 0000:01:00.0
Identifying board
Control Protocol Version: 2
Firmware Version: 4.16.2 (release,app,extended context switch buffer)
Logger Version: 0
Board Name: Hailo-8
Device Architecture: HAILO8
Serial Number: XXXXXXXXXXXXXXXX
Part Number: HM218B1C2FAE
Product Name: HAILO-8 AI ACC M.2 M KEY MODULE EXT TEMP
```

[image]

geerlingguy, 2024-03-15:
@JonLech - Awesome! Thanks for posting your results—I've updated the Hailo 8 module page on the site with a 'Yes' for Pi 5: https://pipci.jeffgeerling.com/cards_m2/hailo-8-ai-module.html (also updating tags on this issue to match...).

cyclux, 2024-03-15:
@JonLech Great, thanks for the info! It seems acquiring Hailo (M.2) isn't straightforward. The documentation hints that not all software is freely accessible: `[..] and each model is accompanied by a binary HEF file, fully supported within the Hailo toolchain and Application suite (accessible to registered users only)`. Does this mean registration / product inquiry is a must, or is there a way to purchase a unit from retail (like https://up-shop.org/hailo-m2-key.html) and still use custom models with the available software?

JonLech, 2024-03-15:
@cyclux The [Hailo Model Zoo](https://github.com/hailo-ai/hailo_model_zoo/blob/master/docs/PUBLIC_MODELS.rst) repo has a lot of models already converted to HEF format that anyone can download.

However, if you have a custom model you want to run on the Hailo-8, then you will need convert it to HEF format first using the Hailo AI Software Suite (available for download on the Hailo website after creating an account). You'll need an x86_64 host with a NVIDIA GPU for this process. It's relatively straightforward.

In terms of integrating use of a HEF model into other software, that can be accomplished with the C/C++/Python APIs in their [hailort](https://github.com/hailo-ai/hailort) repo (although docs & tutorials are only available on their website with an account).

cyclux, 2024-03-17:
@JonLech Nice, Thanks a lot for your helpful info!

swdee, 2024-03-18:
In recent [Raspberry Pi news](https://www.raspberrypi.com/news/velo-ai-smart-bike-light/), there is a company Velo AI which reports they are using a CM4 and Hailo chip, so things look positive as for it working on the CM4 too.

I am trying to get hold of a Hailo, when I do I will be able to confirm the specifics of setup on the CM4.

swdee, 2024-04-03:
I received the Hailo-8 A+E Key card and confirm its fully functional on the CM4.  It is very much plug and play without any dramas with the following process.

I used a [Waveshare A->B key adaptor](https://www.waveshare.com/m.2-m-key-to-m.2-a-key.htm) and [M.2 to PCIe adaptor](https://www.waveshare.com/pcie-to-m.2-a.htm) and mounted this in the CM4 Carrier board.

[image]

The CM4 has Debian 11 (Bullseye) installed with Python v3.9.

## Driver Installation

Install Linux kernel headers and dkms
```
apt install linux-headers dkms
```

Download `.deb` PCIe driver and HailoRT packages from [here](https://hailo.ai/developer-zone/software-downloads/) and install,  during install select DKMS install.
```
dpkg -i hailort-pcie-driver_4.16.0_all.deb
```

Reboot

Check dmesg for successful driver loading
```
$ dmesg | grep -i hailo

[    4.501251] hailo_pci: loading out-of-tree module taints kernel.
[    4.599730] hailo: Init module. driver version 4.16.0
[    4.600451] hailo 0000:01:00.0: Probing on: 1e60:2864...
[    4.600476] hailo 0000:01:00.0: Probing: Allocate memory for device extension, 11592
[    4.600565] hailo 0000:01:00.0: enabling device (0000 -> 0002)
[    4.600591] hailo 0000:01:00.0: Probing: Device enabled
[    4.600666] hailo 0000:01:00.0: Probing: mapped bar 0 - 00000000f78ea782 16384
[    4.600693] hailo 0000:01:00.0: Probing: mapped bar 2 - 0000000097da2126 4096
[    4.600716] hailo 0000:01:00.0: Probing: mapped bar 4 - 0000000071b50fb9 16384
[    4.600737] hailo 0000:01:00.0: Probing: Setting max_desc_page_size to 4096, (page_size=4096)
[    4.600775] hailo 0000:01:00.0: Probing: Enabled 64 bit dma
[    4.600789] hailo 0000:01:00.0: Probing: Using userspace allocated vdma buffers
[    4.600808] hailo 0000:01:00.0: Disabling ASPM L0s 
[    4.600829] hailo 0000:01:00.0: Successfully disabled ASPM L0s 
[    4.772490] hailo 0000:01:00.0: Firmware was loaded successfully
[    4.809954] hailo 0000:01:00.0: Probing: Added board 1e60-2864, /dev/hailo0
```

## Install Hailo Runtime

Install HailoRT tools.
```
dpkg -i hailort_4.16.0_arm64.deb 
```

Check that device is recognised.
```
$ hailortcli scan

Hailo Devices:
[-] Device: 0000:01:00.0
```

```
$ hailortcli fw-control identify

Executing on device: 0000:01:00.0
Identifying board
Control Protocol Version: 2
Firmware Version: 4.16.0 (release,app,extended context switch buffer)
Logger Version: 0
Board Name: Hailo-8
Device Architecture: HAILO8
Serial Number: ***
Part Number: HM218B1C2KA
Product Name: HAILO-8 AI ACCELERATOR M.2 A+E KEY MODULE
```

## Python API Installation

Download pyHailoRT libraries by getting the python wheel file from [here](https://hailo.ai/developer-zone/software-downloads/)

Install Python `pip`

```
apt install python3-pip
```

Install pyHailoRT wheel file.
```
pip install hailort-4.16.0-cp39-cp39-linux_aarch64.whl
```

## Run Example Inference

Get example code.
```
git clone https://github.com/hailo-ai/Hailo-Application-Code-Examples.git hailo-examples
```

Run inference code example which uses resnet50
```
cd hailo-examples/runtime/python/inference_run_opencv_classification/
./get_sources.sh
```

Install python requirements (if need)

```
pip install -r requirements.txt
```

Run inference example
```
./run_inference_with_image_opencv.py resnet_v1_50.hef zidane.jpg --labels imagenet1000_clsidx_to_labels.txt
```

## Product Comparison

I have benchmarks applicable to my usage across a number of devices which consists of image classification using an EfficentNet-Lite0 model.   The following table provides a comparison of the performance.

| Device | First Inference | Second Inference |
| --- | --- | --- |
|  Jetson Orin Nano 8GB - CUDA|  3-4 sec | 14-18ms  |
|  Jetson Orin Nano 8GB - CPU|  N/A | 30ms  |
| Raspberry Pi 4B |  150ms | 92ms |
| Raspberry Pi 5 | 67ms | 50ms |
| Khadas VIM3 Pro |  106ms |  78ms |
| Rock Pi 5B - CPU | 65-70ms | 44ms |
| Rock Pi 5B - NPU | 12ms | 6-7ms |
| Raspberry Pi CM4 with Hailo-8 (Blocking API) | 11ms | 4.2ms |
| Raspberry Pi CM4 with Hailo-8 (Streaming API) | N/A | 1.2ms |
| Threadripper Workstation - USB3 Coral |  | 9-11ms  |
| Raspberry Pi CM4 - USB2 Coral |  | 20-27ms  |
| Raspberry Pi 5  - USB2 Coral |  | 20-24ms |
| Raspberry Pi 5  - USB3 Coral |  | 9-12ms |
| Raspberry Pi 4B  - USB2 Coral |  | 20-27ms |
| Raspberry Pi 4B  - USB3 Coral |  | 11-18ms |

The Hailo-8 Blocking API runs the image inference serially and provides a direct comparison to how inference is run on all the other platforms.   The Hailo-8 device is also highly parallel and performs best using the Streaming API.

The Google Coral is a USB device and performs differently depending on if its connected to a USB2 or USB3 port.

gillygils, 2024-05-08:
I see people are getting these to work on a RPi 5. Does anyone have any documentation on how they did it? Like the install steps, what was needed to install, as well as what version of Linux/ or RPi OS they were on. Maybe I missed that information reading through all of this. I have a vision system project I am working on and I think this would be the way to go if it works on Debian 11 or 12, since the Coral TPU really only has some workarounds. Don't get me wrong I have the work around working but I want other to be able to use this project for themselves and they just are the most user friendly workarounds. 

Any help is much appreciated.

Thanks!

geerlingguy, 2024-05-24:
@swdee - Regarding your benchmarking — have you tried the Hailo on the Pi 5, at PCIe Gen 3 speed? I'm wondering if Gen 3 makes any difference vs Gen 2 specifically.

geerlingguy, 2024-05-24:
@gillygils - Outside of registering an account and viewing the 'Developer Zone' documentation, just browsing around the Hailo GitHub repos, it looks like:

  - https://github.com/hailo-ai/hailort-drivers contains a [firmware download script](https://github.com/hailo-ai/hailort-drivers/blob/master/download_firmware.sh) and a [pcie driver](https://github.com/hailo-ai/hailort-drivers/tree/master/linux/pcie#readme) you can install with `make`
  - https://github.com/hailo-ai/hailort contains the runtime, APIs, and CLI
  - https://github.com/hailo-ai/hailo_model_zoo contains models ready to be used with Hailo and instructions for training your own models — see the [Getting Started Guide](https://github.com/hailo-ai/hailo_model_zoo/blob/master/docs/GETTING_STARTED.rst)

swdee, 2024-05-24:
@geerlingguy I tried our Model with the Halio-8 on Pi 5 with Gen 2.0 and 3.0 speeds but this made little difference.

| Device | Second Inference |
| ---- | ---- |
| Pi 5 with Hailo-8 - PCIe Gen 2.0 (Blocking API) |  4.18ms |
| Pi 5 with Hailo-8 - PCIe Gen 3.0 (Blocking API) |  4.14ms |
| Pi 5 with Hailo-8 - PCIe Gen 2.0 (Streaming API) | 1.19ms |
| Pi 5 with Hailo-8 - PCIe Gen 3.0 (Streaming API) | 1.15ms |

The speeds I am getting out of the Hailo-8 are within the same range as Hailo's published values in their Model Zoo.

Since I posted I have also acquired the Radxa Rock 5C Lite and CM5 with their benchmarks as follows.

| Device | Average Inference |
| ---- | ---- |
| Radxa Rock 5C Lite - NPU (Single Core) | 8.83ms |
| Radxa Rock 5C Lite - NPU (3 Cores, 9 Threads) | 2.29ms |
| Radxa CM5 - NPU (Single Core) |  5.95ms |
| Radxa CM5 - NPU (3 Cores, 9 Threads) |  1.21ms |

I managed to get better performance out of the Rockchip NPU by setting CPU Affinity of the program to run on the fast Cortex-A76 cores only.  Before it was switching between the A55 and A76 so performance was slower and was unable to saturate the NPU cores fully.

gillygils, 2024-05-24:
> @gillygils - Outside of registering an account and viewing the 'Developer Zone' documentation, just browsing around the Hailo GitHub repos, it looks like:
> 
>     * https://github.com/hailo-ai/hailort-drivers contains a [firmware download script](https://github.com/hailo-ai/hailort-drivers/blob/master/download_firmware.sh) and a [pcie driver](https://github.com/hailo-ai/hailort-drivers/tree/master/linux/pcie#readme) you can install with `make`
> 
>     * https://github.com/hailo-ai/hailort contains the runtime, APIs, and CLI
> 
>     * https://github.com/hailo-ai/hailo_model_zoo contains models ready to be used with Hailo and instructions for training your own models — see the [Getting Started Guide](https://github.com/hailo-ai/hailo_model_zoo/blob/master/docs/GETTING_STARTED.rst)
> 
> 
> Also looking around online, it seems there's at least one article with a detailed guide for [setting up the Hailo on the Pi 5](https://ucl.nuee.nagoya-u.ac.jp/blog/2024/2024-05-14-raspi5hailo/), though it's in Japanese, so run it through a translator if you need it in a different language!

Awesome! I appreciate it!

gillygils, 2024-05-24:
> @geerlingguy I tried our Model with the Halio-8 on Pi 5 with Gen 2.0 and 3.0 speeds but this made little difference.
> Device 	Second Inference
> Pi 5 with Hailo-8 - PCIe Gen 2.0 (Blocking API) 	4.18ms [...]

What model are you using on the Rock Chip NPU? I have also been messing around with a RK3588S and can only manage to get the test models they provide form their toolkit to work. Shoot I can't even get yolov8 to convert but that seems like a known issue. Now I did find a pre-converted yolov8 model, and converted yolov5 but both give the same error when I run them. Can't recall what that is off the top of my head, or post code since I am not currently with that device. Would love to know what model you used and if you have ever used yolo on the NPU and have some tips as to how to get that working.

swdee, 2024-05-24:
> What model are you using on the Rock Chip NPU? I have also been messing around with a RK3588S and can only manage to get the test models they provide form their toolkit to work. Shoot I can't even [...]

I'm using the RK3588/S/S2/82 and wrote some [Go bindings](https://github.com/swdee/go-rknnlite) to the RKNN-Toolkit.

For YOLO models you have to use Rockchips Ultralytics fork [v5](https://github.com/airockchip/yolov5/blob/master/README_rkopt.md) and [v8](https://github.com/airockchip/ultralytics_yolov8/blob/main/RKOPT_README.md) as they have made some optimizations so it can run faster on the NPU.

gillygils, 2024-05-24:
> > What model are you using on the Rock Chip NPU? I have also been messing around with a RK3588S and can only manage to get the test models they provide form their toolkit to work. Shoot I can't even get yolov8 to convert but that seems like a known issue. Now I did find a pre-converted yolov8 model, and converted yolov5 but both give the same error when I run them. Can't recall what that is off the top of my head, or post code since I am not currently with that device. Would love to know what model you used and if you have ever used yolo on the NPU and have some tips as to how to get that working.
> 
> I'm using the RK3588/S/S2/82 and wrote some [Go bindings](https://github.com/swdee/go-rknnlite) to the RKNN-Toolkit.
> 
> For YOLO models you have to use Rockchips Ultralytics fork [v5](https://github.com/airockchip/yolov5/blob/master/README_rkopt.md) and [v8](https://github.com/airockchip/ultralytics_yolov8/blob/main/RKOPT_README.md) as they have made some optimizations so it can run faster on the NPU.
> 
> Some details on the conversion process of a YOLOv8 model [here](https://github.com/swdee/go-rknnlite/tree/master/example/alpr)

Sweet! I have a yolov8n.rknn file, but I will try the fork to see if I can get that to convert. I'll also check out the Go bindings and see if I can get that to work. I'm still a bit confused as to why I get errors when running the some code to use the yolo.rknn model. Do you have example code for running without the Go binding by chance? If not no worries, I know I will figure this out at some point.

swdee, 2024-05-24:
> Sweet! I have a yolov8n.rknn file, but I will try the fork to see if I can get that to convert. I'll also check out the Go bindings and see if I can get that to work. I'm still a bit confused as to [...]

Rockchips Model Zoo has C++ and Python examples [here](https://github.com/airockchip/rknn_model_zoo/tree/main/examples/yolov8).  I have run all the C++ code as that is what I ported to Go.

gillygils, 2024-05-24:
> Rockchips Model Zoo has C++ and Python examples [here](https://github.com/airockchip/rknn_model_zoo/tree/main/examples/yolov8). I have run all the C++ code as that is what I ported to Go.

Awesome! I feel like when I got that file, on my main pc, I didn't see any example code in the yolov8 folder. I appreciate you pointing that out. I am going to give that a shot when I get home, and hopefully get that working decently well. Been trying to make a vision system to control some lights and haven't been able to get any SBCs to give the performance I want. Tried RPi 5 with coral tpu, was decent but wasn't a fan of needing to jump through hoops to get it to work, RK3588s works great on the test files with different models, hopefully now I can get yolo to work, and now I'm working with a Jetson Nano but seem to have a bad install and re flashing this has been a real pita since it was flashed to the emmc (came like that wasn't my doing). Ideally I want to use the RK3588s as its way cheaper. Here's hoping that now being pointed in the right direction I can get something working! Thank you!

gillygils, 2024-05-24:
> Rockchips Model Zoo has C++ and Python examples [here](https://github.com/airockchip/rknn_model_zoo/tree/main/examples/yolov8). I have run all the C++ code as that is what I ported to Go.

Okay, I think I am stupid or something. I was trying to run the code on the RK3588 chip, and of course get an error. Module not found RKNN. If im not mistaken, doesnt the ARM version use RKNNLite? I see most of the other examples call RKNNLite and not RKNN or am I missing something here? I have RKNNLite installed but never got RKNN to install and figure that was just the x86 version of the software or something. But maybe it is able to be installed and I have done something wrong? Any advice is much appreciated.

swdee, 2024-05-24:
> Okay, I think I am stupid or something. I was trying to run the code on the RK3588 chip, and of course get an error. Module not found RKNN. If im not mistaken, doesnt the ARM version use RKNNLite? I [...]

Sounds like your using Python.  The Quickstart and Userguide [docs](https://github.com/airockchip/rknn-toolkit2/tree/master/doc) for installation and usage may be of help.  However it does depend on what OS you have installed and which SBC vendor's product your using. Take the Radxa Rock 5B for example, the official Debian image has everything already installed.

geerlingguy, 2024-06-04:
FYI, Hailo maintains a set of Raspberrry Pi 5 examples here: https://github.com/hailo-ai/hailo-rpi5-examples — these were just published yesterday, as the 8L is included in Raspberry Pi's new AI Kit, announcement here: https://www.raspberrypi.com/news/raspberry-pi-ai-kit-available-now-at-70/

See full Pi documentation here: https://www.raspberrypi.com/documentation/accessories/ai-kit.html

It seems like support is rolling into [`picamera2`](https://github.com/raspberrypi/picamera2) and [`rpicam-apps`](https://github.com/raspberrypi/rpicam-apps) now as well.

geerlingguy, 2024-06-04:
Oooh... after running `sudo rpi-update` to run the latest kernel, and setting the following in `/boot/firmware/config.txt` and rebooting:

```
kernel=kernel8.img
dtoverlay=pineboards-hat-ai
```

I get successful initialization of both Hailo-8 and Hailo-8L behind my 12-port PCIe switch...

```
pi@ai-pi:~ $ dmesg | grep hailo
[    3.423816] hailo: Init module. driver version 4.17.0
[    3.425961] hailo 0000:05:00.0: Probing on: 1e60:2864...
[    3.425971] hailo 0000:05:00.0: Probing: Allocate memory for device extension, 11600
[    3.425996] hailo 0000:05:00.0: enabling device (0000 -> 0002)
[    3.426004] hailo 0000:05:00.0: Probing: Device enabled
[    3.426035] hailo 0000:05:00.0: Probing: mapped bar 0 - 00000000fd800105 16384
[    3.426043] hailo 0000:05:00.0: Probing: mapped bar 2 - 00000000d598fb0e 4096
[    3.426048] hailo 0000:05:00.0: Probing: mapped bar 4 - 0000000071524406 16384
[    3.426053] hailo 0000:05:00.0: Probing: Force setting max_desc_page_size to 4096 (recommended value is 4096)
[    3.426062] hailo 0000:05:00.0: Probing: Enabled 64 bit dma
[    3.426066] hailo 0000:05:00.0: Probing: Using userspace allocated vdma buffers
[    3.426072] hailo 0000:05:00.0: Disabling ASPM L0s 
[    3.426076] hailo 0000:05:00.0: Successfully disabled ASPM L0s 
[    3.634179] hailo 0000:05:00.0: Firmware was loaded successfully
[    3.649101] hailo 0000:05:00.0: Probing: Added board 1e60-2864, /dev/hailo0
[    3.649225] hailo 0000:08:00.0: Probing on: 1e60:2864...
[    3.649229] hailo 0000:08:00.0: Probing: Allocate memory for device extension, 11600
[    3.649250] hailo 0000:08:00.0: enabling device (0000 -> 0002)
[    3.649258] hailo 0000:08:00.0: Probing: Device enabled
[    3.649277] hailo 0000:08:00.0: Probing: mapped bar 0 - 0000000025baed26 16384
[    3.649283] hailo 0000:08:00.0: Probing: mapped bar 2 - 00000000b242b3f1 4096
[    3.649287] hailo 0000:08:00.0: Probing: mapped bar 4 - 00000000a74f7f8d 16384
[    3.649291] hailo 0000:08:00.0: Probing: Force setting max_desc_page_size to 4096 (recommended value is 4096)
[    3.649301] hailo 0000:08:00.0: Probing: Enabled 64 bit dma
[    3.649304] hailo 0000:08:00.0: Probing: Using userspace allocated vdma buffers
[    3.649309] hailo 0000:08:00.0: Disabling ASPM L0s 
[    3.649320] hailo 0000:08:00.0: Successfully disabled ASPM L0s 
[    3.899110] hailo 0000:08:00.0: Firmware was loaded successfully
[    3.927612] hailo 0000:08:00.0: Probing: Added board 1e60-2864, /dev/hailo1
```

Need to see which change actually made this work. I was making those changes to try out the Coral Edge TPU on latest Bookworm install, and noticed there were no more Hailo errors in dmesg...

geerlingguy, 2024-06-04:
Inference works fine on the Hailo as well, using rpicam-apps. Not sure if that application benefits from multiple Hailo-8's being attached though!

[image]

geerlingguy, 2024-06-04:
I've narrowed it down to this overlay that makes the Hailo work behind a PCIe bridge:

```
dtoverlay=pineboards-hat-ai
```

Therefore, I've opened: [Unable to use Hailo-8L from AI Kit behind PCI Express Bridge/switch](https://github.com/raspberrypi/firmware/issues/1895). And now I need to go eat dinner!

geerlingguy, 2024-06-05:
Looks like all you need is:

```
dtoverlay=pciex1-compat-pi5,no-mip
```

The `pineboards-hat-ai` overlay also solves it, but this one seems more fitting.

Raflimrv, 2024-07-10:
Still new to things like this, can Hailo-8 M.2 run tensorflow lite files with the float data type. Because i've read that USB Coral can only run on integer 8. Thanks

Raflimrv, 2024-07-18:
@Skillnoob I see, thank

saurabh-git-dev, 2024-09-23:
@Skillnoob 
Are you able to convert the yolov8/custom model to hef now? I am trying to find good resources. Hailo docs are not very helpful. I tried different models. But not able to convert. 
Thanks.

Skillnoob, 2024-09-23:
@saurabh-git-dev I haven't tried it yet, but there's this repo which should hopefully work.
https://github.com/cyclux/HailoConverter
Its currently configured for 1 class so you'd need to change that.

saurabh-git-dev, 2024-09-23:
> @saurabh-git-dev I haven't tried it yet, but there's this repo which should hopefully work. https://github.com/cyclux/HailoConverter Its currently configured for 1 class so you'd need to change that.

Thanks, I will try.

luckylinux, 2024-11-11:
I unfortunately did NOT notice that the AI Kit+ 26 TOPS Hailo-8 (unlike the AI Kit) has the Hailo-8 soldered directly onto it. So yeah, there go my Plans on taking it out and putting it into a Normal Server PC. Damn :cry:. Great Way to lock you down to using their Platform (or possibly some other Radxa SBC).

It's true that also back then (when I bought the AI Kit 13 TOPS with the Hailo-8L) the Hailo-8 M.2 was quite a bit more Expensive.

Any Idea on a "Reverse" Adapter ? Unfortunately that will still hit Performance quite Hard at PCIe 3.0 x 1 :disappointed:. I was thinking some Adapter like 
[image]

plus another M.2 to M.2 Extender + M.2 to PCIe Adapter.

Or a PCIe to m.2 Adapter ?

geerlingguy, 2024-11-12:
@luckylinux - There area a couple adapters, but none commercially available, just the board design you could have made yourself: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/663 and https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/661

luckylinux, 2024-11-12:
@geerlingguy: Thanks for your Reply. Right now I initiated a Return, also because I'd be limited to PCIe x1 anyways.

And since the Hailo-8L was cheaper per-TOPS I grabbed 5 of them from a trusted Source.

Just to realize now ... maybe that was NOT so Smart, as I might need an Active PCIe Adapter that doesn't require Bifurcation :disappointed: .

wisdpi, 2024-12-19:
> @luckylinux - There area a couple adapters, but none commercially available, just the board design you could have made yourself: #663 and #661

@geerlingguy @luckylinux  this board is in stock now.  Thanks to @tltangliang
https://www.wisdpi.com/products/hat-to-pcie-m-2
[image]

geerlingguy, 2024-12-19:
@wisdpi - Could you open a separate issue (and maybe add a PR) to add these in as PCIe devices so we can have them up on the main page?

This'd also be great to allow developers to test Pi 5 HATs on CM4 and CM5 IO Boards!

wisdpi, 2024-12-19:
> @wisdpi - Could you open a separate issue (and maybe add a PR) to add these in as PCIe devices so we can have them up on the main page?
> 
> This'd also be great to allow developers to test Pi 5 HATs on CM4 and CM5 IO Boards!

Ok. soon

Marcus-Forte, 2025-02-24:
@luckylinux in their software stack, look for anything like `*libOpenCL.so*` or `*.icd`. I've seen CUDA being mentioned in their stack so it's not far fetched that they do something with OpenCL.

As for checking if it supports SPIRV, only method I know is to connect the whole setup and check OpenCL properties for IR (Interpreter language)

I've sent mail to HAILO as well, waiting for the reply

Marcus-Forte, 2025-02-24:
@luckylinux Thank you very much for investigating!

I don't have the module myself, in fact, I was exactly looking for its OpenCL capability before buying one.

So, the main application does not necessarily have to be linked to its OpenCL implementation. Generally the GPU vendors add OpenCL their implementation as an `.so` library together with every other lib that comes with their software Stack - so your user application would be able to link to it to call CL functions.

E.g with CUDA:

[image]

So in your case it could be somewhere in the filesystem, I suppose, because I don't see a scoped directory with the libraries from HAILO alone

luckylinux, 2025-02-24:
@Marcus-Forte:

Output of `find /usr/ -iwholename "*hailo*" > hailo_files.txt`
[hailo_files.txt](https://github.com/user-attachments/files/18943372/hailo_files.txt)

There are some Open*GL* Tets / Unsupported "Stuff" but no Open*CL*.

In `/lib` there are only these Files with "Hailo" in the Name:
```
/lib/firmware/hailo
/lib/firmware/hailo/hailo8_fw.bin
/lib/modules/6.10.11+bpo-amd64/updates/dkms/hailo_pci.ko.xz
/lib/modules/6.8.12-6-pve/updates/dkms/hailo_pci.ko
```

Not a whole lot of stuff in the Source Code either:
```
root@HOST:/usr/src# grep -ri opencl /usr/src/hailo
/usr/src/hailo/hailort/hailort/external/protobuf-src/third_party/benchmark/README.md:An example use case for this is benchmarking GPU execution (e.g. OpenCL
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:option(TRISYCL_OPENCL "triSYCL OpenCL interoperability mode" OFF)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:mark_as_advanced(TRISYCL_OPENCL)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:# Find OpenCL package
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:if(TRISYCL_OPENCL)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:  find_dependency(OpenCL REQUIRED)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:    $<$<BOOL:${TRISYCL_OPENCL}>:${OpenCL_INCLUDE_DIRS}>
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:    $<$<BOOL:${TRISYCL_OPENCL}>:${BOOST_COMPUTE_INCPATH}>)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:    $<$<BOOL:${TRISYCL_OPENCL}>:${OpenCL_LIBRARIES}>
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindTriSYCL.cmake:    $<$<BOOL:${TRISYCL_OPENCL}>:TRISYCL_OPENCL>
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindComputeCpp.cmake:find_dependency(OpenCL REQUIRED)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindComputeCpp.cmake:if(NOT TARGET OpenCL::OpenCL)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindComputeCpp.cmake:  add_library(OpenCL::OpenCL UNKNOWN IMPORTED)
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindComputeCpp.cmake:  set_target_properties(OpenCL::OpenCL PROPERTIES
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindComputeCpp.cmake:    IMPORTED_LOCATION             "${OpenCL_LIBRARIES}"
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindComputeCpp.cmake:    INTERFACE_INCLUDE_DIRECTORIES "${OpenCL_INCLUDE_DIRS}"
/usr/src/hailo/hailort/hailort/external/eigen-src/cmake/FindComputeCpp.cmake:    INTERFACE_LINK_LIBRARIES         "OpenCL::OpenCL"
/usr/src/hailo/hailort/hailort/external/eigen-src/unsupported/Eigen/CXX11/src/Tensor/TensorDeviceSycl.h:// All devices (even AMD CPU with intel OpenCL runtime) that support OpenCL and
/usr/src/hailo/hailort/hailort/external/eigen-src/unsupported/Eigen/CXX11/src/Tensor/TensorDeviceSycl.h:    // OpenCL doesnot have such concept
/usr/src/hailo/hailort/hailort/external/eigen-src/unsupported/Eigen/CXX11/src/Tensor/TensorDeviceSycl.h:    // OpenCL doesnot have such concept
/usr/src/hailo/hailort/hailort/external/eigen-src/bench/tensors/eigen_sycl_bench_contract.sh:${COMPUTECPP_PACKAGE_ROOT_DIR}/bin/compute++  tensor_contract_sycl_bench.cc -I ../../ -I ${COMPUTECPP_PACKAGE_ROOT_DIR}/include/ -std=c++11 -O3 -DNDEBUG -DEIGEN_MPL2_ONLY -DEIGEN_USE_SYCL=1 -no-serial-memop -mllvm -inline-threshold=10000 -fsycl-ih-last -sycl-driver -Xclang -cl-mad-enable -lOpenCL -lComputeCpp -lpthread -o tensor_contract_sycl_bench ${@:1}
/usr/src/hailo/hailort/hailort/external/eigen-src/bench/tensors/eigen_sycl_bench.sh:-lOpenCL \
```

The README in `/usr/src/hailo/hailort/hailort/external/protobuf-src/third_party/benchmark/README.md` contains this if you are interested
```
## Manual timing
For benchmarking something for which neither CPU time nor real-time are
correct or accurate enough, completely manual timing is supported using
the `UseManualTime` function.

When `UseManualTime` is used, the benchmarked code must call
`SetIterationTime` once per iteration of the `KeepRunning` loop to
report the manually measured time.

An example use case for this is benchmarking GPU execution (e.g. OpenCL
or CUDA kernels, OpenGL or Vulkan or Direct3D draw calls), which cannot
be accurately measured using CPU time or real-time. Instead, they can be
measured accurately using a dedicated API, and these measurement results
can be reported back with `SetIterationTime`.
```

```c++
static void BM_ManualTiming(benchmark::State& state) {
  int microseconds = state.range(0);
  std::chrono::duration<double, std::micro> sleep_duration {
    static_cast<double>(microseconds)
  };

  while (state.KeepRunning()) {
    auto start = std::chrono::high_resolution_clock::now();
    // Simulate some useful workload with a sleep
    std::this_thread::sleep_for(sleep_duration);
    auto end   = std::chrono::high_resolution_clock::now();

    auto elapsed_seconds =
      std::chrono::duration_cast<std::chrono::duration<double>>(
        end - start);

    state.SetIterationTime(elapsed_seconds.count());
  }
}
BENCHMARK(BM_ManualTiming)->Range(1, 1<<17)->UseManualTime();
```

There is no "Dedicated" Folder where the HailoRT is installed to. It's Binaries in `/usr/local/bin`, Libraries in `/usr/local/lib`, Includes in `/usr/local/include/hailo`.

Plus the Source Code which I used for Installation in `/usr/src/hailo/`.

Marcus-Forte, 2025-02-24:
@luckylinux

Huh. The references to it in the `.../external/...` are from third-parties, so not from them..

Yes. It looks a lot like they really don't supported it. Pity ;(
I'm assuming then any calls to their hardware are compiled and sent to the device via the kernel module + their API to it. Not sure how it works but beyond that's their magic.

Thank you once more!

luckylinux, 2025-02-24:
@Marcus-Forte: no worries. Maybe still worth to wait for their Answer, it could be they will implement at a later stage ?

Although I'd say that since this is a Small Module (compared to a full-blown GPU) I wouldn't hold my breath on it. Probably the Hardware doesn't have that many dedicated Resources so a quite direct Interface was required (instead of using something like OpenCL).

Marcus-Forte, 2025-02-24:
Yes, I'm a tiny bit hopeful still.

Main reason that having OpenCL be an interface is that is a step towards enabling your device to use increasingly popular [SYCL](https://sycl.tech/getting-started) (which I successfully used with the Raspberry ARM CPU as an OpenCL device). IMO that opens a lot of doors in High Performance Computing, not only AI stuff.  But I understand those kinds of chips are much more AI oriented than a general purpose GPU - as OpenCL is made for.

For such an affordable price in comparison to use something like Nvidia Jetson or a full blown Raspberry + Nvidia, it would be 
a very good alternative.

vanguard478, 2025-05-22:
@swdee : I was referring to the inference timing table you have posted. As is common with the AI accelerators, the model must be in the board native format, but can you clarify what was the model parameters type ( INT8, Float16, bfloat16,.. etc) for these inference run timings.

swdee, 2025-05-22:
> [@swdee](https://github.com/swdee) : I was referring to the inference timing table you have posted. As is common with the AI accelerators, the model must be in the board native format, but can you [...]

They were all compiled to their native format as INT8.   Keep in mind those results are for an EfficientNet-Lite0 model, as each Model uses different Math functions you could get a relative different set of results if using a YOLO model for example.  So I find bench marking AI/Inference to be highly application specific.