geerlingguy/raspberry-pi-pcie-devices #165: CM4-based Super 8 Camera Digitizer using Pi HQ Camera Module

geerlingguy, 2021-07-02:
See the [Super 8 Camera Digitizer with Raspberry Pi 4 Compute Module](https://www.youtube.com/watch?v=3EvFHfOq9m4) by @mayermakes (also [on GitHub](https://github.com/mayermakes/Super8)):

[image]

[image]

Here's the [project page](https://www.element14.com/community/docs/DOC-96758/l/episode-505-super-8-camera-digitizer?CMP=SOM-YOUTUBE-PRG-E14PRESENTS-EP505-COMM) on element14's website. The PCB actually reads "Film replacement cartridge system", and it is CC-BY-SA-NC.

closed 2021-07-02 by geerlingguy

mayermakes, 2021-07-02:
Thanks for featuring, will add the new files to the project repo soonish.

geerlingguy, 2021-07-02:
My hope is to do a video summarizing the explosion of new CM4 projects in the next week or two. (Who knew people would start doing more with the CM4 when it was actually shipping in volume?)

mayermakes, 2021-07-02:
I think the most important change since the CM4 released is that building a project with embedded Compute module is now actually smaller/compact than with a normal RPi. the old compute module never appealed to me as a viable option.
my only gripe with the CM4 is the fragility of the connector if replugged often and availability.
Butt I can see myself using it much more often.

commit 9bdb0879 (2021-07-02): Fixes #165: Add Super 8 film cartridge digitizer project.g