geerlingguy/raspberry-pi-pcie-devices #168: Test Intel QuickAssist (QAT) Adapter 8970

Doridian, 2021-07-03:
Trying to even boot the Raspberry Pi with this card inserted causes it to refuse to boot at all.
I no longer even get the "rainbow" bootloader screen, nothing...
Putting the card into an x86 system and it works perfectly fine.
Card has a green indicator LED which likely just indicates power is going to the card, which is lit up solid in both cases.

[image]
[image]

Is there anything I could try? I ran rpi-update, so I am on the latest FW. Any idea how to debug this further, to see where the Pi actually gets stuck?

I can run rpiboot if I disable EEPROM booting and that makes the Pi successfully appear as a storage device.

geerlingguy, 2021-07-03:
If you have the Pi plugged into an HDMI display, do you see anything at all on the screen when it boots? Any kernel panic or anything? If there is literally nothing (not even bootloader), then maybe it's a power issue.

What kind of adapter are you using to plug the card into the Pi? Maybe that adapter is bad? Does the Pi boot up normally again once you unplug the card?

Doridian, 2021-07-03:
> If you have the Pi plugged into an HDMI display, do you see anything at all on the screen when it boots? Any kernel panic or anything? If there is literally nothing (not even bootloader), then maybe [...]

I get absolutely nothing on the HDMI screen at all.
Power adapter? It is a lab bench power supply which I can see is completely stable at 12.00V.

Removing the QAT card and replacing it with any other PCIe card I have (LSI SAS HBA, 10GbE NIC) makes the Pi boot. Obviously then there's driver issues, but point being, with the QAT card I don't even get the rainbow screen from the bootloader, or any display output at all.

em-et, 2021-07-19:
Have you tried checking the logs with `journalctl`?

Coreforge, 2021-07-23:
I kinda doubt it, but if you're only supplying 12V and the riser you're using doesn't have an extra power connector, it could be that the card draws more on the 5V/3.3V rail than the regulators on the IO Board can deliver, and it doesn't boot because of that. It would be kinda weird though that it works with USB boot in that case, so I wouldn't be too sure this is it.

Doridian, 2021-07-24:
> I kinda doubt it, but if you're only supplying 12V and the riser you're using doesn't have an extra power connector, it could be that the card draws more on the 5V/3.3V rail than the regulators on [...]

That sounds like an idea to explore. I'll see if I can see any drops with a multimeter (and failing that, if I can see any with my oscilloscope, but I doubt those short drops would or should cause a persisting boot failure)