geerlingguy/raspberry-pi-pcie-devices #170: Uptime Labs' upberry - Tiny CM4 carrier

geerlingguy, 2021-07-05:
The [Upberry](https://www.instagram.com/p/CQga-_YL5OL/) is a tiny CM4 carrier with only HDMI, USB-C for power and OTG support, and status and activity LEDs:

[image]

[image]

There will be a machined aluminum case that keeps with the trim aesthetic, and it should be able to fit in as a tiny display computer, or something like a Chromecast, if you have a WiFi version on the CM4.

geerlingguy, 2021-09-02:
It's on the site now!

closed 2021-09-02 by geerlingguy

mazyargholami, 2024-09-16:
Where can I found this product or kicad file?