geerlingguy/raspberry-pi-pcie-devices #173: Add CM4 to CM3 adapter board

dev-dull, 2021-07-12:
They've a few different versions including one that adds a Coral ml module. I've linked the base model below. I'm curious to know how well these work with the Turing Pi

clicky link: https://www.gumstix.com/cm4-uprev.html

geerlingguy, 2021-07-13:
I should note that it's been tested and discussed on the Turing Pi Discord, and so far nobody's been able to get the Uprev to work with the Turing Pi. Not sure why, but there could be signaling or power issues.

geerlingguy, 2021-07-13:
It looks like the [development board](https://pipci.jeffgeerling.com/boards_cm/gumstix-raspberry-pi-cm4-devboard.html) is on the site, but not the Uprev. Labeling accordingly.

geerlingguy, 2021-08-24:
I seem to have acquired a couple of these thanks to @ahmadexp! I'll hopefully test them out soon in a few different scenarios. Physically they don't fit a number of CM <= 3+ boards, unfortunately.

punnypenguins, 2022-04-11:
Live site page for this board: https://pipci.jeffgeerling.com/boards_cm/gumstix-rpi-cm4-uprev.html

geerlingguy, 2022-04-11:
@paulwratt - They do not seem to work in the Turing Pi, unfortunately :(

Haven't tested elsewhere since then, though. But if I can get my hands on a CM4S I want to see if _that_ will work.

nowtp, 2024-03-18:
> @paulwratt - They do not seem to work in the Turing Pi, unfortunately :(
> 
> Haven't tested elsewhere since then, though. But if I can get my hands on a CM4S I want to see if _that_ will work.

Just as an FYI ... but more as I was going through google looking for information.

https://discord.com/channels/754950670175436841/1191718968000516136/1217032548740366377

Seems that someone has got 6 nodes instead of 7 (supply 12V, 4.17a), running on the TPi1 board.