geerlingguy/raspberry-pi-pcie-devices #176: Works fine till update

l0gical, 2021-07-19:
Hey,
Thanks for the hardwork making our lives easier to spin up a Marvel Card and CM4
Fresh install of buster, working all good.
Once tested with LSBLK and all fine, sudo reboot and still all good.
APT-GET Update/Upgrade, no drives visible any more.
Also tried a fresh Install of buster, did the APT Update/Upgrade before following the guide and doesn't find the disks either.
Totally out of my depth on this, presume there is some conflict but cannot find it :/
Chris

geerlingguy, 2021-07-20:
I'm trying to figure out what is being asked—it sounds like maybe you're trying to do something with SATA drives, but maybe something else. In order for anyone to help, we'd need to know exactly what you're trying to do, exactly what kind of card or drives or equipment you're testing, and what you have done to set things up (e.g. did you compile a custom kernel, did you just flash Raspbian to your Pi, are you using a Pi?

l0gical, 2021-07-20:
HI,
Sorry, I should have been clearer.
Pi CM4 and the official IO Board
Marvel 9215 4Port PCIE/Sata which looks identical to the IO-Crest one you have.

sudo apt install -y git bc bison flex libssl-dev make libncurses5-dev
git clone --depth=1 https://github.com/raspberrypi/linux
cd linux
export KERNEL=kernel7l                             # use kernel8 for 64-bit, or kernel7l for 32-bit
make defconfig #(used the menu system rather than nano editor)
#nano .config
#CONFIG_ATA=m
#CONFIG_SATA_AHCI=m
#CONFIG_SATA_MV=m #(isnt showing as required)

make -j4 zImage modules dtbs # 'Image' on 64-bit #compile it on the Pi (yeah its slow but just leave it to it)
sudo make modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img

#did add the prefix to the build as mentioned on your guide/post.

sudo shutdown now
#power cycle the device.

LSBLK (all drives showing)
All good up till this point, do APT-GET UPDATE/UPGRADE and reboot the device, the Card has vanished along with drives etc, also appears that the PCI Bridge has vanished and expecting this could be the issue?
Don't claim to be a Linux expert in the slightest sadly.

OS wise, latest version of 32bit Raspbian

Also, if I do a fresh install of Raspbian then do the Update/Upgrade, follow your guide as above, has the exact same issue, wondering if something in the latest Update/Upgrade Libs are breaking something and perhaps someone else has come across this, feel free to chime in.

Thanks
Chris

6by9, 2021-07-21:
```apt update && apt upgrade``` gets the latest stable kernel which is only updated every few months. It knows nothing about you having built your own kernel, so it will overwrite it.

Support for ATA and SATA_AHCI was added in https://github.com/raspberrypi/linux/commit/1ea02333d959ea9005eb7215a89add2a6da3b25a on 21st April.

Checking the latest raspberrypi-kernel package info, that appears to be 20210527 (ie 27th May), so I would have expected that to have included the relevant patch. Check by using ```uname -a``` which will give you the kernel version and build date. If it's after 21st April 2021 then it should have the relevant modules built in.

SuNNjek, 2021-07-22:
@6by9 going by https://github.com/raspberrypi/linux/commits/raspberrypi-kernel_1.20210527-1 the commit doesn't seem to be included yet

l0gical, 2021-07-22:
Hey all,

Unsure if this is useful.
Popped another MicroSd Card in, Fresh install of Raspbian, ran the APT Update/Upgrade.
It installed Kernel 20210527-1 installed over 20210430

LSPCI showing the card, no probs, LSBLK no drives visible.
Ran DMESG as I saw in one of Jeff's posts, is hitting the bar limit.
Followed one of Jeff's guides on upping the bar limit but a load of warnings and wouldn't allow to copy the DTB file to the /Boot.
Kinda stuck on this as definitely out of my depth.

As above, unsure if this info is of any use to this issue.

l0gical, 2021-07-22:
# Recompile the Device Tree from the dts (source) file.
dtc -I dts -O dtb ~/test.dts -o ~/test.dtb
`/home/pi/test.dtb: Warning (unit_address_vs_reg): /soc: node has a reg or ranges property, but no unit name
/home/pi/test.dtb: Warning (unit_address_vs_reg): /soc/axiperf: node has a reg or ranges property, but no unit name
/home/pi/test.dtb: Warning (unit_address_vs_reg): /soc/gpiomem: node has a reg or ranges property, but no unit name
/home/pi/test.dtb: Warning (unit_address_vs_reg): /emmc2bus: node has a reg or ranges property, but no unit name
/home/pi/test.dtb: Warning (unit_address_vs_reg): /scb: node has a reg or ranges property, but no unit name
/home/pi/test.dtb: Warning (unit_address_vs_reg): /v3dbus: node has a reg or ranges property, but no unit name
/home/pi/test.dtb: Warning (avoid_unnecessary_addr_size): /soc/firmware: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
/home/pi/test.dtb: Warning (unique_unit_address): /soc/mmc@7e300000: duplicate unit-address (also used in node /soc/mmcnr@7e300000)
/home/pi/test.dtb: Warning (unique_unit_address): /soc/firmwarekms@7e600000: duplicate unit-address (also used in node /soc/smi@7e600000)
/home/pi/test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 1 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 3 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 5 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 7 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 9 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/cprman@7e101000:clocks: cell 11 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e201000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e201000:clocks: cell 2 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/mmc@7e202000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/i2s@7e203000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/spi@7e204000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/i2c@7e205000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dpi@7e208000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dpi@7e208000:clocks: cell 2 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 2 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dsi@7e209000:clocks: cell 4 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/aux@7e215000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e215040:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/spi@7e215080:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/spi@7e2150c0:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/pwm@7e20c000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/hvs@7e400000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dsi@7e700000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dsi@7e700000:clocks: cell 2 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/dsi@7e700000:clocks: cell 4 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/i2c@7e804000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/vec@7e806000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/usb@7e980000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 2 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 4 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/watchdog@7e100000:clocks: cell 6 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e201400:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e201400:clocks: cell 2 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e201600:clocks: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e201600:clocks: cell 2 is not a phandle reference
/home/pi/test.dtb: Warning (clocks_property): /soc/serial@7e201800:clocks: cell 0 is not a phandle reference
[... cut ...]
/home/pi/test.dtb: Warning (gpios_property): /sd_vcc_reg:gpio: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (gpios_property): /cam1_reg:gpio: cell 0 is not a phandle reference
/home/pi/test.dtb: Warning (gpios_property): /__symbols__:gpio: property size (19) is invalid, expected multiple of 4`

# Copy the new Device Tree into place.
sudo mv ~/test.dtb /boot/bcm2711-rpi-cm4.dtb

`pi@raspberrypi:~ $ sudo mv ~/test.dtb /boot/bcm2711-rpi-cm4.dtb
mv: failed to preserve ownership for '/boot/bcm2711-rpi-cm4.dtb': Operation not permitted
`

6by9, 2021-07-22:
@ SuNNjek Thanks - I forget that the apt releases are tagged in Github.
It looks like 20210527-1 was a special done to add support for the PoE+ HAT rather than a major kernel bump.

rpi-update will get you the latest kernel which will include the relevant modules, but the normal warnings apply about potential regressions. And again apt will not know that you've done it, so it will overwrite and potentially downgrade your kernel on you.

6by9, 2021-07-22:
@l0gical All of those BAR warnings are for IO bars. Those are not supported on ARM, but are also unnecessary for SATA cards except for legacy emulation modes.

l0gical, 2021-07-22:
@6by9 
Thanks, in that case, am even more at a loss.
Not one of those people that tries to be lazy by getting others to sort issues for them, but really unsure how to proceed on this.
Would you have any *suggestion* on what I should be looking at, again, don't mind trying to figure this out myself, just unsure where to look.
Thanks either way

6by9, 2021-07-22:
What does ```lspci -k``` report? It should give you the device info, and the kernel driver in use with it.

l0gical, 2021-07-22:
Hi @6by9 
No Kernal driver that I can see

pi@raspberrypi:~ $ lspci -k
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20)
        Kernel driver in use: pcieport
01:00.0 SATA controller: Marvell Technology Group Ltd. Device 9215 (rev 11)
        Subsystem: Marvell Technology Group Ltd. Device 9215
pi@raspberrypi:~ $
----------------------------------------------------
Also for good measures, I did a:
pi@raspberrypi:~ $ sudo lspci -v
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 63
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 00000000-00000fff
        Memory behind bridge: 00000000-000fffff
        Capabilities: [48] Power Management version 3
        Capabilities: [ac] Express Root Port (Slot-), MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [180] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
        Capabilities: [240] L1 PM Substates
        Kernel driver in use: pcieport

01:00.0 SATA controller: Marvell Technology Group Ltd. Device 9215 (rev 11) (prog-if 01 [AHCI 1.0])
        Subsystem: Marvell Technology Group Ltd. Device 9215
        Flags: fast devsel
        I/O ports at 0000
        I/O ports at 0000
        I/O ports at 0000
        I/O ports at 0000
        I/O ports at 0000
        Memory at 600040000 (32-bit, non-prefetchable) [size=2K]
        [virtual] Expansion ROM at 600000000 [size=256K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit-
        Capabilities: [70] Express Legacy Endpoint, MSI 00
        Capabilities: [e0] SATA HBA v0.0
        Capabilities: [100] Advanced Error Reporting
pi@raspberrypi:~ $

elmeyer, 2021-07-22:
> @SuNNjek Thanks - I forget that the apt releases are tagged in Github.
> It looks like 20210527-1 was a special done to add support for the PoE+ HAT rather than a major kernel bump.
>
> rpi-update will [...]

@l0gical Seems pretty clear cut to me that this is what happens. You recompile and install the kernel yourself, but it gets unconditionally replaced by running `apt-get upgrade` afterwards. You need to reinstall your self-compiled kernel again afterwards, or recompile again based on the new sources.

Don't know off the top of my head where exactly the kernel is placed on the RPi, but your self-compiled kernel _should_ still be there, and if I'm not mistaken it's a simple matter of pointing the bootloader back at that kernel. Hopefully someone else can help you with the specifics.

6by9, 2021-07-22:
I think I agree with @elmeyer here.

If you've still got your self-built kernel tree.
Assuming you've built it on the Pi, rerun the 
```
sudo make modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img
```
steps from https://www.raspberrypi.org/documentation/linux/kernel/building.md.
If cross-compiling, then run the "Install directly onto the SD card" steps from that page.

l0gical, 2021-07-22:
Thank you all very much, 6by9 did state that the update/upgrade replaces the Kernel but has no idea of previous modifications/self compiled etc.
I will copy my self compiled kernel from my other MicroSd card over and see how that works out, will of course post my findings here as may help someone else in the same boat.

@6by9 Sure was self compiled on the Pi itself (other uSD card) so going to try that now.
In my defence my confusion was "Support for ATA and SATA_AHCI was added in raspberrypi/linux@1ea0233 on 21st April." was assuming that the Kernel didn't need touching hence not even thinking of that.

Back shortly with an update on this, thanks again, Chris

l0gical, 2021-07-23:
Thanks and you are indeed correct, it is now working.

Genuinely appreciated on this, can now sort my nas system out. :)

closed 2021-07-23 by l0gical

SuNNjek, 2021-07-29:
@l0gical fyi, a new kernel has been released now that includes the SATA drivers out-of-the-box

l0gical, 2021-07-29:
Hey,

Awesome and thanks!
I will have a play with that over the weekend as I currently have a queue of Chia Plots transferring to the disks.
Will give that a try (Fresh image of Raspbian, APT Upgrade&&Update) and see if it detects the disks, will post the results here.

By the way, currently works fantastically with CM4 + Io Board + Marvel PciE 9215 card running 4x 8TB NAS N300 Sata Drives for anyone else reading this, chucked OMV+Rclone on it and built a very frugal and compact NAS.
Transfers 100GB in just over 20 minutes via GbE so 2x USB2.0 by my estimates.

Chris

l0gical, 2021-07-30:
@SuNNjek 
Fabulous, that works straight out of the box!
Downloaded Pi OS Lite (7th May 2021), imaged the card, ran the Sudo APT Update && Upgrade.
Rebooted
All SATA Drives visible via LSBLK

Nice work guys, will make a lot of peoples lives easier who are on limited Linux knowledge as per myself.
Chris

App-Teck, 2021-07-30:
Oh thanks that's worked for me update && upgrade and reboot . Thanks a lot lovely guys.