geerlingguy/raspberry-pi-pcie-devices #177: Add Waveshare Dual Gigabit Ethernet Base Board

geerlingguy, 2021-07-21:
I saw [Waveshare's Dual Gigabit Ethernet Base Board for Raspberry Pi Compute Module 4](https://thepihut.com/products/dual-gigabit-ethernet-base-board-for-raspberry-pi-compute-module-4) in a tweet from @ThePiHut today:

[image]

Looks to have a 2nd Gigabit Ethernet port off an internal USB 3.0 bus (with a Realtek chip for the NIC), similar to the USB 3.0-based approach [Seeed used on their board](https://pipci.jeffgeerling.com/boards_cm/seeed-dual-gig-cm4-routerboard.html). Biggest question is if it can perform well on that 2nd interface. It does have a lot more IO than the other boards, so that could make it a nice platform for DIY routers.

guibom, 2021-08-29:
Another Waveshare board that I recommend is the [Type B version of the POE](https://www.waveshare.com/Compute-Module-4-PoE-Board-B.htm) carrier board. 
The difference from the POE one you already documented is that this one has a PCIe 1x slot!

I've been using it for my tests with SATA boards and it works well.

[image]

geerlingguy, 2021-08-29:
@guibom - Can you add a new issue for that board? It does look nice, especially the more compact layout, the color-coded everything, and that slot—though why they don't make it open ended is a mystery to me!

YumingChang02, 2022-01-21:
There is another small alternative to Waveshare's Dual Gigabit Ethernet Base Board.
This configuration moves one of the USB3 from a type a receptacle to a M.2 type B ( usb3 only, no pcie ), which I am testing with an HP branded EM7565 modem inside,
minor changes will be moving type c to the other side ( so antennas will not collide with type c power ), RTC battery is moved to underside of the board
[CM4-DUAL-ETH-4G/5G-BASE ](https://www.waveshare.com/wiki/CM4-DUAL-ETH-4G/5G-BASE)
[image]
[image]

geerlingguy, 2022-01-21:
@YumingChang02 - Can you add this new alternative (with the modem) as a new issue on this repo for easier tracking?

That looks like a really nice way to build a complete 4G/5G, WiFi, WAN/LAN router!

geerlingguy, 2022-01-21:
Moved the 4G/5G variant to it's own issue: #367

iwalker2000, 2022-02-07:
Did anyone test this board? I tried to start it and found that ANY USB storages (USB sticks, USB HDD/SSD) are significantly slow for READ (like 3MB/s max). Even SSD mount process takes a few minutes. Same time USB storage WRITE speed is good (about 250MB/s). Any ideas/suggestions?