geerlingguy/raspberry-pi-pcie-devices #178: Add CM4 gigabit switch board.

tlalexander, 2021-07-23:
I've just uploaded the design files for my CM4 gigabit switch board to github so I suppose it could be added!

https://github.com/tlalexander/rpi-cm4-switch-board

[image]

geerlingguy, 2021-07-23:
This is a pretty cool idea—is the gist of it that you'd have both Pis on internal Gigabit connections, then the switch would also have three Gigabit ports? I think the main thing I'm trying to see based on this single picture is how the ports are routed to the Pis (or if they're just part of the switched backplane, and the Pis are on the same thing).

tlalexander, 2021-07-23:
Yeah I guess it's not a proper router because each ethernet connection goes directly to the switch. So maybe not a router board so much as an interesting tool. Really this is a stepping stone to a version of the board with four CM4s and one ethernet jack, all passed through the gig switch.

TheGuyDanish, 2021-07-23:
Passing through the switch's management interface to one of the CM4's via SPI or i2c could make for an interesting addition as well. For setting things like VLANs.

tlalexander, 2021-07-23:
Ah yes that's right. I haven't looked in to the software side to see if drivers exist, but I suppose I should make sure to hook that up and worry about software later.

TheGuyDanish, 2021-07-23:
> Ah yes that's right. I haven't looked in to the software side to see if drivers exist, but I suppose I should make sure to hook that up and worry about software later.

Without having read the datasheet for the switch chip yet, I'll assume it exposes all it's configuration registers over either of those buses. So even if software doesn't exist, you could still manually push settings to it by pushing values to the register in something like Python over SPI. You could build software around that, but I'd say for the first prototype, just being able to read and write to the registers is a huge win.

TheGuyDanish, 2021-07-23:
https://github.com/torvalds/linux/blob/d7c5303fbc8ac874ae3e597a5a0d3707dc0230b4/drivers/net/dsa/microchip/ksz9477.c

Looks like you may be in luck though. ;)

tlalexander, 2021-07-23:
Oh that is great!

geerlingguy, 2021-09-10:
Board has been added; if there's new information at some point feel free to open up a new issue/PR to get the page updated!

closed 2021-09-10 by geerlingguy