geerlingguy/raspberry-pi-pcie-devices #181: Add Raspberry Pi CM4 TV Stick

geerlingguy, 2021-07-29:
The [Raspberry Pi CM4 TV Stick](https://hackaday.io/project/180969-raspberry-pi-cm4-tv-stick) is a carrier that plugs directly to HDMI port. Perfect for Kodi, RetroPie, or just using a Pi attached directly to a monitor or TV for any use case. Power is through USB-C, though.

[image]

Hat tip to [Matthew Turland](https://twitter.com/elazar/status/1420511405460148232) for the link.

magic-blue-smoke, 2021-08-22:
GitHub home for this project: https://github.com/magic-blue-smoke/Raspberry-Pi-CM4-TV-Stick
Twitter: https://twitter.com/magic__smoke (double underscore)
Boards are in production now and expected to be available for purchase late September.
Production boards will be 2mm longer to fit ambilight connector. Status LEDs moved from the edge under CM4 to the edge around USB-C connector. Third USB port available as solder pads "just in case"

geerlingguy, 2021-08-23:
@magic-blue-smoke - Excellent, thanks for posting the updated info! And for making this board, it looks great!

closed 2021-09-09 by geerlingguy

geerlingguy, 2023-11-22:
Page: https://pipci.jeffgeerling.com/boards_cm/raspberry-pi-cm4-tv-stick.html