geerlingguy/raspberry-pi-pcie-devices #185: Test Waveshare's slim PCIe to USB 3.0 board built specifically for CM4 IO board

geerlingguy, 2021-08-04:
See: [PCI-E to USB 3.2 Expansion Card for Raspberry Pi Compute Module 4 IO Board](https://amzn.to/2VfqbcR)

[image]

[image]

geerlingguy, 2021-08-04:
Ordered one today!

nezra, 2021-08-22:
This works, but there is a nasty quirk, fyi.

The raspberry pi external header is based on the atx floppy header. This puts it as 5v gnd gnd 12v.

This card from waveshare has the same physical header attached, but they don't use standard atx floppy pinouts for the wiring. They have it as 12v gnd gnd nc. They don't include a wiring loom, so if you attempt to hook it up to an atx supply, you'll be putting 5v on the 12v input port unless it's switched.

I caught the mistake before I hooked it all up and powered it on, so I am unsure what happens if you supply it with 5v. It seems as though it may not be needed to connect this header however.

geerlingguy, 2021-08-23:
@nezra - That seems... surprising. If 5v, it seems like at least it wouldn't blow anything up, but it still is weird to use that connector but not the standard pinout.

I double-checked on mine and you're right, just 12v/Gnd/Gnd/NC.

geerlingguy, 2021-10-21:
You would need to connect power to the power header on this board (but see @nezra's comment above).

michaelarmstrong, 2022-05-15:
I bought one of these recently and have had nothing but trouble even after wiring up my own breadboard pins to the alternate pin-out. 

Wondering if there is a more elegant solution for PCI-e power on the CM4 I/O board rather than 2 incoming feeds... Failing that, it'd be interesting to see another vendor come up with a 12V input I/O board that doesn't step down to 5V immediately.