geerlingguy/raspberry-pi-pcie-devices #193: UptimeLab MinCab minimal CM4 carrier board

just-grizzle, 2021-08-10:
Hey,

just read about the MinCab CM4 board over at [CNXSoft](https://www.cnx-software.com/2021/08/10/mincab-smallest-raspberry-pi-cm4-carrier-board/) .

[image]

It would nicely fit on your list of CM4 boards.
Even though the articles links towards the UpLab website, I could not find any information there. But as you have good connections, you'll surely find out more. 

Regards and until next time!

geerlingguy, 2021-08-10:
Yes, I noticed Ivan posted about it on Twitter too: https://twitter.com/Merocle/status/1424701350567350276

> Regards and until next time!

Heh.

I think the MinCab takes the crown of the smallest possible CM4 carrier board now!

just-grizzle, 2021-08-20:
That's for sure. But I don't get the use case yet. Like it though.

geerlingguy, 2021-09-08:
Added!

closed 2021-09-08 by geerlingguy