geerlingguy/raspberry-pi-pcie-devices #203: Add JMicron JMB362 SATA board

guibom, 2021-08-28:
This JMB362 is an older JMicron card and controller, but it works really well.
No need for custom drivers or settings, it worked out-of-box with the latest kernel.
Seems like a good solution for somebody that needs only 2 drives, and doesn't want to modify their 1x PCIe slot.

[image]

```
pi@raspberrypi:~ $ sudo lspci -v

01:00.0 SATA controller: JMicron Technology Corp. JMB362 SATA Controller (rev 10) (prog-if 01 [AHCI 1.0])
        Subsystem: JMicron Technology Corp. JMB362 SATA Controller
        Flags: bus master, fast devsel, latency 0, IRQ 66
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        I/O ports at <unassigned> [disabled]
        Memory at 600010000 (32-bit, non-prefetchable) [size=512]
        [virtual] Expansion ROM at 600000000 [disabled] [size=64K]
        Capabilities: [8c] Power Management version 3
        Capabilities: [50] Express Legacy Endpoint, MSI 00
        Capabilities: [94] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Kernel driver in use: ahci
        Kernel modules: ahci
```
```
pi@raspberrypi:~ $ sudo hdparm -tT /dev/sda

/dev/sda:
 Timing cached reads:   2016 MB in  2.00 seconds = 1009.23 MB/sec
 Timing buffered disk reads: 446 MB in  3.00 seconds = 148.52 MB/sec
```

geerlingguy, 2021-08-28:
@guibom - Thanks! Can I use your image in the page on the database site? This does indeed look like a great way to pop one or two SATA devices on the CM4.

guibom, 2021-08-28:
Yeah for sure, you can use the image!

geerlingguy, 2021-09-06:
Added to the site. Thanks again!

closed 2021-09-06 by geerlingguy

fentonfentonfenton, 2023-07-10:
There's no affiliate link for this :)