geerlingguy/raspberry-pi-pcie-devices #204: Add Intel I210 NIC - supports hardware PTP timestamping

geerlingguy, 2021-08-31:
In tandem with my work on the Time Card (see #199), I noticed @lasselj had tested the [Intel I210]() on a Raspberry Pi and it seems to be working:

[image]

@lasselj - I was wondering, did you have to do anything special to get Intel's driver to work on Pi OS in your testing?

lasselj, 2021-09-03:
@geerlingguy It's [this](https://thepihut.com/products/metal-enclosure-for-raspberry-pi-compute-module-4-io-board-with-cooling-fan) case. Kudos to whomever decided to make this slice in the case to make a standard PCIe card fit:
[image]

geerlingguy, 2021-09-06:
Added to the site!

closed 2021-09-06 by geerlingguy

antevens, 2023-05-11:
For future reference I think it's probably worth noting that the I210 (and from what I can tell any device using the Intel igb driver) can't send ONESTEP PTP packets, it can only do two step.

coconutneu, 2025-04-25:
> For future reference I think it's probably worth noting that the I210 (and from what I can tell any device using the Intel igb driver) can't send ONESTEP PTP packets, it can only do two step.

Can you provide more details on this? This is the description from the datasheet of i210
7.8.2.2.4 1-step Time Synchronization Phase Procedure
Packet processing in the master and the Slave for 1-step procedure is almost identical to the 2-step procedure as follow:
• The master software sends the SYNC packet while indicating it to the hardware by the 1STEP_1588 flag. Doing so, the hardware inserts the transmission time in the SYNC packet.
• The slave samples the reception time of the SYNC and extract its transmission time at the master.
• From this point the flow is identical to the 2-step procedure.
So from this, it seems that the I210 should be capable of sending 1-step SYNC packets via hardware. Or maybe I'm wrong?

antevens, 2025-04-25:
It's been a while since I looked at this, I remember trying to get onestep to work and recompiling but the hardware simply never sent it.

In the end I gave up on using an RPi and had to revert back to a hardware GM source for our gNodeBs.

lasselj, 2025-04-25:
Hi. So I don't remember the exact details for the i210 (odds are the same), but for the i225/i226, the issue is that whereas the hardware can update the sync timestamp, it cannot recalculate the UDP header checksum. I once submitted kernel patches to enable then layer 2 PTP one-step, but Richard Cochran turned them down on netdev for what I at the time considered no compelling reason. As my life is definitely too short to quarrel with him, I never revisited. Use 2-step. It works just as well in 99.5% of cases.

coconutneu, 2025-05-16:
Hi, thanks a lot for sharing these insights! I’ve made some driver modifications myself, but unfortunately the one-step functionality still doesn’t seem to be working correctly. If you happen to have any guidance or tips on that, I’d be truly grateful.

lasselj, 2025-05-16:
@coconutneu It should be in the netdev mailing list archive, but I’m happy to have a look and forward patches. Please bear with me 🙂