geerlingguy/raspberry-pi-pcie-devices #219: Generic JMicron JMB363

rocketeerbkw, 2021-09-06:
I was able to get an old PCI-E SATA card working with a [CM4108000](https://www.raspberrypi.org/products/compute-module-4/?variant=raspberry-pi-cm4108000) and the [waveshare PoE board](https://www.waveshare.com/wiki/Compute_Module_4_PoE_Board_(B)). I'm using Ubuntu 21.04 so I had to compile a custom kernel with SATA support.

There are no markings for manufacturer that I can see and it's been sitting in a box long enough that I don't remember where it came from, but it's using a JMB363 chip, so I think any other card with the same chip should work. There aren't any similar cards on amazon, but a google shopping search shows a bunch of clones on aliexpress for <$10.

The card I have can use eSATA or SATA, I have only tested the two internal SATA ports. It also has a single IDE connector which I haven't tested. There is a "JMicron PATA Support" kernel module that might enable the IDE support.

Looks like #203 with IDE support. The position of the PCI-E connector on the IO board is just a bit too far away from the edge requiring the removal of the bracket (really annoying oversight by waveshare).

```bash
lspci -v
01:00.0 SATA controller: JMicron Technology Corp. JMB363 SATA/IDE Controller (rev 03) (prog-if 01 [AHCI 1.0])
	Subsystem: JMicron Technology Corp. JMB363 SATA/IDE Controller
	Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@1,0/usb@1,0
	Flags: bus master, fast devsel, latency 0, IRQ 68
	Memory at 600010000 (32-bit, non-prefetchable) [size=8K]
	Expansion ROM at 600000000 [virtual] [disabled] [size=64K]
	Capabilities: [68] Power Management version 2
	Capabilities: [50] Express Legacy Endpoint, MSI 01
	Kernel driver in use: ahci
	Kernel modules: ahci

01:00.1 IDE interface: JMicron Technology Corp. JMB363 SATA/IDE Controller (rev 03) (prog-if 85 [PCI native mode-only controller, supports bus mastering])
	Subsystem: JMicron Technology Corp. JMB363 SATA/IDE Controller
	Flags: fast devsel
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	Capabilities: [68] Power Management version 2
```
[image]
[image]

6by9, 2021-09-06:
> ```
> 01:00.1 IDE interface: JMicron Technology Corp. JMB363 SATA/IDE Controller (rev 03) (prog-if 85 [PCI native mode-only controller, supports bus mastering])
> 	Subsystem: JMicron Technology Corp. [...]

Unlikely to work as I/O addressing is not supported on ARM (everything is memory mapped, unlike on x86).

rocketeerbkw, 2021-09-06:
Unlikely enough that it's not worth testing further? I could scrounge up a IDE drive if there's a chance it could work.

The 2 SATA ports work and that's all I wanted the card for at the moment.

geerlingguy, 2021-09-06:
Heh, I think I still have an IDE drive sitting around in one of my boxes. Though it has nothing worthwhile on it. Far more interesting would be if old SCSI and SCSI-II cards would work somehow. I have a few of those drives that I still have to plug through an old Mac laptop to get them to work.

6by9, 2021-09-06:
> Unlikely enough that it's not worth testing further? I could scrounge up a IDE drive if there's a chance it could work.

PATA_JMICRON enables https://elixir.bootlin.com/linux/v5.14.1/source/drivers/ata/pata_jmicron.c#L42
That calls into [libata-sff](https://elixir.bootlin.com/linux/v5.14.1/source/drivers/ata/libata-sff.c). Looking at that code there are a load of ```ioreadX``` and ```iowriteX``` calls.
My understanding of those calls is that they are pretty much exclusively x86 ```in``` and ```out``` opcodes for I/O access, but there's no harm in trying them.

I would offer to send you a PATA HDD as I still have a fair number lying around, but it'd cost too much to ship.

Old SCSI cards are likely to hit the same issue of living in an x86-centric world. At least with SCSI being (in theory) backwards compatible, finding a SCSI-III card is plausible, and that should still work with SCSI-I and II drives.
The only SCSI drive I still have around is a 170MB drive from my old Atari ST! I don't know if it will still spin up.

rocketeerbkw, 2021-09-09:
I recompiled with PATA_JMICRON and it looks like that was the correct module, but it doesn't work still

```
01:00.1 IDE interface: JMicron Technology Corp. JMB363 SATA/IDE Controller (rev 03) (prog-if 85 [PCI native mode-only controller, supports bus mastering])
	Subsystem: JMicron Technology Corp. JMB363 SATA/IDE Controller
	Flags: fast devsel
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	I/O ports at <unassigned> [disabled]
	Capabilities: [68] Power Management version 2
	Kernel modules: pata_jmicron          <<<<<< NEW!
```

I have these errors in dmesg:
```
[    6.362147] pata_jmicron 0000:01:00.1: of_irq_parse_pci: failed with rc=-22
[    6.362273] pata_jmicron 0000:01:00.1: no available native port
```
I don't see any changes in the `pci` output in dmesg. I thought i'd save/diff the output between the two kernels but I guess things don't always run in the same order so it was a lot of noise, if there's something else to look for in there I don't know what it is.

The disk does power up when the pi boots, so that gives me some indication that the IDE port works, but I haven't tested it (the card or the disk) in a x86 machine.

punnypenguins, 2021-09-10:
Adding for anyone trying to get/use their own JMicron JMB363:

### Purchasing the JMicron JMB363:
[**This**](https://store.cwc-group.com/jmb363.html) appears to be the only storefront on the internet still selling the card. As of August 2021, they're listed at **$33 USD** per unit, regular price. They're on sale right now for $18 USD, but I have no idea how long that will last.

### Using the JMicron JMB363:
**Drivers** - I'm not linking one because I can't pretend to know what equipment and software everyone will be working with. If you don't have a working driver yet and want to try Googling it, YMMV, but it shouldn't be too hard to find the right thing. Alternatively, I'd suggest just asking someone else who already has used the card to point you towards the right driver. Save yourself time!

**User Guide** I hunted it down—you can find the full PDF for online access OR download [**here**](https://manualzz.com/doc/23852319/jmb36x-raid-ap-user-guide-v1.0). I'm also attaching the [JMicron JMB363 User Guide.pdf](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/7146436/JMicron.JMB363.User.Guide.pdf) to this comment if you want to grab it from here instead.

geerlingguy, 2021-09-11:
I'll leave this issue open for now in case you want to post back any more test results with the IDE port. But feel free to close if there's nothing else you want to test, as the card's up on the site now too!