geerlingguy/raspberry-pi-pcie-devices #268: Test Radxa Taco 5x SATA NAS with 2.5G and 1G Ethernet

geerlingguy, 2021-10-06:
See original issue: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/202

[image]

I have a Taco (well, the Penta main board that goes inside) and would like to do some testing on it; run some benchmarks, test compiling ZFS, etc.

oDn, 2021-10-06:
This is VERY exiting!

andyattebery, 2021-10-07:
I was able to get the Realtek driver working without having to recompile the kernel with the module on the newest version of Raspberry Pi OS. Jeff mentions in his [post about 2.5GbE](https://www.jeffgeerling.com/blog/2020/testing-25-gbps-ethernet-on-raspberry-pi-cm4), he had trouble compiling it, so maybe something has changed.

1. Install raspberrypi-kernel-headers (`sudo apt install raspberrypi-kernel-headers`)
2. Download [2.5G Ethernet LINUX driver r8125 for kernel up to 5.6](https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software)
3. Unarchive driver (`untar -xf r8125-9.006.04.tar.bz2`)
4. Build and install driver (`cd r8125-9.006.04; sudo ./autorun.sh`)

[image]
[image]
[image]

geerlingguy, 2021-10-07:
@andyattebery - Good to know! Did you use 32-bit Pi OS or 64-bit? A lot has changed in the past year, it might install easily on both now.

andyattebery, 2021-10-07:
Both 32-bit and 64-bit work. On one 32-bit installation `lspci` wasn't returning anything and the SATA drives weren't showing up. However, I just did a fresh install, updated the packages, installed the driver, and everything is working fine.

mjeshurun, 2021-10-08:
> @mjeshurun @bydorfler we have the 4x PCIe switch for the CM4, which are used as following:
> 
> * 1x for 2.5 GbE
> * 1x for 5x SATA
> * 1x for NVMe
> * 1x for WiFi 6
> 
> which do you think not important and want to replace? The idea was to build a NAS with NVMe as cache and fast WiFi streaming.

Hi @hipboi , I think all the PCIe ports you mentioned are important. 
@bydorfler 's suggestion might be a better alternative. To replace 1x SATA port with a PCIe connector. 
4x SATA ports are plenty to connect enough TB using SSD/HDD, so an extra PCIe will be much more valuable.

geerlingguy, 2021-10-08:
@mjeshurun - Sure, but it's limited to a maximum of around 70 Mbps (90 if you're lucky and have a really, really clear signal). That's good enough for some use cases :)

geerlingguy, 2021-11-08:
First boot—used [this 12V 8A power supply with 2.5mm barrel plug](https://amzn.to/30cGCJ8), booted into Pi OS 64-bit lite, on a 4 GB CM4 Lite with a Sandisk Extreme 32GB microSD card.

A few initial observations:

  - There's one green 'power' LED that is on when the board is powered.
  - There's no 'ACT' LED like other CM4 boards, so besides a wired Ethernet connection, no direct indication on the board when Pi is actively doing anything (nice to have)

`lspci` with nothing plugged into anything and system not updated:

<details>
<summary>Click to expand output</summary>
  
```
pi@taco:~ $ sudo lspci -vvvv
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20) (prog-if 00 [Normal decode])
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Bus: primary=00, secondary=01, subordinate=06, sec-latency=0
	I/O behind bridge: 00000000-00000fff
	Memory behind bridge: c0000000-c01fffff
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <1us, L1 <2us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt+ ABWMgmt+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna- CRSVisible+
		RootCap: CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE# ARIFwd-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled ARIFwd-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=1us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us

01:00.0 PCI bridge: ASMedia Technology Inc. Device 1806 (rev 01) (prog-if 00 [Normal decode])
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Bus: primary=01, secondary=02, subordinate=06, sec-latency=0
[... 395 lines cut ...]
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [148 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [168 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Capabilities: [178 v1] Transaction Processing Hints
		No steering table available
	Capabilities: [204 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [20c v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=150us PortTPowerOnTime=150us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [21c v1] Vendor Specific Information: ID=0002 Rev=4 Len=100 <?>
```
</details>

geerlingguy, 2021-11-10:
Getting Realtek 2.5G NIC working, attempt number 1:

  1. Download the [2.5G Ethernet LINUX driver r8125 for kernel up to 5.6](https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software) driver version `9.006.04` (had to solve an annoying math captcha first).
  2. Run:

     ```
     tar vjxf r8125-9.006.04.tar.bz2
     cd r8125-9.006.04/
     sudo ./autorun.sh
     ```

  3. TODO.

geerlingguy, 2021-11-10:
Testing on 2.5 Gbps—one thing I noticed right away, and this may be a bug: neither the amber connection light nor the green activity light on the 2.5 Gbps port light up when I connect it to my 10G switch. Connected to my 1G switch, I get a blinking amber light, but no green light.

So maybe some wires are crossed with the port LEDs to the NIC? It might support multi-mode LEDs for 1/2.5G indication and maybe the board schematic is crossing some connections somewhere.

Anyways, a trip to `ip a` shows that the connection is active. And using `ethtool`, it looks like it's negotiating the correct speeds:

```
$ ethtool eth1
Settings for eth1:
	Supported ports: [ TP ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Half 1000baseT/Full 
	                                     10000baseT/Full 
	                                     2500baseT/Full 
	                                     5000baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
	MDI-X: Unknown
Cannot get wake-on-lan settings: Operation not permitted
	Current message level: 0x00000033 (51)
			       drv probe ifdown ifup
	Link detected: yes
```

geerlingguy, 2021-11-10:
Now comes the fun part... let's see if a [Sabrent Rocket 8TB NMVe SSD](https://amzn.to/3wyh4Cf) works in the M.2 slot...

[image]

It fits, even though it's a double-sided card (luckily the M.2 port is nice and tall).

[image]

The M.2 slot uses an M2.5 screw (and my board didn't come with one). That seemed a little odd since most of my M.2 devices/slots seem to come with M2-size screws (M2 meaning the measurement of the ISO metric screw thread, not 'screw meant for M.2'). I was lucky I had a pack of said screws from my Pi PoE HATs!

[image]

TODO: Results.

geerlingguy, 2021-11-10:
And next... 5x [Samsung 870 QVO 8TB SATA SSDs](https://amzn.to/30bLjDl) (and no, that's not a typo... lol):

[image]

I noticed that all the drive status LEDs are not only glaringly bright blue (aaah!!), they are also all lined up directly under the middle slot... so good luck seeing them. Not sure if the official case will use some very creative light pipes to show drive status, or if you'll just operate the drives without blinkenlights.

[image]

I also noticed the preinstalled headers on the board's edges are basically mm away from the drives once installed. It would be nice to have a smidge more clearance—and 3.5" drives will _definitely_ not fit on this board with those headers in place.

Formatted one drive (`sda`) and mounted to `/mnt/sda` (found device with `sudo parted -l`):

```
TODO
```

geerlingguy, 2021-11-11:
Testing out various RAID levels by [creating a RAID array in Linux with mdadm](https://www.jeffgeerling.com/blog/2021/htgwa-create-raid-0-array-linux-mdadm).

RAID 0 Results:

TODO.

geerlingguy, 2021-11-11:
Still working on the above comment—but have to wait a bit for the resync. So until that's done, I'm going to take a gander at the [work @joshuaboud did in this Pi kernel fork](https://github.com/45Drives/linux/tree/storinator-jr) to get ZFS compiling on the Pi.

My plan is to try to get ZFS running after I finish testing the RAID 5 setup, so I can test RAIDZ1 speeds, and then do some network file copy tests maybe with both NFS and SMB.

geerlingguy, 2021-11-12:
Before ZFS tests, I'm going to test Samba performance to/from my Mac. I set up a Samba share [following my simple guide](https://www.jeffgeerling.com/blog/2021/htgwa-create-samba-smb-share-on-raspberry-pi).

Benchmark methodology:

```
# All run on Mac:
mkfile 5G test.zip && \
rsync -h --stats test.zip /Volumes/shared/test.zip && \
rm -f test.zip && \
rsync -h --stats /Volumes/shared/test.zip test.zip && \
rm -f /Volumes/shared/test.zip  && \
rm -f test.zip
```

In RAID 0:

| Source/destination | Result |
| --- | --- |
| From Taco to Mac | TODO MB/sec |
| From Mac to Taco | TODO MB/sec |

In RAID 5:

| Source/destination | Result |
| --- | --- |
| From Taco to Mac | 139.48M MB/sec |
| From Mac to Taco | 70.66 MB/sec |

In RAIDZ1:

| Source/destination | Result |
| --- | --- |
| From Taco to Mac | TODO MB/sec |
| From Mac to Taco | TODO MB/sec |

> **NOTE**: `rsync` that ships with macOS is absurdly slow (it is a bit older than the latest, too). Finder copies were going like 4x faster than CLI copies with `rsync`. So I made sure to upgrade to a later version with `brew install rsync`.

markwort, 2021-11-12:
Any ideas why the rsync performance might be so slow, compared to direct disk access?
Plenty of older CPUs struggle with modern ssh ciphers due to lack of acceleration, so for connecting to my "trash NAS" (sporting a Phenom II X4 965 ) I use the follwoing option passed to rsync:
`-e 'ssh -c chacha20-poly1305@openssh.com'`

[Here's](https://gist.github.com/joeharr4/c7599c52f9fad9e53f62e9c8ae690e6b) a bash script from someone else to test transfer speeds of different ssh ciphers and [here](https://blog.twogate.com/entry/2020/07/30/benchmarking-ssh-connection-what-is-the-fastest-cipher) is a blog post with some benchmarks (coincidentially on "Raspberry Pi 3B+" and "Mac mini 2018 Core i7-8700B").

I don't think you'd need to test this for all RAID configurations - if this cipher overhead is a problem for one, then it is a problem for all :smile:  .

geerlingguy, 2021-11-12:
@markwort - I'm using `rsync` here not to copy between a remote server and local, but rather local to a locally-mounted volume. So the transfers are going across the network, and I've also verified (as close as I can) that direct Finder copies are showing the same (or at least similar, ±10%) performance.

markwort, 2021-11-12:
@geerlingguy sorry for not seeing the forest for the trees, it must have been too early for me.
I thought I had something useful to contribute, and was too busy lining it out and didn't take a closer look at what exactly you're actually measuring!

geerlingguy, 2021-11-12:
I also just tested the power button functionality—after you shut down the Pi, assuming it's still plugged in, you can press the power button and it will boot back up.

Note that when powered down, it seems the PCIe circuits may still have power, because the green LED on the board and the blue LEDs for the SATA drives are all lit up.

Also, when soft-powering this way, it seems like the SATA drives didn't come up normally; maybe the PCIe bus link was down :/

geerlingguy, 2021-11-12:
Power consumption:

  - Idle: 11.2W
  - Writing over network: 

As a point of comparison, my ASUSTOR Lockerstor 4 is pretty similar—about 10W idle (with no drives powered up), and 20-40W in operation.

geerlingguy, 2021-11-12:
And just to verify, I also plugged one of my older SATA spinny 3.5" hard drives in using a [combo SATA power/data extension cable](https://amzn.to/3kwTSPO), and it comes up quite nicely:

```
pi@taco:~ $ sudo hdparm -I /dev/sda

/dev/sda:

ATA device, with non-removable media
	Model Number:       WDC WD5000AVDS-63U7B1                   
	Serial Number:      WD-WCAV9Y717334
	Firmware Revision:  01.00A01
	Transport:          Serial, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6
Standards:
	Supported: 8 7 6 5 
	Likely used: 8
```

geerlingguy, 2021-11-12:
Alrighty, I think I've tested this board to death. Working on a script for a video now, and hopefully I'll have that up next week! Feel free to continue any discussions in this issue! I should have those new 10G transceivers in today :)

closed 2021-11-12 by geerlingguy

andyattebery, 2021-11-13:
I set up a 4 drive, striped + mirrored zpool with recordsize=1M on my hacked together build with a 8GB CM4 pictured below. 

I'm seeing some wild numbers using your `iozone` tests: 900+ MB/s 1M reads/writes even with the file size set to 20 GB and 4-5 MB/s 4K reads/writes, so I don't know what's going on there.

I also ran your `rsync` test using a 16GB file from my Mac mini (with 10GbE) to the 2.5 GbE port on the Taco with more reasonable results:
- Mac to Taco: 140.28M bytes/sec
- Taco to Mac: 130.68M bytes/sec

I'd be happy to run some more tests if you'd like; I have another 8TB hard drive I can add to the array. However, its looking like PCIe 2.0 lane is going to be the real bottleneck even for spinny drives.

About the build... I know the PSU is probably unnecessary, but I didn't want to buy $35 in cables only to find out the Taco couldn't drive 4-5 spinny hard drives.

[image]

[image]

geerlingguy, 2021-11-13:
@andyattebery So with stripe + mirror, I can definitely see the writes being a bit faster. You lose out on another drive's worth of storage and have to stick with an even number, but I didn't even think about trying RAID10 or striped mirror zpool in my tests.

For the iozone tests, I kept getting wild numbers with lower total sizes so kept increasing until the numbers stabilized (and seemed more rational). Before increasing to `5000M` for the 1M random test and `500M` for the 4K test, I was getting numbers anywhere from 100-1200 MB/sec, which seemed a bit wild, but must've been based on ram caches.

I think once you get past a certain threshold the RAM cache mechanism breaks down and you see what ZFS is actually able to do on the disks.

I'm wondering if a striped mirror in ZFS really can put through more data than the bus would indicate? More testing required... As I'll mention in my video next week, I'm very new to ZFS and it seems a lot more complicated than the filesystems and RAID setups I'm used to!

markwort, 2021-11-13:
If you want a quick way to pretty much disable all of the fancy ARC caching that ZFS does, you can change the `primarycache` setting.
Get the current values:
```
root@host:~# zfs get primarycache
NAME  PROPERTY      VALUE  SOURCE
data  primarycache  all    local
```
change the value:
```
root@host:~# zfs set primarycache=metadata data
```

The other available settings are all (=cache files and metadata) and none. I think allowing ZFS to keep the metadata cache is fair.

AFAICT, when using a ZFS filesystem, the ARC should be the only thing between you and the disks, so there should not be any additional buffering happening in the kernel.

geerlingguy, 2021-11-15:
Since I think it might be acceptable to add r8125 support to the Pi kernel to make these common 2.5G adapters work out of the box, I've opened an issue upstream: https://github.com/raspberrypi/linux/issues/4699

geerlingguy, 2021-11-15:
So this is cool... apparently support was added recently, so if you want an even quicker fix, you can run `sudo rpi-update` and it'll work out of the box. Hopefully that makes its way to stable firmware soon so it'll be available everywhere!

iandk, 2021-11-16:
I can't get my board to detect my microSD/ SATA SSD.
The boot always fails with this output:

```
USB xHC init failed
SD: card not detected
```

Any ideas?
Do you have to set a jumper or something to set the boot device?

geerlingguy, 2021-11-16:
@iandk - Do you mean you're trying to boot with a USB hard drive, and not microSD? Or you're trying to boot of a microSD card with Pi OS on it? It's not clear from your comment what you're trying to do—and are you doing this on a Taco board or some other CM4 board?

iandk, 2021-11-16:
I’ve tried booting off a microSD as well as a SATA SSD which I connected directly to the SATA slots. 
Both had the newest version of PiOS installed. 
Yea, I’m testing a Taco Board

geerlingguy, 2021-11-16:
Is the CM4 you're using a Lite version or one with eMMC installed? If using an eMMC compute module, it will never allow any kind of microSD card boot.

SATA boot support is not implemented on the Raspberry Pi currently; see https://github.com/raspberrypi/firmware/issues/1653

iandk, 2021-11-16:
oh, I thought I had the lite version without emmc but just checked it indeed has emmc. 

That means I probably have to get another board which allows me to flash PiOS the emmc flash?

geerlingguy, 2021-11-16:
@iandk - Yeah, it looks like (afaict so far) the Taco doesn't have the ability to boot the Pi into usbboot/rpiboot and allow flashing eMMC compute modules.

hipboi, 2021-11-17:
yeah, this is the issue we did not consider for the first version. Now for the new hw revision, we added a small button for the usbboot.

geerlingguy, 2021-11-17:
Video is live! https://www.youtube.com/watch?v=G_px298IF2k

vebmaster, 2021-12-06:
> Video is live!

Thx. Where did you buy the Radxa Taco board and for how much?

geerlingguy, 2021-12-06:
@vebmaster - The one I used in the video was provided by Radxa; it should hopefully be available soon on their website.

ThomasKaiser, 2021-12-22:
Wrt your benchmarking:

  * it's well known that both [Finder and Windows Explorer do auto-tuning of network settings](https://www.helios.de/web/EN/support/TI/157.html). How did you measure the block size `rsync` and 'the network stack' were using? This number has a huge impact on performance therefore at least I prefer to test with tools that use predictable block sizes (like e.g. Helios Lantest)
  * Have you checked what the RPi was doing while benchmarking? For example using `sbc-bench -m`?
  * Have you checked samba settings on both NAS using `testparm`? I just noticed that a bunch of former optimisations that were in my 'OMV for SBC install script' have disappeared in the meantime...

geerlingguy, 2021-12-22:
@ThomasKaiser - For the benchmark monitoring, I ran the tests in three different conditions: 1. monitoring with `atop` at 2s intervals, 2. monitoring with `top` in 2s intervals, and 3. not running any tool to monitor resource usage at all during the benchmark. I do that third test because even though it *should* be minimal, injecting the calls to get monitoring data could conceivably impact performance (no matter how minimally).

Nothing else was running during the benchmarks (I actually ran them all again without OMV installed at all) besides what is preinstalled in the lite Pi OS 64-bit image.

For the network file copy I used [this rsync command](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/268#issuecomment-966797635), and I also set up a separate comparison (that I didn't fully document in this issue) where I did the following:

1. Took a 36 GB folder with about 1,200 files in it—one of my video projects.
2. Started a screen recording in iShowU Instant on my mac at 60fps.
3. Measured the time it took from dropping the file into the NAS in the finder to the time the copy progress dialog disappeared. Using that time and the exact folder size in bytes to get a MB/s rating.
4. Did the same copy operation using `rsync` to copy the folder.

And the final result between rsync and Finder were within about 1s of each other (which surprised me... it _felt_ like rsync was slower, just watching my network graph in iStatMenus).

I haven't done any more advanced checking of the SMB connection details—but it seems other people in the Pi community have noticed similar issues with samba file copies not being as fast as they were a year or two ago.

iandk, 2021-12-22:
I can confirm that the Samba performance dropped with the kernel upgrade from 5.4 to 5.10.
I did lot's of testing, going back to 5.4 always improved the performance.

ThomasKaiser, 2021-12-22:
@geerlingguy apologies for asking in the wrong issue, then deleting the comment here while you were already answering.

AFAIK neither `atop` nor `top` show actual CPU clockspeeds? Do you set the cpufreq governor always manually to performance prior to testing or do you trust in the clockspeeds being fully ramped up anyway?

And as [already mentioned](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/162#issuecomment-999782619) block sizes matter in a world where network stacks implement auto tuning of settings. You would either need a network sniffer or a tool that has control over the block size to get what you test (Helios Lantest for example).

BTW: I'm in the wrong issue since my concerns are not about 'RPi samba performance degradation' but about the measured differences between the RTD1296 thing and the Taco. Without checking details like the negotiated block size (or smbd tunables) it's hard to interpret the numbers or attribute them to 'hardware' while they're in reality just different settings.

geerlingguy, 2021-12-22:
@ThomasKaiser heh, we just keep going back and forth! I've moved my comment body over to https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/162, let's keep the discussion there since it has more of the detailed benchmarking in that issue :)

ThomasKaiser, 2021-12-22:
@iandk just glanced through the referenced rpi issue. The drop in SMB write performance after some seconds looks like 'buffer full' situations.

In case you can still reproduce it this way with more recent kernels than 5.4 in combination with Samba I would run on the RPi in a terminal `vmstat 2` (or maybe `vmstat 1` if the performance drop happens too fast) and start a SMB write effort.

`free   buff  cache` are interesting and `wa` on the right (that's %iowait).

iandk, 2021-12-22:
I think the current state is, that it’s always stuck to around 70-80mb/s Write, instead of the 112mb/s stable with kernel 5.4

ThomasKaiser, 2021-12-23:
Well, I fired up my RPi 4B (early model with just 1GB) to check for this (5.10 kernel). No Windows involved but with macOS 12 as SMB client [I get ~100 MB/s in both directions](https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/162#issuecomment-1000368318) which is perfectly fine to me...

iandk, 2021-12-23:
You should remove all of those cache settings, no send/ receive file, no caching. 
With 5.4 I was able to get a sustained 112mb/ in both directions.

ThomasKaiser, 2021-12-23:
> You should remove all of those cache settings, no send/ receive file, no caching.

Why exactly should I do that?

iandk, 2021-12-23:
The Pi 4 was (with kernel 5.4) able to saturate the 1Gbit/s port without a problem, you don't need a cache.
This will just result in performance drops over time, as you transfer larger files.

iandk, 2021-12-23:
Additionally lots of users have exactly the same issue, the current performance is quite a bit slower than a few months ago. 
Even one of the devs could confirm the problem, see my GitHub issue.  

I stillt think that this is related to the kernel update 5.4 -> 5.10

ThomasKaiser, 2021-12-23:
> Additionally lots of users have exactly the same issue, the current performance is quite a bit slower than a few months ago.

Yeah, I heard this. Jeff also said the same. And everybody refuses [active benchmarking](https://www.brendangregg.com/activebenchmarking.html) and looking into settings. You said above

> it’s always stuck to around 70-80mb/s

I gave it a try and have to report: nope, it's not. I truly believe that you had more before and that 70-80 MB/s are pathetic especially given [how Windows Explorer accelerates SMB transfers since Windows 7](https://www.helios.de/web/EN/support/TI/157.html). But what's the point in using bad settings and not actively benchmarking stuff to get a clue _what_ has changed? :)

As for "remove all of those cache settings, no send/ receive file, no caching" and "you don't need a cache. This will just result in performance drops over time, as you transfer larger files." I don't understand what you mean (but it doesn't matter that much anyway since I'm fine with the performance I get with '5.10' or as I would call it 'appropriate settings'). There's the `write cache size = 524288` setting which `man smb.conf` helps to explain. Nothing will slow down with 'large files' with this sort of cache 512K in size.

iandk, 2021-12-24:
I did lot's of testing and benchmarking a few months ago.
Tested different clients, drives, samba versions, kernel versions, usb enclosures.... etc. 

But again, there's no reason to use a cache, as it was always running perfectly fine and hitting the Gbit limits without one.

ThomasKaiser, 2021-12-24:
> `rsync` that ships with macOS is absurdly slow (it is a bit older than the latest, too). Finder copies were going like 4x faster than CLI copies with `rsync`.

Have you had a look with e.g. `tshark` with which chunk size data has been transmitted with either version? IIRC older `rsync` versions used something as low as 700 bytes while newer ones go up to 128K or something like this...

ctsrc, 2023-11-20:
I have a Radxa Taco and I would like to put it in a case. Since no official case exists yet, my idea is to buy a small PC case made for mATX or ITX mobo NAS, and use extender cables to connect the drives from the bays to the Radxa Taco.

Do the two holes in the Radxa Taco board line up so that I can screw the board in place inside of a case made for mATX or ITX?

judokan9, 2023-11-24:
@geerlingguy  Thanks for the videos about the Radxa Taco!

I have mine fitted with a CM4 and it is running very stable so far, however I have had problems with the screws supplied for the CPU cooling being too short and I can't get the RTC to work.

Has anyone got the RTC chip to work?
There is also a question from another user in the [forum ](https://forum.radxa.com/t/how-to-connect-rtc/14882) that has been unanswered since February and I have not been able to address the chip either.

racky1, 2023-12-15:
I'm late to this discussion but will we ever be able to boot from M.2 NMVe on the Radxa Taco?

geerlingguy, 2024-12-18:
@racky1 - On the CM5, it may be possible—it can boot through the ASMedia Switch chip on here... however, today I tried getting the CM5 to boot on it, but couldn't (tried with a Lite CM5). Not sure why not.

I also plugged in a CM4, and was able to boot off microSD, and could see the M.2 slot, but didn't spend much time trying to boot off it. I'm not sure if the CM4-generation firmware was ever updated (or _could be_ updated) to boot through a PCIe switch.

I also wanted to mention that at least as of Pi OS 12 / Bookworm today, the 2.5 Gbps Ethernet NIC works out of the box, as well as the SATA ports, I didn't have to tweak any settings on a default Pi OS install! (Yay)

Though I did only get 1.8 Gbps down and 1.2 Gbps up when testing through my 2.5 Gbps network. I would normally expect 2.3 Gbps...

koutheir, 2024-12-20:
Does anybody know if there is a similar solution (based on a single board computer) for a direct-attached storage (DAS) instead of a network-attached storage (NAS)?

hipboi, 2024-12-24:
@koutheir You can check Radxa SBC with USB 3.0 OTG such as ROCK 5A or 5C with Penta SATA HAT. The speed is limited to around 200 ~ 300MB/s for the USB OTG data transfer.

koutheir, 2024-12-25:
@hipboi, thank you very much for the recommendation!

bakadave, 2025-03-20:
> I'm late to this discussion but will we ever be able to boot from M.2 NMVe on the Radxa Taco?

I have made some progress on that front. What I've done so far:

- built a minimal ARM64 kernel with kexec support
- compiled a statically linked kexec binary
- created a custom initramfs with busybox
- wrote an init script to:
  - load PCIe switch driver
  - find and mount the SSD
  - sets up kexec and executes it
- successfully booted over to Raspbian on the SSD via kexec
 
Unfortunately I haven't been able to actually complete the boot, I can see text whizzing by as usual with a boot, but after a while it must get stuck somewhere because the screen goes blank. I hope to figure it out at a later time.

Also the whole image on the SD is <50MB, and shouldn't write anything to the card itself. The whole process takes around 10-15 seconds.

Since I am not an embedded Linux expert, I used Sonnet 3.7 exentsively both for help and to learn stuff.