geerlingguy/raspberry-pi-pcie-devices #284: m.2 Marvell 88se9230 RAID10, Hyperduo 4x SATA

BeauSlim, 2021-10-19:
[image]

I don't think this particular chip has been mentioned here yet.  It is the successor to the 88se9130, with 2x PCIe 2.0 lanes and SATA III.  It features Marvell's "Hyperduo" feature set that allows SSDs to act as caches for HDDs, as well as striping, mirroring or RAID 10.  TRIM is supported along with port multipliers.  Apparently the card will even do disk encryption with a modified ROM.  

Like the enterprise RAID cards Jeff has talked about, this chip helps the Pi not be bogged down with software RAID like it does with the JMB585 card, for example.   I'm getting good Samba file sharing performance with lots of CPU headroom to keep the Pi snappy and spare bus bandwidth to write files locally.  

A tiny m.2 version is available.  It is wider than most m.2 cards and requires a riser to fit onto many boards.  I paid $50 CAD for a Syba branded board (SD-ADA40118) on Amazon and am using it with the Waveshare CM4-IO-BASE-B.  

This really seems like the perfect SATA card for a CM4 based NAS, but there's a catch:  Management and monitoring is all via boot x86 BIOS, x86/64 web software, or x64 EFI, none of which works with the card in the Pi.  I had to create my RAID configuration on a Linux PC and then move the card over.  There may be some way of getting it working under emulation, but I am doubtful.  Not being able to monitor the health of a mirror will likely be a deal-killer for many.

geerlingguy, 2022-05-09:
On the site: https://pipci.jeffgeerling.com/cards_storage/marvell-88se9230-host-controller.html

closed 2022-05-09 by geerlingguy

Tony763, 2022-08-25:
Hi, just a note. In order to run it on Ubuntu Server 22.04, `linux-modules-extra-xxxxxx-raspi` must be installed, as `sata_mv.ko` is not present in the default package.

BenSeventy9, 2022-08-28:
I am currently testing a HighPoint r640l 4 port card. I struggled with update firmware until I found this: https://www.asrockrack.com/support/faq.asp?k=Marvell9230 Works like a charm.

This would be nice to have for arm64:
https://github.com/stegm/marvell_msu_docker
Have tested it on Ubuntu 22.04 and does the job.

I have no experience with emulation but is there a way to get that working with Box64?

BenSeventy9, 2022-09-02:
I have got mvcli running with Box64.
Follow this steps to install Box64: https://pimylifeup.com/raspberry-pi-x64
See readme for installing mvcli: https://support.lenovo.com/ch/en/downloads/ds547431-mvcli-command-line-tool-for-linux

What I don't understand is that I have to run install.sh everytime after a reboot. Error messages:
```
sudo mvcli info -o hba
Dynarec for ARM64, with extension: ASIMD CRC32 PageSize:4096
Box64 with Dynarec v0.1.9 b185afa built on Sep  2 2022 18:22:00
Using default BOX64_LD_LIBRARY_PATH: ./:lib/:lib64/:x86_64/:bin64/:libs64/
Using default BOX64_PATH: ./:bin/
Counted 16 Env var
Looking for /usr/local/bin/mvcli
argv[1]="info"
argv[2]="-o"
argv[3]="hba"
Rename process to "mvcli"
Error loading needed lib libmvraid.so
Error loading one of needed lib
Error: loading needed libs in elf /usr/local/bin/mvcli
```
After run install.sh:
```
sudo mvcli info -o hba
Dynarec for ARM64, with extension: ASIMD CRC32 PageSize:4096
Box64 with Dynarec v0.1.9 b185afa built on Sep  2 2022 18:22:00
Using default BOX64_LD_LIBRARY_PATH: ./:lib/:lib64/:x86_64/:bin64/:libs64/
Using default BOX64_PATH: ./:bin/
Counted 16 Env var
Looking for /usr/local/bin/mvcli
argv[1]="info"
argv[2]="-o"
argv[3]="hba"
Rename process to "mvcli"
Using emulated libmvraid.so
Using native(wrapped) libpthread.so.0
Using native(wrapped) libdl.so.2
Using native(wrapped) libc.so.6
Using native(wrapped) ld-linux-x86-64.so.2
Using native(wrapped) librt.so.1
Using emulated /lib/x86_64-linux-gnu/libstdc++.so.6
Using native(wrapped) libm.so.6
Using emulated /lib/x86_64-linux-gnu/libgcc_s.so.1
SG driver version 3.5.36.

Adapter ID:                          0
Product:                             1b4b-9230
Sub Product:                         1b4b-9230
Chip revision:                       A1
Max PCIe speed:                      5Gb/s
Current PCIe speed:                  5Gb/s
Max PCIe link:                       2
Current PCIe link:                   1
BIOS version:                        1.0.0.1024
Firmware version:                    2.3.0.1063
Boot loader version:                 2.1.0.1008
# of ports:                          4
Buzzer:                              Not supported
Supported port type:                 SATA
Supported RAID mode:                 RAID0 RAID1 RAID10 hc(hypper capacity) hs(hypper safe)
Maximum SSD in one HyperDuo:         3
Maximum SSD Segment:                 3
Maximum disk in one VD:              4
PM:                                  Supported
Expander:                            Not supported
Maximum supported disk:              7
Maximum supported VD:                4
Max total blocks:                    128
Features:                            rebuild
Advanced features:                   event sense code,multi VD,spc 4,image health,phy ctrl,timer,smart poll,bga rate,ata pass through,remap,access register
Max buffer size:                     3
Stripe size supported:               32K 64K
Image health:                        Healthy
Autoload image health:               Healthy
Boot loader image health:            Healthy
Firmware image health:               Healthy
Boot ROM image health:               Healthy
HBA info image health:               Healthy
```
Any ideas or suggestions?

BenSeventy9, 2022-09-03:
Found a newer firmware(2.3.0.1076): https://www.station-drivers.com/index.php/en/component/remository/Drivers/Marvell/SATA/MV9xxx-(88SE91xx-92xx)/bios/Marvell-MV-92xx-SATA-6G-Controller-Firmware-Version-2.3.0.1076/lang,en-gb/
```
 $ sudo mvcli info -o hba
Dynarec for ARM64, with extension: ASIMD CRC32 PageSize:4096
Box64 with Dynarec v0.1.9 b185afa built on Sep  2 2022 18:22:00
Using default BOX64_LD_LIBRARY_PATH: ./:lib/:lib64/:x86_64/:bin64/:libs64/
Using default BOX64_PATH: ./:bin/
Counted 16 Env var
Looking for /usr/local/bin/mvcli
argv[1]="info"
argv[2]="-o"
argv[3]="hba"
Rename process to "mvcli"
Using emulated libmvraid.so
Using native(wrapped) libpthread.so.0
Using native(wrapped) libdl.so.2
Using native(wrapped) libc.so.6
Using native(wrapped) ld-linux-x86-64.so.2
Using native(wrapped) librt.so.1
Using emulated /lib/x86_64-linux-gnu/libstdc++.so.6
Using native(wrapped) libm.so.6
Using emulated /lib/x86_64-linux-gnu/libgcc_s.so.1
sh: 1: cannot create /tmp/vgname: Permission denied
sh: 1: cannot create /tmp/pvinfo: Permission denied
SG driver version 3.5.36.

Adapter ID:                          0
Product:                             1b4b-9230
Sub Product:                         1b4b-9230
Chip revision:                       A1
slot number:                         0
Max PCIe speed:                      5Gb/s
Current PCIe speed:                  5Gb/s
Max PCIe link:                       2
Current PCIe link:                   1
BIOS version:                        1.0.0.1027
Firmware version:                    2.3.0.1076
Boot loader version:                 2.1.0.1008
# of ports:                          4
Buzzer:                              Not supported
Supported port type:                 SATA
Supported RAID mode:                 RAID0 RAID1 RAID10 hc(hypper capacity) hs(hypper safe)
Maximum SSD in one HyperDuo:         3
Maximum SSD Segment:                 3
Maximum disk in one VD:              4
PM:                                  Supported
Expander:                            Not supported
Rebuild:                             Supported
Background init:                     Not supported
Sync:                                Not supported
Migrate:                             Not supported
Media patrol:                        Not supported
Foreground init:                     Not supported
Copy back:                           Not supported
Maximum supported disk:              7
Maximum supported VD:                4
Max total blocks:                    128
Features:                            rebuild
Advanced features:                   event sense code,multi VD,spc 4,image health,phy ctrl,timer,smart poll,bga rate,ata pass through,remap,access register
Max buffer size:                     3
Stripe size supported:               32K 64K
Image health:                        Healthy
Autoload image health:               Healthy
Boot loader image health:            Healthy
Firmware image health:               Healthy
Boot ROM image health:               Healthy
HBA info image health:               Healthy
```
Found newer CLI (4.1.13.31): https://www.dell.com/support/home/de-ch/drivers/driversdetails?driverid=67hr5
```
$ sudo mvcli
Dynarec for ARM64, with extension: ASIMD CRC32 PageSize:4096
Box64 with Dynarec v0.1.9 b185afa built on Sep  2 2022 18:22:00
Using default BOX64_LD_LIBRARY_PATH: ./:lib/:lib64/:x86_64/:bin64/:libs64/
Using default BOX64_PATH: ./:bin/
Counted 16 Env var
Looking for /usr/local/bin/mvcli
Rename process to "mvcli"
Using emulated libmvraid.so
Using native(wrapped) libpthread.so.0
Using native(wrapped) libdl.so.2
Using native(wrapped) libc.so.6
Using native(wrapped) ld-linux-x86-64.so.2
Using native(wrapped) librt.so.1
Using emulated /lib/x86_64-linux-gnu/libstdc++.so.6
Using native(wrapped) libm.so.6
Using emulated /lib/x86_64-linux-gnu/libgcc_s.so.1
sh: 1: cannot create /tmp/vgname: Permission denied
sh: 1: cannot create /tmp/pvinfo: Permission denied
SG driver version 3.5.36.
CLI Version: 4.1.13.31   RaidAPI Version: 5.0.13.1071
Welcome to RAID Command Line Interface.

>
```

paulwratt, 2022-09-04:
looks like that newer FW and CLI dont know about "HyperDuo" by that output

BenSeventy9, 2022-09-04:
You are right. I may have mixed some links. Here is an other Lenovo link to mvcli: https://datacentersupport.lenovo.com/nz/en/downloads/DS547073

```
----------------------------------------------------------------------
1.1 Usage information
----------------------------------------------------------------------

1. chmod 777 install.sh & mvcli
2. ./install.sh

root@loclhost/ mvcli help
```

BenSeventy9, 2022-09-08:
Detailed instructions are now here: https://github.com/BenSeventy9/rpi_cm4_mvcli

paulwratt, 2022-09-11:
ok, nice to see you use
`lnvgy_utl_bootstor_mvcli-2.3.10.1095-0_linux_x86-64.tgz`
along with
`mvcli-4.1.13.31_A01.zip`

I presume those installer scripts were the ones you were having problems with before you posted those instructions just now ..

BenSeventy9, 2022-09-11:
> ok, nice to see you use `lnvgy_utl_bootstor_mvcli-2.3.10.1095-0_linux_x86-64.tgz` along with `mvcli-4.1.13.31_A01.zip`
> 
> I presume that `install.sh` script you were having problems with, was before you posted those instructions just now with multiple update scripts?

The install Script isn't the problem. Box64 isn't locking correctly for the needed library.

paulwratt, 2022-09-11:
but you fixed that with `BOX64_LD_LIBRARY_PATH=/lib` right?

If you have to do that I'd move it to a /lib/x86_64 directory, which might actually be where its looking for it. Actually there is cli tool to check that (libtool?)

BenSeventy9, 2022-09-11:
> but you fixed that with `BOX64_LD_LIBRARY_PATH=/lib` right?

Yes
> If you have to do that I'd move it to a `/lib/x86_64` directory, which might actually be where its looking for it.

I have tried that. Placing it in ``/lib``, ``/lib64``, ``/x86_64``, ``/bin64`` or ``/libs64`` doesn't matter.

I think it is an issue with box64. Because it is allready locking for that directory:``Using default BOX64_LD_LIBRARY_PATH: ./:lib/:lib64/:x86_64/:bin64/:libs64/``
These paths have no leading ``/``
I think it should be:``Using default BOX64_LD_LIBRARY_PATH: ./:/lib:/lib64:/x86_64:/bin64:/libs64``

I have proven that by this simple test:
- Copy ``mvcli`` into ``/bin``
- Create a random directory for example in your home directory ``myrandomfolder``.
- Go in that directory and create an another directory named ``lib`` (or ``lib64``, ``x86_64``, ``bin64`` or ``libs64`` doesn't matter) and copy ``libmvraid.so`` into it.
- Now run ``sudo mvcli`` (in your random directory ``myrandomdirectory``)

What a surprise: Box64 is finding your ``libmvraid.so`` in your random directory! That means box64 is locking for ``$PWD/lib`` (or ``lib64``, ``x86_64``, ``bin64`` or ``libs64`` doesn't matter). And that is why adding ``BOX64_LD_LIBRARY_PATH=/lib`` is fine.

BenSeventy9, 2022-10-05:
## Detailed instructions on my GitHub

**My new solution so you don't need installing Box64!**
**Run mvcli in a docker container:** https://github.com/BenSeventy9/rpi-cm4-mvcli-docker

**Instructions based on this discussion:** https://github.com/BenSeventy9/rpi-cm4-mvcli

Tony763, 2022-10-05:
Hi, @BeauSlim URL to docker repo does not work, isn't it set as personal only?

BenSeventy9, 2022-10-05:
> Hi, @BeauSlim URL to docker repo does not work, isn't it set as personal only?

Forgot to change visibility