geerlingguy/raspberry-pi-pcie-devices #292: Testing the SOQuartz / Model-A hostboard

rtreffer, 2021-10-30:
The SOQuartz is a CM4 compatible board (see [the announcement](https://www.pine64.org/2021/06/15/june-update-new-hardware-and-more-on-the-way/)) that should become [available next week](https://www.pine64.org/2021/10/29/october-update-follow-up/) (`The SOquartz module will be available for purchase in the Pine Store next week`).

This board is based on the [RK3566](http://www.boardcon.com/download/Rockchip_RK3566_Datasheet_V1.1.pdf). It might (or might not) work better with the PCIe devices listed.

Is there any interest in rerunning tests with this device?

In related news: I would expect their `Model-A hostboard` to work with the CM4, too. It's a very compact board ([blog post image](https://www.pine64.org/wp-content/uploads/2021/10/photo_2021-10-29_07-56-16.jpg)) optimized for PCIe cards. It should work very well for PCIe testing.

geerlingguy, 2021-10-31:
I would absolutely love to get my hands on a SOQuartz... do you know if it'll be available for sale in the US next week or just certain countries?

I actually have a little Radxa-built CM4-compatible (but slightly upgraded, hardware-wise) module that I'm testing too. It would be interesting to compare the three head to head. And see if any of them solve the PCIe issues I've had on the Pi ;)

rtreffer, 2021-11-07:
They are online now: https://pine64.com/product-category/soquartz/

geerlingguy, 2021-11-08:
I just saw that, and ordered a 2GB model.

What's odd is I only found out about it in roundabout ways. I went to the main site where I look for Pine64 stuff, https://www.pine64.org, and there's not even a product page for the SOQuartz!

rtreffer, 2021-11-09:
Yeah, they released the Pinephone Pro and their ePaper system and are super busy overall right now (as far as I know). I got to know about it through their discord (bridged to whatever else you might prefer - IRC, matrix,....).

They seem to be mostly a HW engineering - software is up to others/community work, marketing/website/community seems to be a one man show etc. They call them selves a community service - mostly getting HW into the hands of developers at a nice price tag.

But they usually deliver on the HW side. I got the original A64 off Kickstarter. I put it aside years ago and revived it recently just to find that all the software stuff is rock solid now. Similar experience for a RockPro64, software/kernel got better every year or so.

It seems like they have ~3 SOC lines. The original A64 (powering an SBC, PinePhone, Module, Pinebook), the RockPro line (again whole set of form factors) and now the Quartz line. So the whole SOC got already some 6 months of software bring up.

We will see what that means, I ordered the 8GB one ... I have high hopes that the PCIe support is better than on the CM4 and I hope to try 2-3 graphics cards to verify if they could work on ARM64 at all / if it's a CM4 issue.

rtreffer, 2021-11-15:
Ohh, they are planning a blade Hostboard.

https://www.pine64.org/2021/11/15/november-update-first-impressions/

I would expect that it works for the CM4 too.

geerlingguy, 2022-04-18:
Closing this issue in favor of https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/336

closed 2022-04-18 by geerlingguy