geerlingguy/raspberry-pi-pcie-devices #293: Add 52Pi CM4 Router Board

geerlingguy, 2021-11-09:
S2Pi's [CM4 Router Board](https://www.seeedstudio.com/CM4-Router-Board-p-5211.html) is an expansion board based on the Raspberry Pi Compute Module 4. It has two full-speed gigabit network ports and offers better performance, lower CPU usage, and higher stability for a long time work compared with a USB network card. Designed especially for OpenWRT.
[image]

closed 2021-11-11 by geerlingguy

paulwratt, 2022-01-15:
S2Pi's what? ;)

geerlingguy, 2022-01-17:
@paulwratt - Oopsie! Got it fixed.

Honestly until this year I always thought it was S2 Pi, because of the stylized logo.