geerlingguy/raspberry-pi-pcie-devices #295: Add MaxLinear (formerly Exar) XR17V358/SP339-0A-EB (evaluation board for XR17V358-0A eight port UART)

ObliqueSinter, 2021-11-10:
Had one of these evaluation boards in inventory, decided to see if it would work with the CM4 and IO board:

[image]
[image]

Sadly, not plug and play; no new TTY devices showed up in /dev.

lspci -vvk
`01:00.0 Serial controller: Exar Corp. Device 0358 (rev 03) (prog-if 02 [16550])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 65
        Region 0: Memory at 600000000 (32-bit, non-prefetchable) [size=16K]
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
                Address: 00000000fffffffc  Data: 6540
        Capabilities: [78] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] Express (v2) Endpoint, MSI 01
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                LnkCap: Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 <1us
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR-, OBFF Not Supported
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Virtual Channel
                Caps:   LPEVC=0 RefClk=100ns PATEntryBits=1
                Arb:    Fixed- WRR32- WRR64- WRR128-
                Ctrl:   ArbSelect=Fixed
                Status: InProgress-
                VC0:    Caps:   PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
                        Arb:    Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
                        Ctrl:   Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
                        Status: NegoPending- InProgress-
        Kernel driver in use: exar_serial`

Relevant(?) dmesg output:
`[    1.164551] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.164586] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.164667] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.164825] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x003fffffff -> 0x0400000000
[    1.196890] brcm-pcie fd500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.197302] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.197324] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.197345] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.197439] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.197685] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.201239] PCI: bus0: Fast back to back transfers disabled
[    1.201264] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.201562] pci 0000:01:00.0: [13a8:0358] type 00 class 0x070002
[    1.201624] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00003fff]
[    1.201942] pci 0000:01:00.0: PME# supported from D0 D3hot
[    1.205403] PCI: bus1: Fast back to back transfers disabled
[    1.205426] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.205506] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.205529] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600003fff]
[    1.205557] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.205583] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    1.206024] pcieport 0000:00:00.0: enabling device (0140 -> 0142)
[    1.206245] pcieport 0000:00:00.0: PME: Signaling with IRQ 64
[    1.213336] exar_serial 0000:01:00.0: enabling device (0140 -> 0142)
[    1.213823] exar_serial 0000:01:00.0: Couldn't register serial port 0, irq 65, type 2, error -28`

If anybody has ideas on what to try, I can run tests as needed.

paulwratt, 2021-11-15:
if this was fully populated (I presume the final releast is) it would be truely epic, one RPi to rule them all :)

Do you have anything (incl. CM4 boards) with RS-232 - there should be some linux software that can talk to (or display) multiple RS-232 ports. Might be the sort of thing to put on the "slow burner" to build up enough hardware and some idea on how to use it (for testing purposes anyway)

geerlingguy, 2022-04-21:
It's on the site now: https://pipci.jeffgeerling.com/cards_other/maxlinear-evaluation.html

closed 2022-04-21 by geerlingguy

xrysf03, 2024-02-15:
I'd like to respond to the message:
exar_serial 0000:01:00.0: Couldn't register serial port 0, irq 65, type 2, error -28
To get past this error message, you may want to add   8250.nr_uarts=32   to your kernel command line.
After that, you probably won't be able to grep "exar" from your dmesg anymore, i.e. this message will vanish, but you will see a couple messages where /dev/ttyS<up_and_up> get detected.
More comments on that kernel command line candy is [available in my old description](http://support.fccps.cz/download/adv/frr/pci-16xx/README.pci-16xx.html) for unrelated multiport serial hardware (the kernel configuration option is generic).