geerlingguy/raspberry-pi-pcie-devices #297: Add Vantec UGT-FW210 (Texas Instruments XIO2213B)

ObliqueSinter, 2021-11-11:
Doesn't work out of the box. I suspect the chance of getting it working is low, given that it has and uses an embedded PCIe-to-PCI bridge.
(IIRC these bridges typically use the I/O BAR which is unsupported on ARM?)

Photo from Vantec website:
[image]

lspci -vvk
```
01:00.0 PCI bridge: Texas Instruments XIO2213A/B/XIO2221 PCI Express to PCI Bridge [Cheetah Express] (rev 01) (prog-if 00 [Normal decode])
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Region 1: Memory at 600100000 (32-bit, non-prefetchable) [disabled] [size=4K]
	Bus: primary=01, secondary=02, subordinate=02, sec-latency=0
	Memory behind bridge: c0000000-c00fffff
	Secondary status: 66MHz+ FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity+ SERR+ NoISA- VGA- MAbort+ >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
		Bridge: PM- B3+
	Capabilities: [60] MSI: Enable- Count=1/16 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [80] Subsystem: Device 3412:7856
	Capabilities: [90] Express (v1) PCI-Express to PCI/PCI-X Bridge, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ SlotPowerLimit 0.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+ BrConfRtry-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <1us, L1 <16us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-

02:00.0 FireWire (IEEE 1394): Texas Instruments XIO2213A/B/XIO2221 IEEE-1394b OHCI Controller [Cheetah Express] (rev 01) (prog-if 10 [OHCI])
	Subsystem: Device 3412:7856
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz+ UDF- FastB2B- ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 255
	Region 0: Memory at 600004000 (32-bit, non-prefetchable) [disabled] [size=2K]
	Region 1: Memory at 600000000 (32-bit, non-prefetchable) [disabled] [size=16K]
	Capabilities: [44] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
```
Note the bridge as well as the controller.

Relevant dmesg output:
```
[    1.164586] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.164619] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.164699] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.164787] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x003fffffff -> 0x0400000000
[    1.196964] brcm-pcie fd500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.197497] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.197519] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.197541] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    1.197647] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.197896] pci 0000:00:00.0: PME# supported from D0 D3hot
[    1.201455] PCI: bus0: Fast back to back transfers disabled
[    1.201481] pci 0000:00:00.0: bridge configuration invalid ([bus ff-ff]), reconfiguring
[    1.201789] pci 0000:01:00.0: [104c:823e] type 01 class 0x060400
[    1.201870] pci 0000:01:00.0: reg 0x14: [mem 0x00000000-0x00000fff]
[    1.202151] pci 0000:01:00.0: supports D1 D2
[    1.205582] PCI: bus1: Fast back to back transfers disabled
[    1.205608] pci 0000:01:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.205796] pci_bus 0000:02: extended config space not accessible
[    1.205977] pci 0000:02:00.0: [104c:823f] type 00 class 0x0c0010
[    1.206042] pci 0000:02:00.0: reg 0x10: [mem 0x00000000-0x000007ff]
[    1.206081] pci 0000:02:00.0: reg 0x14: [mem 0x00000000-0x00003fff]
[    1.206307] pci 0000:02:00.0: supports D1 D2
[    1.206324] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot
[    1.209747] PCI: bus2: Fast back to back transfers disabled
[    1.209769] pci_bus 0000:02: busn_res: [bus 02-ff] end is updated to 02
[    1.209799] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 02
[    1.209846] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6001fffff]
[    1.209870] pci 0000:01:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    1.209890] pci 0000:01:00.0: BAR 1: assigned [mem 0x600100000-0x600100fff]
[    1.209921] pci 0000:02:00.0: BAR 1: assigned [mem 0x600000000-0x600003fff]
[    1.209948] pci 0000:02:00.0: BAR 0: assigned [mem 0x600004000-0x6000047ff]
[    1.209977] pci 0000:01:00.0: PCI bridge to [bus 02]
[    1.210007] pci 0000:01:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    1.210048] pci 0000:00:00.0: PCI bridge to [bus 01-02]
[    1.210073] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6001fffff]
[    1.210508] pcieport 0000:00:00.0: enabling device (0140 -> 0142)
[    1.210754] pcieport 0000:00:00.0: PME: Signaling with IRQ 64
```

Tried plugging in an Apple iSight camera (A1023), only one line showed up in dmesg:
`[  185.989047] v3d fec00000.v3d: MMU error from client L2T (0) at 0x6961000, pte invalid`

No new devices in /dev, with the camera attached or otherwise.

paulwratt, 2021-11-15:
BAR I/O: not unsupported, just limited and quirky in use/implimentation

(see various graphics threads/issues to see more on BAR I/O)

ObliqueSinter, 2021-11-25:
Got kernel cross compiling working, enabled Firewire drivers:
`Device Drivers -> IEEE 1394 (FireWire) support-> FireWire driver stack -> OHCI-1394 controllers, Storage devices (SBP-2 protocol), IP networking over 1394`
No dice. Card still shows up in `lspci` but no drivers are attached.

geerlingguy, 2021-12-12:
I went ahead and added the card to the site as 'None' functionality, I'll go ahead and close out this issue.

closed 2021-12-12 by geerlingguy

t5b6de, 2022-05-05:
Sorry, i know it is closed BUT:

i have a firewire 400 card with via vt6307 Chipset and thats a native PCI card, but this one has a asm1083 on it.

with 32 bit dma forced and ohci drivers build with the kernel it is working on a pi, flawless.

I'm currently transferring my miniDV Videos from my old video camera that is working without any issues. 
So other cards do work!

dmesg output:
[dmesg ohci firewire.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/8635488/dmesg.ohci.firewire.txt)

And this is lspci -vv output:
[lspci-vv.txt](https://github.com/geerlingguy/raspberry-pi-pcie-devices/files/8635499/lspci-vv.txt)

its incredible what's working on the pi already.

My videos from my Camera are perfecly fine on my hdd now without any issues. It worked out of the box after compiling kernel with firewire support.

unfortunately this is the rev1 of the asm1083 chipset, which has a bug: the interrupt is not reset after handled. So the linux kernel will disable the interrupt and is using irqpoll for further handling.

kind regards, and sorry for answering on a closed issue.

WACOMalt, 2022-05-14:
> 

Thank you for updating with this info. I'm currently attempting to get the same setup working but am stumbling on a few points. I purchased this card: https://www.ebay.com/itm/175158502561 Which appears to have the same chipsets you mentioned in your post, though it is a PCIe card directly. Upon testing dvgrab I am getting the usual "Error: no camera exists" with no extra info. I'm sure the main issue I'm hitting is my lack of knowlege in adding the "ohci drivers" to the kernel. I'll continue testing and check in if I have any success. Getting this working is paramount in a project I'm currently building, so I appreciate the info you shared in your previous reply!

geerlingguy, 2022-05-19:
@WACOMalt / @t5b6de - Can you follow up with any more info in this new issue I created for that card and the chipset it uses? https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/443

tbords, 2024-12-07:
Hey all,

I am struggling to get the firewire-ochi module onto my Raspberry Pi CM4. Is there any chance you have this kernel available? Or any tips?