geerlingguy/raspberry-pi-pcie-devices #303: Add will whang's CM4 Pi 4 Cluster HAT

geerlingguy, 2021-11-12:
@will127534 is back with a [CM4 Pi 4 Cluster HAT](https://twitter.com/will_whang/status/1459080800314425345) which is by far the most compact and powerful Pi cluster on the planet, packing four Compute Module 4s onto a HAT the size of a Pi 4 model B, and it even throws in a Raspberry Pi Pico for management tasks!

[image]

It uses a built-in Gigabit network switch to provide networking to all the CM4s, and you can find more details in [@will_whang's introductory Twitter thread](https://twitter.com/will_whang/status/1459080800314425345).

will127534, 2021-11-28:
Here is the [Github repo](https://github.com/will127534/Miniature-CM4-Cluster/) and  [Github Page](https://will127534.github.io/Miniature-CM4-Cluster/)!

Thanks @geerlingguy for adding this to the list!

geerlingguy, 2021-11-29:
@punnypenguins - Could you update the PR with those links? Thanks!

punnypenguins, 2021-11-29:
@geerlingguy Done!

closed 2021-12-12 by geerlingguy