geerlingguy/raspberry-pi-pcie-devices #313: Test Compex WLE200NX 802.11n 2x2 mini PCIe WiFi card

geerlingguy, 2021-11-17:
Alftel just shipped me a Seaberry board to test, and it came with a few PCI Express devices pre-installed.

One such device is the [Compex WLE200NX](https://amzn.to/3cigkI9), a WiFi 802.11n mini PCIe card.

TODO: Picture here.

I'll be testing it and will see if I can get a driver working on the CM4.

geerlingguy, 2021-11-17:
```
$ sudo lspci -vvvv
0b:00.0 Network controller: Qualcomm Atheros AR928X Wireless Network Adapter (PCI-Express) (rev 01)
	Subsystem: Qualcomm Atheros AR928X Wireless Network Adapter (PCI-Express)
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 66
	Region 0: Memory at 600200000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: [40] Power Management version 2
		Flags: PMEClk- DSI- D1+ D2- AuxCurrent=375mA PME(D0+,D1+,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit-
		Address: 00000000  Data: 0000
	Capabilities: [60] Express (v1) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <512ns, L1 <64us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset-
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L1, Exit Latency L1 <64us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 128 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
	Capabilities: [90] MSI-X: Enable- Count=1 Masked-
		Vector table: BAR=0 offset=00000000
		PBA: BAR=0 offset=00000000
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [140 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [160 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Kernel modules: ath9k
```

geerlingguy, 2021-11-17:
Interesting that it sees `ath9k`. But according to dmesg, something is not enabling:

```
pi@seaberry:~ $ dmesg | grep ath
[    3.539876] systemd[1]: Reached target Paths.
[    5.817290] ath9k 0000:0b:00.0: enabling device (0000 -> 0002)
[    7.417223] ath: EEPROM regdomain sanitized
[    7.417238] ath: EEPROM regdomain: 0x64
[    7.417242] ath: EEPROM indicates we should expect a direct regpair map
[    7.417253] ath: Country alpha2 being used: 00
[    7.417257] ath: Regpair used: 0x64
[    7.417514] ath: phy0: Failed to allocate tx descriptors: -12
[    7.417562] ath9k 0000:0b:00.0: Failed to initialize device
[    7.417662] ath9k: probe of 0000:0b:00.0 failed with error -12
```

Maybe needs a country code set?

geerlingguy, 2021-11-18:
```
pi@seaberry:~ $ iw reg get
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

I ran `sudo raspi-config`, went to Localization, and set the WiFi country to `US`, and now:

```
pi@seaberry:~ $ iw reg get
global
country US: DFS-FCC
	(2400 - 2483 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A)
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

Rebooted and checked again:

```
[   22.772037] ath: Country alpha2 being used: 00
```

Not sure why it's still unset, but anyways, in relation to the tx descriptors, I found:

  - https://github.com/GalliumOS/galliumos-distro/issues/253
  - https://forums.raspberrypi.com/viewtopic.php?t=306166 (bug with the exact same chipset on CM4 IO Board...)
  - https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1927037 (Ubuntu for Pi bug report)

geerlingguy, 2021-11-18:
Aha, from [this comment](https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1927037/comments/9) in the Ubuntu bug report, I found the suggestion to add the following to `/boot/config.txt`:

```
dtoverlay=pcie-32bit-dma
```

There are some other older cards that needed the same overlay added.

For reference:

```
Name:   pcie-32bit-dma
Info:   Force PCIe config to support 32bit DMA addresses at the expense of
        having to bounce buffers.
Load:   dtoverlay=pcie-32bit-dma
Params: <None>
```

geerlingguy, 2021-11-18:
To test the card is working, I added the following to `/boot/config.txt` to disable the Pi's built-in WiFi:

```
dtoverlay=disable-wifi
```

Then after a reboot, only one WLAN interface showed up.

To connect to wifi, you can use `sudo raspi-config` or just add the following in `/etc/wpa_supplicant/wpa_supplicant.conf`:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
        ssid="my-network-ssid"
        psk="my-password"
}
```

To get a list of visible SSIDs:

```
sudo iwlist wlan0 scan | grep SSID
```

geerlingguy, 2021-11-18:
Well... I'm getting 70/70 signal to my home WiFi, and have the right credentials, but I'm seemingly stuck in a loop here:

```
[   20.547443] wlan0: authenticate with 3c:7c:3f:6a:fa:c4
[   20.568500] wlan0: send auth to 3c:7c:3f:6a:fa:c4 (try 1/3)
[   21.822322] wlan0: send auth to 3c:7c:3f:6a:fa:c4 (try 2/3)
[   22.814323] wlan0: send auth to 3c:7c:3f:6a:fa:c4 (try 3/3)
[   23.806324] wlan0: authentication with 3c:7c:3f:6a:fa:c4 timed out
[   27.056332] wlan0: authenticate with 3c:7c:3f:6a:fa:c4
[   27.076820] wlan0: send auth to 3c:7c:3f:6a:fa:c4 (try 1/3)
[   30.297474] wlan0: authenticated
[   30.338349] wlan0: associate with 3c:7c:3f:6a:fa:c4 (try 1/3)
[   31.806401] wlan0: associate with 3c:7c:3f:6a:fa:c4 (try 2/3)
[   32.830374] wlan0: associate with 3c:7c:3f:6a:fa:c4 (try 3/3)
[   33.822359] wlan0: association with 3c:7c:3f:6a:fa:c4 timed out
```

geerlingguy, 2021-11-18:
Not sure why it won't connect—I've tried my 2.4 GHz network and 5 GHz network, so I'll maybe just mark this card as 'maybe' working and move on...

RodrigoRosmaninho, 2024-01-25:
To anyone struggling with this issue on the new Raspberry Pi 5: The _pcie-32bit-dma_ dts overlay is only compatible with the CM4 and does not work on the Pi 5.
I have created a PR (https://github.com/raspberrypi/linux/pull/5897) to add a new _pcie-32bit-dma-overlay-pi5_ dts overlay that fixes this issue. 
I've confirmed that this works correctly on a Raspberry Pi 5 connected to Pineberry Pi's "Hat AI!", to which I plugged this wireless card and also a MikroTik R11e-5HnD (Atheros AR9580), using a M.2 E-key to Mini-PCIe adapter.