geerlingguy/raspberry-pi-pcie-devices #314: Test IOCrest mini PCIe dual SATA card (ASM1061 chipset)

geerlingguy, 2021-11-17:
Alftel just shipped me a Seaberry board to test, and it came with a few PCI Express devices pre-installed.

One such device is the [IOCrest mini PCIe dual SATA card](https://amzn.to/3oF77iz), which has an ASM1061 chip on it.

TODO: Picture here.

Hopefully it should just work 'out of the box' with the CM4 and a relatively recent version of Pi OS.

geerlingguy, 2021-11-17:
```
06:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 01) (prog-if 01 [AHCI 1.0])
	Subsystem: ASMedia Technology Inc. ASM1062 Serial ATA Controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 67
	Region 0: I/O ports at <unassigned> [disabled]
	Region 1: I/O ports at <unassigned> [disabled]
	Region 2: I/O ports at <unassigned> [disabled]
	Region 3: I/O ports at <unassigned> [disabled]
	Region 4: I/O ports at <unassigned> [disabled]
	Region 5: Memory at 600110000 (32-bit, non-prefetchable) [size=512]
	Expansion ROM at 600100000 [virtual] [disabled] [size=64K]
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit-
		Address: fffffffc  Data: 6540
	Capabilities: [78] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [80] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <1us, L1 <8us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 5GT/s, Width x1, ASPM not supported
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABC, TimeoutDis+ NROPrPrP- LTR-
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Kernel driver in use: ahci
	Kernel modules: ahci
```

Works great!

closed 2021-11-17 by geerlingguy

geerlingguy, 2021-11-18:
Just leaving a note that I was able to test a Crucial 2TB SSD (MX500), and here are the results:

Running my [disk-benchmark.sh](https://github.com/geerlingguy/raspberry-pi-dramble/blob/master/setup/benchmarks/disk-benchmark.sh):

```
pi@taco:~ $ wget https://raw.githubusercontent.com/geerlingguy/raspberry-pi-dramble/master/setup/benchmarks/disk-benchmark.sh
pi@taco:~ $ chmod +x disk-benchmark.sh 
pi@taco:~ $ nano disk-benchmark.sh
pi@taco:~ $ sudo DEVICE_UNDER_TEST=/dev/sda1 DEVICE_MOUNT_PATH=/mnt/ssd ./disk-benchmark.sh
```

Results:

| Benchmark | Result |
| --- | --- |
| fio 1M sequential read | 398 MB/s |
| iozone 1M random read | 326 MB/s |
| iozone 1M random write | 353 MB/s |
| iozone 4K random read | 27.43 MB/s |
| iozone 4K random write | 56.76 MB/s |

czosnkowy666, 2023-07-24:
small question about this card, how you connected it with CM4 & IO Board?

commit e3b0a6ba (2021-11-17): Fixes #314: dual SATA board works great