geerlingguy/raspberry-pi-pcie-devices #327: Test Radxa CM3 on CM4 Boards

geerlingguy, 2021-11-18:
I have a Radxa CM3, which has a RockChip RK3566. I want to test it with some things, for science. This issue will be a compendium of all my knowledge.

geerlingguy, 2021-11-18:
First up, getting the thing to boot (either to eMMC flashing mode, or to boot normally) is... interesting.

## Booting with CM4 IO Board

It looks like if you just supply power (e.g. plug in the board), it won't boot by itself. The process for boot seems to be:

  1. Plug in power
  2. Put a jumper on pins 13-14 on the 14-pin header (GLOBAL_EN).

## Getting into eMMC mass storage mode with CM4 IO Board

Similarly, to flash the built-in eMMC through mass storage / OTG mode, I have to perform a little bit of a dance:

  1. Plug in power
  2. While depressing the little gold button between the SoC and WiFi chip...
  3. Put a jumper on pins 13-14 on the 14-pin header (GLOBAL_EN).

At this point, if I run `lsusb` I can see a new device appear:

```
Bus 002 Device 004: ID 2207:350a Fuzhou Rockchip Electronics Co., Ltd. Composite Device
```

geerlingguy, 2021-11-18:
## Flashing a new OS to the CM3

  1. Download an image from the [rock-3-images-released repo](https://github.com/radxa/rock-3-images-released/releases).
  2. TODO.

geerlingguy, 2021-11-19:
Right now, no matter what I do, this thing boots to this desktop after I see a 'RockChip kernel' splash screen for a couple seconds:

[image]

robtech21, 2021-11-19:
Interesting, what are OS and desktop?

geerlingguy, 2021-11-19:
@robtech21 - No clue, this was an earlier board that someone else was testing with, so it seems to be something kinda custom they were working on. I can't get USB to work and can't log in over the network, so I'm just trying to figure out how to re-flash the whole thing so it doesn't boot to this.

paulwratt, 2021-11-20:
with the lack of USB I summize there is debug output somewhere

the desktop looks like android with a menu bar, maybe try with a touch screen?

robtech21, 2021-11-22:
> with the lack of USB I summize there is debug output somewhere
> 
> the desktop looks like android with a menu bar, maybe try with a touch screen?

The UI kinda reminds me of an XFCE interface, but maybe that's just me going slowly insane after using it for a long time. Radxa does offer Debian, Ubuntu, and Android iirc

geerlingguy, 2021-12-05:
FYI I'm also beginning to test the SOQuartz: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/336

geerlingguy, 2021-12-09:
Testing this with a new set of instructions from Discord:

1. Plug in 12V barrel plug power.
2. Hold down the button next to the WiFi shield AND the button next to the SPI flash chip.
3. Place jumper across 13/14 (GLOBAL_EN) on CM4 IO Board.
4. Release buttons
5. Run `rkdeveloptool ld` (should get `DevNo=1  Vid=0x2207,Pid=0x350a,LocationID=202	Maskrom`).
6. Run `rkdeveloptool db rk356x_spl_loader_ddr1056_v1.06.110.bin` (should get `Downloading bootloader succeeded.`)

geerlingguy, 2021-12-09:
Still nothing—it is still booting to that Qt looking custom desktop, and USB still doesn't work.

I tried [attaching to UART](https://www.jeffgeerling.com/blog/2021/attaching-raspberry-pis-serial-console-uart-debugging) through a USB-to-Serial adapter, and tried testing baud rates of `1500000`, `115200`, and `9600`, but I just get gibberish no matter what (but there is gibberish output during boot as if it's outputting something... just not sure what baud):

[image]

ProKn1fe, 2021-12-09:
What with cpu and memory performance vs CM4?

geerlingguy, 2021-12-09:
@ProKn1fe - I'll be covering that more later, for now go to the links in the original comment and it has all the specs. I am just trying to get the thing in a bootable state for now, before I can work on any kind of comparisons.

geerlingguy, 2021-12-09:
Ah, just realized I also have a spare USB-UART adapter that Broadcom sent when they sent a MegaRAID card for testing:

```
Bus 002 Device 024: ID 0403:6001 Future Technology Devices International Limited FT232R USB UART  Serial: AR0JTLV4
```

Unfortunately, it doesn't seem like that adapter worked either. I tried `screen /dev/tty.usbserial-AR0JTLV4 1500000` and got no output (though it would drop a question mark on each boot).

Here's the setup:

[image]

(Tested with both SOQuartz and Radxa. Also validated that everything worked at `115200` baud with a Pi, so I guess the old chip doesn't work with 1.5 Mbps serial data either.)

hipboi, 2021-12-10:
On macOS, if you want to get serial output at 150000, which is not a standard baud rate for macOS, the working usb to ttl chip is FT series(they provide universal driver for m1&intel), the working serial tool we tested is Coolterm(universal app).

https://wiki.radxa.com/Rock3/dev/serial-console#macOS

geerlingguy, 2021-12-10:
@hipboi - Thanks, didn't realize `screen` wouldn't support it, at least not the version that ships with macOS. I'll try CoolTerm soon and report back what happens.

hipboi, 2021-12-14:
update: picocom works with 1.5Mb baud rate

https://forum.radxa.com/t/macos-pi-4-e-s-working-1-5mbps-ttl-baudrate-with-picocom/8121

geerlingguy, 2021-12-14:
No clue how, but suddenly after booting it up again today, I can get the keyboard and mouse working! Things are looking much more positive after fighting with the board last week.

geerlingguy, 2021-12-14:
I also just wrote ubuntu focal to the eMMC using the same method (`rkdeveloptool wl 0 rockcm3_ubuntu_focal_server_arm64_20210914_0552-gpt.img`), and after rebooting, it still boots into the Rockchip Kernel buildroot OS. No clue how to bypass that.

hipboi, 2021-12-15:
@geerlingguy This is because the `rkdeveloptool` write the image to SPI flash, which is absolutely wrong but no error reported. We feedback this to Rockchip to fix it. At the meanwhile, if you want to write to eMMC, you will have to disable the SPI flash by press and hold the button besides it.

geerlingguy, 2021-12-15:
@hipboi - I did that too; I held both the SPI flash button _and_ the button next to the WiFi chip, and then booted it, and then wrote the image (tried with both Debian and Ubuntu), and it's still always booting into the rockchip buildroot kernel environment. Image for reference:

[image]

geerlingguy, 2021-12-15:
I just installed `CoolTerm` (`brew install --cask coolterm`), and will test it out later, too.

geerlingguy, 2021-12-15:
Good news! Even my cheap `CP2102` adapter sees to be able to handle 1.5 Mbps using CoolTerm — so I can finally get UART.

All right, so copying all the data from right after issuing `reboot now` to when the kernel started booting up: https://gist.github.com/geerlingguy/fafd85e7984b0a67f444778a6f620af4

Could this be the issue?

```
U-Boot SPL board init
U-Boot SPL 2017.09-ga1f6fc00a0-210413 #ldq (Apr 13 2021 - 11:35:00)
unknown raw ID phN
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC2
MMC error: The cmd index is 1, ret is -110
Card did not respond to voltage select!
mmc_init: -95, time 10
spl: mmc init failed with error: -95
Trying to boot from MMC1
SPL: A/B-slot: _a, successful: 0, tries-remain: 7
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256+ OK
## Checking uboot 0x00a00000 ... sha256+ OK
## Checking fdt 0x00b221f0 ... sha256+ OK
## Checking atf-2 0xfdcc9000 ... sha256+ OK
## Checking atf-3 0xfdcd0000 ... sha256+ OK
## Checking optee 0x08400000 ... sha256+ OK
Jumping to U-Boot(0x00a00000) via ARM Trusted Firmware(0x00040000)
Total: 220.547 ms
```

geerlingguy, 2021-12-15:
Also, I plugged in a SATA PCIe card, and it initially gets identified in the Rockchip kernel, but boot stalls out after a while with:

```
[   61.540550] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   61.540569] rcu: .2-...0: (1 GPs behind) idle=522/1/0x4000000000000000 softirq=60/61 fqs=5964 
[   61.540574] rcu: .(detected by 1, t=18002 jiffies, g=-1055, q=108)
[   61.540582] Task dump for CPU 2:
[   61.540587] rk-pcie         R  running task        0    54      2 0x0000002a
[   61.540596] Call trace:
[   61.540609]  __switch_to+0xc0/0x124
[   61.540616]  0x3
```

hipboi, 2021-12-16:
First, I am sorry for the confusion, the sample is a very early engineering sample, to write image to eMMC, you should do:

1. Hold both buttons to go to maskrom mode, `rkdeveloptool ld` should see maskrom
2. Now hold only the SPI flash button, and run `rkdeveloptool db rk356x_spl_loader_ddr1056_v1.06.110.bin`, so that rkdeveloptool will *not* write to SPI flash
3. After step 2, you can release SPI button now since the loader is running and it thinks there is no spi flash, now you can write to emmc with `rkdeveloptool wl 0 xxx.img`

hipboi, 2021-12-16:
Btw, we've fixed the picocom on M1. You can install from source with:

    brew install --build-from-source radxa/picocom/picocom

geerlingguy, 2021-12-16:
@hipboi - ah, I wasn't holding the SPI button during the `db` command. I'll try that and report back!

jcdutton, 2022-03-11:
Has anyone found device tree .dts files for this board?
There are multiple parts to the device tree.
1) The CM3 module itself.
2) The motherboard the CM3 is plugged into.  (this be will vary if one uses different boards with the module)
3) Optional device tree overlays - so one can switch features on/off

Once one has the correct device tree files, I would expect it to work ok with almost any arm64 linux kernel.
If you are playing with .dtb files, you are fighting a loosing battle.

hipboi, 2022-03-12:
dtsi for the cm3 module itself: https://github.com/radxa/kernel/blob/stable-4.19-rock3/arch/arm64/boot/dts/rockchip/rk3566-radxa-rock-3-compute-module.dtsi

dts for using on rpi cm4 io board: https://github.com/radxa/kernel/blob/stable-4.19-rock3/arch/arm64/boot/dts/rockchip/rk3566-radxa-cm3-rpi-cm4-io.dts

dts for using on radxa cm3 io board: https://github.com/radxa/kernel/blob/stable-4.19-rock3/arch/arm64/boot/dts/rockchip/rk3566-radxa-cm3-io.dts

jcdutton, 2022-03-12:
Looking at the logs posted by @geerlingguy 
It looks like the device tree is embedded in the linux binary image being used.
That was the old way of doing things, but it is generally better to have the linux binary image and the device tree .dtb file as separate files, that uboot can load.
The advantage of this is that one can then distribute a linux binary that can work on all cm3 modules, no matter what io board they are plugged into. The .dtb file can stay (matching the io board it is plugged into) the same on each linux binary update.
So, I would advise the manufacturer of the board to move to a separate linux binary image and .dtb file.
Now, for this particular boot, it is unclear which device tree file has been embedded into the linux binary image.
Is it actually the right one for the io board you have plugged the radxa cm3 module into?
One will have to use the radxa kernel, because it includes extra PCIe drivers that are not in Linux mainline yet.
I believe the correct PCIe drivers will be in kernel 5.18.
I also notice from the boot logs, that it is booting quite an old kernel image. You might want to use a more up to date one. It is booting "Linux version 4.19.172-01402-g0ae0afc2911a-dirty"  when a more up to date one for radxa is "4.19.193-10-rockchip-g152e34a538b4"

benivyhuen, 2022-07-28:
any progress？ very interesting project

jcdutton, 2022-08-06:
You should be able to edit the file:
rk3566-radxa-rock-3-compute-module.dtsi
rockchip,baudrate = <1500000>;  /* Only 115200 and 1500000 */
Just change the number to change the rate.

But looking at the logs you have got, it is still stuck in uboot, and not even found a Linux kernel to boot from yet.

For initially getting it to work, it is probably best to avoid PCIe devices. 

You should be able to short out 2 pins on the header to stop it booting from the SPI storage. My guess it that it is booting from that which is why you do not get the sdcard or emmc to boot.

Another alternative, if you escape it during the uboot part, you should be able to modify the uboot config to boot from something else.

I don't actually have a rock cm3, so the exact method to get it booting might be different.

gopromaster, 2023-01-23:
Has anybody managed to get the CM3 working? Thinking about getting one for a Radxa Taco NAS setup...

raphaelscholle, 2023-01-24:
I use the CM3, but it's quite limited because of lacking documentation and simply no hardware support to anything.
You can try my image, which is porpuse made for OpenHD (a digital video transmission system)
Just deactivate the openhd and qopenhd service and it should be mostly clean from junk

username and password are openhd

https://drive.google.com/file/d/1QRbU5oiU57-5uAG5HaoWUrlaPjNLeWj9/view?usp=share_link

RicoKirstein, 2023-03-03:
Hi, I stumbled upon this thread while trying to boot up a cm3 on the cm4io board. 
I finally got it working by flashing the image directly to the emmc memory, despite the wiki saying we can either boot from sd or internal memory. Does anyone know if we have to do something special to get it booting from sd card?
Or is it like the raspberry cm4 where you can only have either sdcard or emmc working?

leftbrainstrain, 2023-04-05:
I got the Radxa Taco working with the CM3 for a NAS, but it took a bit of work. The biggest issue right now, though, is availability.

The lack of documentation and design liberties can also be annoying at times. If you try booting the CM3 on non-Radxa boards, you'll find that you can't boot off of microSD because the CM3 is expecting one of the Reserved pins on the 100-pin connectors to be wired to the microSD card sense pin. On Radxa boards this is connected like the CM3 expects, so you can boot off of microSD cards on Radxa boards with the CM3.

After booting for the first time, you will probably want to update packages. You'll need to Add Radxa's key first to do this:

  wget -O -  apt.radxa.com/bullseye-stable/public.key | sudo apt-key add - 
  apt update && apt upgrade

You may also want to update your hostname, add a new user account, disable the GUI, etc. It's up to you, but in case you want to:

  hostnamectl set-hostname mynewhostname
  echo "127.0.1.1 mynewhostname" >> /etc/hosts
  
  systemctl get-default
  systemctl enable multi-user.target --force
  systemctl set-default multi-user.target

Next you'll want to see if your drives and the second network port are detected. You will find that nothing connected to PCIe is showing up to your system, however. To enable PCIe in your device tree, run these commands:

  echo "dtoverlay=radxa-cm3-io-pcie2x1" >> /boot/config.txt
  update_extlinux.sh
  reboot

If you'd like to use the fan included with the kit, and you can't get the instructions on the Taco page working, you can manually set the GPIO pin controlling the fan to high:

  cd /sys/class/gpio
  
  echo 22 > export
  cd gpio22
  
  echo out > direction
  echo 1 > value     # output high

Just keep in mind the above commands are manually run and won't run on every boot like the Radxa scripts should. I've included the above commands only because the URL of the .deb file resulted in a 404 error when I first got my Taco, then it worked but only for RPi CM4. It looks like it might work for Radxa CM3 now, but I haven't tried it. Unfortunately, I've not figured out a way to use PWM on the pin controlling the fan.

If you're looking for a case and have a 3D printer, there are 1 or 2 on Thingiverse and Printables. I just published one on Printables with ID 443909 that has some obvious flaws, but it's an option. Good luck have fun.

RicoKirstein, 2023-04-06:
Interestingly enough, the uboot, once flashed, will absolutely boot from the sd card on a cm4io board. Until the kernel loads, then it will revert back to reading from the internal flash.
I'm working on designing a custom board for a product  and this was just to evaluate the mipi-dsi interface for my application.
I had to do quite a bit custom dtc work to get everything working.
I also started on porting the dtc for the 6.12 kernel. I got wired network working for the cm4io board.
Somehow Radxa removed a lot from the dtc files compared to the 4.19 kernel and i had to revert back to get the dsi driver working correctly.
I could've went for any radxa carrier board, but the cm4-io just had much better availability where i live. 
also if you want to use a custom kernel, you have to change the file system of the boot partition, since dpkg will fail if it is exFAT or FAT formated.

Jamos1988, 2023-04-12:
Anybody had issues with the WiFi chip not working?
Also, does anybody know of a Bullseye distro available for this board?

geerlingguy, 2023-05-31:
See also makerbymistake's CM3 (eMMC) review: https://dphacks.com/2023/05/30/radxa-rock-cm3-review-all-you-need-to-know/

Andalite1999, 2023-06-05:
Hey so also using the exact same setup as @geerlingguy from https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/327#issuecomment-973477069.  My main issue is that everywhere I go to find a .img file to flash the OS (as outlined in step 1 at https://github.com/radxa/rock-3-images-released/releases) I keep finding .img followed by .bmap, .gz, and md5.txt.  I know that it’s based on the Radxa’s modifications, but I cannot figure out a way to whip them all up together to get a single .img file.  I downloaded a bunch of isos and converted them to .img but I can’t get the cm3 to boot.  How can I get the right OS file so that I only have to flash a single .img file?