geerlingguy/raspberry-pi-pcie-devices #331: Test AMD Radeon Pro WX2100 Graphics Adaptor

ObliqueSinter, 2021-11-25:
Using the stock `amdgpu` driver, the CM4 will fully boot, but HDMI output will be an empty screen with a blinking cursor.
System is still accessible via the UART/serial console and appears fully functional.
There is no output on any of DisplayPort connectors on the WX2100 card.

[image]

[image]

Output on `lspci -vvk`:
```
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Lexa XT [Radeon PRO WX 2100] (prog-if 00 [VGA controller])
	Subsystem: Dell Lexa XT [Radeon PRO WX 2100]
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 51
	Region 0: Memory at 600000000 (64-bit, prefetchable) [size=256M]
	Region 2: Memory at 610000000 (64-bit, prefetchable) [size=2M]
	Region 4: I/O ports at <unassigned> [disabled]
	Region 5: Memory at 618000000 (32-bit, non-prefetchable) [size=256K]
	Expansion ROM at 618040000 [disabled] [size=128K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0-,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [58] Express (v2) Legacy Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 8GT/s, Width x8, ASPM L1, Exit Latency L0s <64ns, L1 <1us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis-, LTR+, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [100 v1] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [200 v1] #15
	Capabilities: [270 v1] #19
	Capabilities: [2b0 v1] Address Translation Service (ATS)
		ATSCap:	Invalidate Queue Depth: 00
		ATSCtl:	Enable-, Smallest Translation Unit: 00
	Capabilities: [2c0 v1] Page Request Interface (PRI)
		PRICtl: Enable- Reset-
		PRISta: RF- UPRGI- Stopped+
		Page Request Capacity: 00000020, Page Request Allocation: 00000000
	Capabilities: [2d0 v1] Process Address Space ID (PASID)
		PASIDCap: Exec+ Priv+, Max PASID Width: 10
		PASIDCtl: Enable- Exec- Priv-
	Capabilities: [320 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [328 v1] Alternative Routing-ID Interpretation (ARI)
		ARICap:	MFVC- ACS-, Next Function: 1
[... 99 lines cut ...]
[    4.828189]  really_probe+0xf4/0x3e0
[    4.828198]  driver_probe_device+0x60/0xc0
[    4.828209]  device_driver_attach+0x7c/0x88
[    4.828218]  __driver_attach+0x60/0xe8
[    4.828228]  bus_for_each_dev+0x78/0xd0
[    4.828236]  driver_attach+0x2c/0x38
[    4.828246]  bus_add_driver+0x158/0x200
[    4.828254]  driver_register+0x6c/0x128
[    4.828266]  __pci_register_driver+0x4c/0x58
[    4.828625]  amdgpu_init+0x74/0x1000 [amdgpu]
[    4.828638]  do_one_initcall+0x4c/0x2d8
[    4.828648]  do_init_module+0x60/0x250
[    4.828656]  load_module+0x214c/0x27a0
[    4.828664]  __do_sys_finit_module+0xb8/0x100
[    4.828672]  __arm64_sys_finit_module+0x28/0x38
[    4.828685]  el0_svc_common.constprop.0+0x84/0x1f8
[    4.828695]  do_el0_svc_compat+0x24/0x50
[    4.828705]  el0_svc_compat+0x20/0x30
[    4.828714]  el0_sync_compat_handler+0x90/0x150
[    4.828722]  el0_sync_compat+0x184/0x1c0
[    4.828731] ---[ end trace 9b253281666531fb ]---
[    4.885532] amdgpu: probe of 0000:01:00.0 failed with error -2
```

Note that `5.10.81-v8-OBSINDEV` is the name of the kernel I compiled.

6by9, 2021-11-25:
```
[    4.825127] amdgpu 0000:01:00.0: Direct firmware load for amdgpu/polaris12_mc.bin failed with error -2
[    4.825152] amdgpu: mc: Failed to load firmware "amdgpu/polaris12_mc.bin"
[    4.825578] [drm:gmc_v8_0_sw_init [amdgpu]] *ERROR* Failed to load mc firmware!
```
It should be in the linux-firmware package, or otherwise is available from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/amdgpu. Copy it to /lib/firmware/amdgpu.

ObliqueSinter, 2021-11-27:
> It should be in the linux-firmware package, or otherwise is available from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/amdgpu. Copy it to /lib/firmware/amdgpu.

Oops, I missed that step, thanks!

Still no joy, though. All screens blank, serial console non-responsive.
Following is serial console output, from power on to freeze:
```recover4.elf not found (6)
recovery.elf not found (6)
Read start4.elf bytes  2241504 hnd 0x0000e984 hash '61557506bbcf746b'
Read fixup4.dat bytes     5411 hnd 0x0000041e hash 'ce9c0be09a40d74a'
0x00a03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0

MESS:00:00:04.673626:0: arasan: arasan_emmc_open
MESS:00:00:04.861663:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.865061:0: brfs: File read: 2125 bytes
MESS:00:00:04.984568:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.994080:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.003594:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.013102:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.022616:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.032126:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.041640:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.051147:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.060661:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.070171:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.074676:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.081197:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.085942:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:05.092256:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.120024:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:05.124662:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:05.132403:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:05.139853:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:05.601839:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.607482:0: *** Restart logging
MESS:00:00:05.610128:0: brfs: File read: 2125 bytes
MESS:00:00:05.679189:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.689226:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.703032:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.713120:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.723215:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.733306:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.743401:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.753487:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.763583:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.773673:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.778751:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.789404:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.799493:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.809589:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.819676:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.829771:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.839860:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.849955:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.860042:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.870137:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.880227:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.885304:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.890907:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.899663:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:05.905132:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:05.911841:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
MESS:00:00:05.922767:0: dtb_file 'bcm2711-rpi-cm4.dtb'
MESS:00:00:05.935646:0: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
MESS:00:00:05.938932:0: Loading 'bcm2711-rpi-cm4.dtb' to 0x100 size 0xc60f
MESS:00:00:05.977752:0: brfs: File read: 50703 bytes
MESS:00:00:05.990558:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:06.155165:0: brfs: File read: 1667 bytes
MESS:00:00:06.159320:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.163034:0: dtparam: audio=on
MESS:00:00:06.191389:0: brfs: File read: 2125 bytes
MESS:00:00:06.207143:0: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
MESS:00:00:06.369666:0: Loaded overlay 'vc4-kms-v3d'
MESS:00:00:06.816540:0: brfs: File read: 3823 bytes
MESS:00:00:06.820077:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:06.823781:0: Read command line from file 'cmdline.txt':
MESS:00:00:06.829638:0: 'console=serial0,115200 console=tty1 root=PARTUUID=cb5aee81-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
MESS:00:00:06.884008:0: CMA size reduced to 256MB (total_mem 1024MB, gpu_mem 76MB)
[... cut ...] : 148 bytes
MESS:00:00:08.851582:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:08.854142:0: Loading 'kernel8.img' to 0x80000 size 0x1487a00
MESS:00:00:08.881809:0: Kernel relocated to 0x200000
MESS:00:00:08.883656:0: Device tree loaded to 0x2eff3300 (size 0xccb0)
MESS:00:00:08.891476:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:08.898987:0: uart: Baud rate change done...
MESS:00:00:08.901004:0:[    1.427186] hub 2-0:1.0: config failed, hub doesn't have any ports! (err -19)```

geerlingguy, 2022-03-23:
Going to close this issue for now and consolidate graphics card discussions down to the older AMD card issues:

  - https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/222
  - https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/4

ffrediani, 2022-04-24:
Hi @geerlingguy would you be able to download a cryptomining software, compile it for ARM and run it for a couple of minutes using this or any other card in this PCI-E slot to find out how much hashes per second it does ?

I am looking for cheap capex and opex solutions (specially in the terms of power usage) and this could be a good solution. Compiling the miner for ARM should not be an issue, but the main doubt is about the current GPU Linux driver how it would perform.

If you are able to do this for a couple of minutes for any GPU card and cryptocurrency we can have some comparison with a traditional one running in either Windows or HiveOS.

Thanks

geerlingguy, 2022-04-24:
@ffrediani - I do not do any kind of activity related to crypto mining, so would not be interested in testing anything in that realm.

FYI most of that stuff wouldn't work until the drivers are working well enough for other things regardless, and some features miners rely on require parts of the driver we haven't even touched. So it might or might not ever be possible to do any crypto-related work on a CM4 (BCM2711) and GPUs.

And I'd rather it not be, because that would probably contribute in at least some small way to extending the shortage of both Raspberry Pi hardware _and_ graphics cards, both things that I would not like to see.

ffrediani, 2022-04-24:
Ok Jeff, thanks for replying.
The suggested was just a test or a proof of concept, so if it had potential other people interested in using this board for this proposal could consider it.
Yes the main doubt is how mature de available drivers are in this architecture.
With regards the possible shortage of RPi and GPUs I think that is more a question of supply/demand for general usage which vendors have to endeavour to try balance that. At the end the economics will always dictate that.
Thanks anyway.

kerberjg, 2023-04-18:
I'd be really curious to see if it supports V4L2 or VAAPI, it seemed like a really good choice for a Jellyfin cluster ✨

@ObliqueSinter @geerlingguy did perhaps either of you try running `ffmpeg -formats` ?