geerlingguy/raspberry-pi-pcie-devices #336: Test Pine64 SOQuartz on CM4 boards

geerlingguy, 2021-12-05:
Much like I want to test the Radxa CM3, I would also like to test Pine64's [SOQuartz](https://wiki.pine64.org/wiki/SOQuartz) with some CM4 boards, since it's supposed to be pin-compatible.

[image]

[image]

@timonsku mentioned the Wiki (linked above) and [this dtb artifact](https://gitlab.com/pgwipeout/quartz64_ci/-/jobs/1830165844/artifacts/file/artifacts/rk3566-soquartz-cm4.dtb.img.xz) are the two best ways to get started with it. I'd like to write up my experience trying to get the thing to boot, and also seeing if it fits and works in a few popular CM4 boards (starting with the official IO Board).

geerlingguy, 2021-12-09:
So I can get this thing to boot... at least somewhat:

  1. I flashed [this dtb image](https://gitlab.com/pgwipeout/quartz64_ci/-/jobs/1830165844/artifacts/file/artifacts/rk3566-soquartz-cm4.dtb.img.xz) to a microSD card using Balena Etcher.
  2. I popped that microSD card and the SOQuartz onto a CM4 IO Board.
  3. I plugged in the board to my Mac [via UART](https://www.jeffgeerling.com/blog/2021/attaching-raspberry-pis-serial-console-uart-debugging).
  4. I plugged my HDMI display into HDMI0.
  5. I plugged in power to the board.

Using `screen /dev/tty.usbserial-0001 115200` I got gibberish. Also tried `1500000` (which seemed to be a default for the RK3566?), and `9600`, and still got gibberish. But it was getting data, which means something's flowing through.

On the HDMI display, after a few seconds, I see a flashing cursor. So that's impressive, I guess :D

[image]

geerlingguy, 2021-12-09:
Also tested UART with a `FT232R` adapter, but that got nothing at all (not even gibberish) at 1.5 Mbps; see this comment: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/327#issuecomment-990317145

S199pWa1k9r, 2021-12-10:
If you disassemble the used installation image of the SD card into parts and extract the DTB file from it.

dtc -I dtb -O dts rk3566-soquartz-cm4.dtb> rk3566-soquartz-cm4.dts

Convert it to DTS filethen you can see such lines in it:

chosen {
              stdout-path = "serial2: 1500000n8";
  };

This means that serial line 2 is used (from the point of view of RADXA CM3) at a speed of 1500000.
If you do not see normal output, there are two possibilities.

1. You connected the cable not to serial2 but to another serial2 and therefore you see garbage. The on-board console is the RPI CM4 console. And you need to use the console from the RADXA CM3 point of view.

2. Your console cable is simply stray (too long) or the FT232R chip is not original.

The original FT232R cable with Radxa QURTZ64A works for me.

I usually use cables based on the CH304B chip, but they have even more fakes. 
You are doing everything right, you just need to find the error.

Good luck

geerlingguy, 2021-12-10:
@S199pWa1k9r - Thanks for the pointers! For now, I have to finish up a couple other things before the weekend, so I'll just hold off and hope the CH340 adapter is genuine and works out of the box (might also try with CoolTerm).

S199pWa1k9r, 2021-12-10:
Maybe you need to use the GPIO14-RTX, GPIO15-RX pins to connect to the UART console?

It is written here
https://files.pine64.org/doc/quartz64/SOQuartz%20Connector%20Pin%20Assignments%20ver%201.0.ods

geerlingguy, 2021-12-15:
Aha! At @hipboi's suggestion, I installed CoolTerm—it works fine with my existing USB-UART adapter at 1.5 Mbps, and now I can see the SOQuartz's output just fine.

I noticed during boot it paused for about 5 seconds on this prompt, then auto-selected #1 ("Buildroot-recovery"):

```
FIT: No FIT image
Could not find misc partition
ANDROID: reboot reason: "(none)"
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
No valid android hdr
Android image load failed
Android boot failed, error -1.
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:5...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
812 bytes read in 4 ms (198.2 KiB/s)
Quartz64 Installer
1:.Buildroot-recovery
2:.Debian-Installer
3:.Boot Root SDMMC
4:.Boot Root eMMC
Enter choice: 1:.Buildroot-recovery
```

So I rebooted, then made sure to press `2`, then `Enter` to get into the Debian installer.

heh... and now the data coming back expects a full terminal with colors and such:

```
[47m .[31m[!!] Select a language.[30m .(0tqqqqqqqqqqqqqqqqqqqqqqqqk.[6;3Hx.[0m.(B.[30m.[47m                                                                         .(0x.[0m.(B.[1m.[37m.[40m .[7;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m Choose the languagsed for the installation process. The        .(0x.[0m.(B.[1m.[3740m .[8;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m selected language lso be the default language for the installed   .(0x.[0m.(B.[1m.7m.[40m .[9;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m system.                                                                 .(0x.[0m.(B.[1m.[37m.[40m .[10;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m                                                                  .(0x.[0m.(B.[1m.[37m.[40m .[11;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.anguage:                                                            .(0x.[0m.(B.[1m.[37m.[40m .[12;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m                                                                      .(0x.[0m.(B.[1m.[37m.[40m .[13;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m                               C                                         .(0x.[0m.(B.[1m.[37m.[40m .[14;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m                               .[7m.[41mEnglish.[30m.[47m                                   .(0x.m.(B.[1m.[37m.[40m .[15;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m                                                                         .(0x.[0m.(B.[1m.[37m.[40m .[16;3H.[0m.(0.[30m.[47mx.[0m.(B.[30m.[47m     <Go Back>                                                          .(0x.[0m.(B.[1m.[37m.[40m .[17;3H.[0m.(0.[30m.                                         .(0x.[0m.(B.[1m.[37m.[4qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj.[0m.(B.[1m.                                              
.[5B.[44m<Tab> mo.[14;35H.[H.[0m.[7m.[m.[32m.[40m[           .[37m.[31m (.[37m1*installer.[31m).[37m  2 shell  3 shell  4- log           .[32m][.[34m Jan 01 .[37m 0:00 .[32m].[14;35H.[0m
```

geerlingguy, 2021-12-15:
Working through the installer with this formatting is ... fun. But I seem to have gotten most of the way through so far. This was a funny little tidbit:

[image]

[image]

Though I'm hitting an error deeper in:

```
No kernel modules were found. This probably is due to a mismatch between the kernel used by this version of the installer and the kernel version available in the archive.
```

Going to put this on hold until I can figure out a way to interact via TTY that doesn't make my brain hurt in serial console... is there a way?

timonsku, 2021-12-15:
I use Putty for this, gave me color menus and everything

miguemely, 2021-12-15:
Curious, since this is such a non-standard baud rate... could using the "Arduino as a TTL" trick still work here?

timonsku, 2021-12-15:
I went through the installation btw. it did not yield a working system. Bootloader is missing and need to manually call the init system, have not figured that out yet

geerlingguy, 2021-12-15:
@miguemely - Yeah, or a Pico or Pi probably, but as it turns out CoolTerm supports 1.5 Mbps with the cheap USB UART adapter I've been using, so at this point it's a matter of CoolTerm settings so my brain doesn't hurt through the interactive installer :)

miguemely, 2021-12-15:
Perfect... Time to pull the 3 SOQuartz I have out of the shelf now.

miguemely, 2021-12-15:
@geerlingguy Are you using GPIO15-16, or 14-15?

martinx72, 2021-12-15:
@geerlingguy 

here is mine test with the image you mentioned above

[image]

Using putty on Windows 11 with FT232B USB to Serial dongle. GPIO14 as TX and GPIO15 as RX 
[image]

martinx72, 2021-12-15:
[image]

still waiting here. hopefully it would boot after the full installation

miguemely, 2021-12-16:
Alright, Arduino = Bad idea here.

https://www.amazon.com/gp/product/B08TX3KTP1/ worked natively on Windows.
Same with https://www.amazon.com/gp/product/B00LZV1G6K/

[image]

JustCommitRandomness, 2021-12-17:
> If you disassemble the used installation image of the SD card into parts and extract the DTB file from it.
> 
> dtc -I dtb -O dts rk3566-soquartz-cm4.dtb> rk3566-soquartz-cm4.dts
> 
> Convert it to DTS filethen you can see such lines in it:
> 
> chosen { stdout-path = "serial2: 1500000n8"; };

so, can the .dts be edited, compiled, and written into the image to set a more conventional speed?

(not that I can plug my terminal into one of these without frying them)

miguemely, 2021-12-25:
Ok, after full install and telling to to boot to SDMMC:
https://www.toptal.com/developers/hastebin/etiguvixoc.yaml

adminy, 2022-01-24:
Any updates on getting something like a debian os with dropbear preinstalled so that the rest of us without a uart to usb can do this too?

pgwipeout, 2022-02-09:
Rockchip devices default to 1.5m baud because they cannot generate a clean standard baud below that, due to the clock source and limitations of the divider.
@miguemely You were running a quartz64-a image on the SoQuartz which **will damage** the board.

hipboi, 2022-02-10:
> You were running a quartz64-a image on the SoQuartz which will damage the board.

Really? Can you explain more? @pgwipeout

pgwipeout, 2022-02-10:
The Quartz64 and SoQuartz have different PMICs, different regulator layouts, and different power domains.
Loading an incorrect DTS is never a good idea, but specifically with the SoQuartz loading the Quartz64-A DTS leads to a situation where hardware support is half broken and the board gets very hot very quickly.

miguemely, 2022-02-10:
Got it, I'll probably need to re-flash the SD card, grab the SoQuartz and give it another go.

pgwipeout, 2022-02-10:
Also, the SoQuartz dtb is specifically for a CM4 carrier module, there shouldn't be much difference with other carrier modules, but the CM4 is the only one I have to test against.

miguemely, 2022-02-10:
That's the only carrier module I have as well. I just put this on the back-burner since I finally was able to get a hold of two CM4s, but have 3-4 of the SoQuartz that I would love to start tinkering with again.

pgwipeout, 2022-02-10:
You can reach out to me on any of the Pine64 hosted services in the Quartz64 room if you have any questions.

dferrg, 2022-02-22:
Has anyone managed to make PCIe work for the soquartz? I'm testing a board I designed and everything works fine except for the PCIe, which isn't detecting the attached card.

pgwipeout, 2022-02-23:
Yes @dferrg. What board are you using?

dferrg, 2022-02-23:
It's my own design, based on CM4_NAS board by mebs. I connected RX, TX and CLK_REF differential pairs, with length matching for every pair, and nRESET and CLK_nREQ lines. 

As it looks like the problem is with my board, and unrelated to this thread, I'll let a link to an [issue](https://gitlab.com/dferrg/soquartz-nas/-/issues/1) in my repo just in case anyone wants to take a look or give me any hint.

jcdutton, 2022-03-11:
Has anyone found device tree .dts files for this board?
There are multiple parts to the device tree.

    The Pine64 SOQuartz module itself.
    The motherboard the Pine64 SOQuartz is plugged into. (this be will vary if one uses different boards with the module)
    Optional device tree overlays - so one can switch features on/off

Once one has the correct device tree files, I would expect it to work ok with almost any arm64 linux kernel.
If you are playing with .dtb files, you are fighting a loosing battle.

pgwipeout, 2022-03-11:
IRT the rk3566 PCIe controller, it seems we have an issue with cache snooping chip wide. Despite having 1GB of addressable space, dGPUs don't like this. With the radeon based cards, I succeeded at getting a generic console, but RING 0 test fails.

Coreforge, 2022-03-11:
I don't know it off of the top of my head anymore, but the cache issue should be fairly simple to get around, as the bcm2711 had the same issues (plus some other ones).
```
bo->flags &= ~(RADEON_GEM_GTT_WC | RADEON_GEM_GTT_UC);
bo->flags |= RADEON_GEM_GTT_UC;
```
Adding these two lines [here](https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/radeon/radeon_object.c#L222) might be enough, if I haven't forgotten anything. The first line might not be needed if write combining works on the RK3566.

pgwipeout, 2022-03-12:
Sadly it makes no difference, either ring test 0 fails, or we faceplant hard with `bo->flags |= RADEON_GEM_GTT_UC;`
```
[    9.056553] radeon 0000:01:00.0: VRAM: 1024M 0x0000000000000000 - 0x000000003FFFFFFF (1024M used)
[    9.060744] radeon 0000:01:00.0: GTT: 1024M 0x0000000040000000 - 0x000000007FFFFFFF
[    9.066184] [drm] Detected VRAM RAM=1024M, BAR=256M
[    9.070998] [drm] RAM width 128bits DDR
[    9.078439] [drm] radeon: 1024M of VRAM memory ready
[    9.081685] [drm] radeon: 1024M of GTT memory ready.
[    9.086887] [drm] Loading TURKS Microcode
[    9.137734] [drm] Internal thermal controller with fan control
[    9.169945] [drm] radeon: dpm initialized
[    9.179102] [drm] GART: num cpu pages 262144, num gpu pages 262144
[    9.191661] [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0
[    9.313054] [drm] PCIE GART of 1024M enabled (table at 0x0000000000162000).
[    9.313454] Unable to handle kernel paging request at virtual address ffffffc00ab13000
[    9.313469] Mem abort info:
[    9.313470]   ESR = 0x96000061
[    9.313474]   EC = 0x25: DABT (current EL), IL = 32 bits
[    9.313478]   SET = 0, FnV = 0
[    9.313480]   EA = 0, S1PTW = 0
[    9.313482]   FSC = 0x21: alignment fault
[    9.313485] Data abort info:
[    9.313486]   ISV = 0, ISS = 0x00000061
[    9.313488]   CM = 0, WnR = 1
[    9.313491] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000003c5b000
[    9.313494] [ffffffc00ab13000] pgd=10000001fffff003, p4d=10000001fffff003, pud=10000001fffff003, pmd=1000000100aa9003, pte=006800010b6bf70f
[    9.313513] Internal error: Oops: 96000061 [#1] PREEMPT SMP
[    9.313519] Modules linked in: crct10dif_ce hci_uart btrtl btbcm radeon(+) ftdi_sio drm_ttm_helper usbserial ttm fuse ip_tables x_tables ipv6
[    9.313552] CPU: 1 PID: 267 Comm: systemd-udevd Tainted: G        W         5.17.0-rc5-00097-gccb1df4cf6b5-dirty #217
[    9.313559] Hardware name: Pine64 RK3566 Quartz64-A Board (DT)
[    9.313563] pstate: 40400009 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    9.313568] pc : __memset+0x16c/0x188
[    9.313583] lr : radeon_wb_init+0x1b8/0x2a0 [radeon]
[    9.313776] sp : ffffffc00b53b6f0
[    9.313779] x29: ffffffc00b53b700 x28: 0000000000000013 x27: 0000000000000100
[    9.313789] x26: ffffffc00a34a000 x25: 0068000000000f13 x24: 0000000000008c00
[    9.313798] x23: 0000000000008b10 x22: 0000000000008a60 x21: 0000000000028200
[    9.313806] x20: ffffff8108754000 x19: 0000000000000000 x18: 0000000000009a3c
[    9.313814] x17: 0000000000009a38 x16: 0000000000009a34 x15: 0000000000009a30
[    9.313821] x14: ffffff81008a2dd1 x13: ffffff810264cd50 x12: ffffffc00ab27000
[    9.313830] x11: ffffff81fffff2b0 x10: ffffffc04ab13000 x9 : 0000000000000000
[    9.313837] x8 : ffffffc00ab13000 x7 : 0000000000000000 x6 : 000000000000003f
[    9.313844] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 0000000000000004
[    9.313852] x2 : 0000000000000fc0 x1 : 0000000000000000 x0 : ffffffc00ab13000
[    9.313860] Call trace:
[    9.313862]  __memset+0x16c/0x188
[    9.313872]  evergreen_startup.part.0+0xef4/0x2554 [radeon]
[    9.314002]  evergreen_init+0x2a4/0x390 [radeon]
[    9.314130]  radeon_device_init+0x4d0/0xa40 [radeon]
[    9.314256]  radeon_driver_load_kms+0x94/0x1ac [radeon]
[    9.314383]  drm_dev_register+0xec/0x220
[    9.314392]  radeon_pci_probe+0xf8/0x17c [radeon]
[    9.314519]  local_pci_probe+0x4c/0xc0
[    9.314529]  pci_device_probe+0x1b0/0x1f0
[    9.314536]  really_probe.part.0+0xa4/0x310
[    9.314545]  __driver_probe_device+0xa0/0x150
[    9.314551]  driver_probe_device+0x4c/0x160
[    9.314556]  __driver_attach+0x104/0x1a0
[    9.314560]  bus_for_each_dev+0x7c/0xe0
[    9.314565]  driver_attach+0x30/0x40
[    9.314570]  bus_add_driver+0x15c/0x200
[    9.314575]  driver_register+0x84/0x140
[    9.314579]  __pci_register_driver+0x50/0x5c
[    9.314585]  radeon_module_init+0x78/0x1000 [radeon]
[    9.314712]  do_one_initcall+0x50/0x2a0
[    9.314720]  do_init_module+0x54/0x270
[    9.314727]  load_module+0x2050/0x28b0
[    9.314731]  __do_sys_finit_module+0xbc/0x110
[    9.314736]  __arm64_sys_finit_module+0x2c/0x40
[    9.314740]  invoke_syscall.constprop.0+0x58/0xf0
[    9.314747]  do_el0_svc+0x128/0x160
[    9.314752]  el0_svc+0x28/0xe0
[    9.314760]  el0t_64_sync_handler+0x1a8/0x1b0
[    9.314765]  el0t_64_sync+0x1a0/0x1a4
[    9.314774] Code: 91010108 54ffff4a 8b040108 cb050042 (d50b7428)
[    9.314780] ---[ end trace 0000000000000000 ]---
```

jcdutton, 2022-03-12:
@pgwipeout 
That oops is just an unaligned write. A relatively easy bug to fix.

pgwipeout, 2022-03-12:
We suspect cache snooping is completely broken on rk356x. This is indicated by the ITS requiring some rather unsightly hacks to work. https://gitlab.com/pgwipeout/linux-next/-/commit/1443e24c848de7da5431ce6e62d1d6962b7de18a
Rockchip has not provided any insight here unfortunately.
DMA to ram doesn't work above 4GB, due to the rk356x having a 32bit interconnect.
Here is the standard failure:

```
[    9.160745] ATOM BIOS: TURKS
[    9.164489] [drm] GPU not posted. posting now...
[    9.172809] radeon 0000:01:00.0: VRAM: 1024M 0x0000000000000000 - 0x000000003FFFFFFF (1024M used)
[    9.175683] radeon 0000:01:00.0: GTT: 1024M 0x0000000040000000 - 0x000000007FFFFFFF
[    9.178484] [drm] Detected VRAM RAM=1024M, BAR=256M
[    9.180912] [drm] RAM width 128bits DDR
[    9.183475] [drm] radeon: 1024M of VRAM memory ready
[    9.185953] [drm] radeon: 1024M of GTT memory ready.
[    9.188530] [drm] Loading TURKS Microcode
[    9.250828] [drm] Internal thermal controller with fan control
[    9.270952] [drm] radeon: dpm initialized
[    9.287221] [drm] GART: num cpu pages 262144, num gpu pages 262144
[    9.292097] [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0
[    9.428160] [drm] PCIE GART of 1024M enabled (table at 0x0000000000162000).
[    9.428557] radeon 0000:01:00.0: WB enabled
[    9.428577] radeon 0000:01:00.0: fence driver on ring 0 use gpu addr 0x0000000040000c00
[    9.428587] radeon 0000:01:00.0: fence driver on ring 3 use gpu addr 0x0000000040000c0c
[    9.435434] radeon 0000:01:00.0: fence driver on ring 5 use gpu addr 0x0000000000072118
[    9.439887] radeon 0000:01:00.0: radeon: MSI limited to 32-bit
[    9.440248] radeon 0000:01:00.0: radeon: using MSI.
[    9.440372] [drm] radeon: irq initialized.
[    9.441440] radeon 0000:01:00.0: enabling bus mastering
[    9.794807] [drm:r600_ring_test [radeon]] *ERROR* radeon: ring 0 test failed (scratch(0x8504)=0xCAFEDEAD)
[    9.798804] radeon 0000:01:00.0: disabling GPU acceleration
[    9.808046] [drm] Radeon Display Connectors
[    9.810315] [drm] Connector 0:
[    9.812323] [drm]   DP-1
[    9.814265] [drm]   HPD4
[    9.816142] [drm]   DDC: 0x6450 0x6450 0x6454 0x6454 0x6458 0x6458 0x645c 0x645c
[    9.818552] [drm]   Encoders:
[    9.820572] [drm]     DFP1: INTERNAL_UNIPHY2
[    9.822807] [drm] Connector 1:
[    9.824909] [drm]   DVI-I-1
[    9.827083] [drm]   HPD1
[    9.829133] [drm]   DDC: 0x6460 0x6460 0x6464 0x6464 0x6468 0x6468 0x646c 0x646c
[    9.831960] [drm]   Encoders:
[    9.833979] [drm]     DFP2: INTERNAL_UNIPHY
[    9.836065] [drm]     CRT1: INTERNAL_KLDSCP_DAC1
[    9.898000] radeon 0000:01:00.0: [drm] Cannot find any crtc or sizes
[    9.903586] [drm] Initialized radeon 2.50.0 20080528 for 0000:01:00.0 on minor 2
[   10.962008] radeon 0000:01:00.0: [drm] Cannot find any crtc or sizes
```

pgwipeout, 2022-03-12:
Also note, Nvidia cards also fail. Nouveau and Nvidia's closed source driver.
I did get a basic framebuffer on an Nvidia card by hacking an x86 rombar emulator into a VM and passing the PCIe controller through with some more fun hacks, but it would crash the second anything real touched it.

Coreforge, 2022-03-12:
DMA to RAM over PCIe? As long as PCIe devices can access the RAM, it should work. If they can't it'll definitely be a bit more (or a lot more) complicated.

pgwipeout, 2022-03-12:
Ugh, it's the DMA issue.
I've forced the ram to be in the DMA range (less than 4G) and forced uncached and ring test succeeds.

Coreforge, 2022-03-12:
It might be possible to only put part of the ram into the DMA range and make sure GTT buffers are in that range, but that would be a bit more complicated again. Only 1GB or less of overlap should be enough.

pgwipeout, 2022-03-12:
It's actually really quite simple.
Same way to fix the ITS. I'll probably need to make all kernel allocations be in the DMA range to be safe on this board.
It also means the issue is uniquely a pain in the rear on my board with 8G of ram, where 4G and 2G boards will only need to force uncached.

Now to figure out why it's not picking up any crtcs.

```
diff --git a/drivers/gpu/drm/radeon/radeon_device.c b/drivers/gpu/drm/radeon/radeon_device.c
index 4f0fbf667431..ad8fa1692033 100644
--- a/drivers/gpu/drm/radeon/radeon_device.c
+++ b/drivers/gpu/drm/radeon/radeon_device.c
@@ -490,7 +490,7 @@ int radeon_wb_init(struct radeon_device *rdev)
 	}
 
 	/* clear wb memory */
-	memset((char *)rdev->wb.wb, 0, RADEON_GPU_PAGE_SIZE);
+	memset_io((char *)rdev->wb.wb, 0, RADEON_GPU_PAGE_SIZE);
 	/* disable event_write fences */
 	rdev->wb.use_event = false;
 	/* disabled via module param */
diff --git a/drivers/gpu/drm/radeon/radeon_object.c b/drivers/gpu/drm/radeon/radeon_object.c
index 56ede9d63b12..3153d04bca58 100644
--- a/drivers/gpu/drm/radeon/radeon_object.c
+++ b/drivers/gpu/drm/radeon/radeon_object.c
@@ -220,6 +220,9 @@ int radeon_bo_create(struct radeon_device *rdev,
 		bo->flags &= ~RADEON_GEM_GTT_WC;
 #endif
 
+//	bo->flags &= ~(RADEON_GEM_GTT_WC | RADEON_GEM_GTT_UC);
+	bo->flags |= RADEON_GEM_GTT_UC;
+
 	radeon_ttm_placement_from_domain(bo, domain);
 	/* Kernel allocation are uninterruptible */
 	down_read(&rdev->pm.mclk_lock);
diff --git a/drivers/gpu/drm/radeon/radeon_ttm.c b/drivers/gpu/drm/radeon/radeon_ttm.c
index 11b21d605584..e286f92e7d2d 100644
--- a/drivers/gpu/drm/radeon/radeon_ttm.c
+++ b/drivers/gpu/drm/radeon/radeon_ttm.c
@@ -507,7 +507,7 @@ static struct ttm_tt *radeon_ttm_tt_create(struct ttm_buffer_object *bo,
 #endif
 	rbo = container_of(bo, struct radeon_bo, tbo);
 
-	gtt = kzalloc(sizeof(struct radeon_ttm_tt), GFP_KERNEL);
+	gtt = kzalloc(sizeof(struct radeon_ttm_tt), GFP_KERNEL | GFP_DMA);
 	if (gtt == NULL) {
 		return NULL;
 	}
 	```

Coreforge, 2022-03-12:
I assume you have a monitor connected?

pgwipeout, 2022-03-12:
Funny, the displayport -> hdmi adaptor died.

Coreforge, 2022-03-12:
I guess that explains that issue.

pgwipeout, 2022-03-12:
Userspace is mad, crashes at the same spot each time:
[   565.882] (EE)
[   565.882] (EE) Backtrace:
[   565.884] (EE) 0: /usr/lib/xorg/Xorg (OsLookupColor+0x188) [0x559488b4c8]
[   565.884] (EE) unw_get_proc_info failed: no unwind info found [-10]
[   565.884] (EE)
[   565.885] (EE) Bus error at address 0x7f95206304
[   565.885] (EE)
Fatal server error:
[   565.885] (EE) Caught signal 7 (Bus error). Server aborting
[   565.885] (EE)
[   565.885] (EE)

Coreforge, 2022-03-12:
That's the same error I'm getting. Does kmscube work?

pgwipeout, 2022-03-12:
Yup, I have a glorious floating cube at 60FPS (I don't see a way to disable vsync)

pgwipeout, 2022-03-12:
```
root@quartz64:~# kmscube
Using display 0x55848826b0 with EGL version 1.5
===================================
EGL information:
  version: "1.5"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_platform_base EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_KHR_platform_x11 EGL_EXT_platform_x11 EGL_EXT_platform_device EGL_KHR_platform_wayland EGL_EXT_platform_wayland EGL_MESA_platform_xcb EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_create_context_robustness EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 21.2.6"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "X.Org"
  renderer: "AMD TURKS (DRM 2.50.0 / 5.17.0-rc5-00097-gccb1df4cf6b5-dirty, LLVM 12.0.1)"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_OES_viewport_array GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_robustness GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_robustness GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_tessellation_point_size GL_EXT_tessellation_shader GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_copy_image GL_EXT_draw_buffers_indexed GL_EXT_draw_elements_base_vertex GL_EXT_gpu_shader5 GL_EXT_polygon_offset_clamp GL_EXT_primitive_bounding_box GL_EXT_render_snorm GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_buffer GL_EXT_texture_cube_map_array GL_EXT_texture_norm16 GL_EXT_texture_view GL_KHR_context_flush_control GL_KHR_robust_buffer_access_behavior GL_NV_image_formats GL_OES_copy_image GL_OES_draw_buffers_indexed GL_OES_draw_elements_base_vertex GL_OES_gpu_shader5 GL_OES_primitive_bounding_box GL_OES_sample_shading GL_OES_sample_variables GL_OES_shader_io_blocks GL_OES_shader_multisample_interpolation GL_OES_tessellation_point_size GL_OES_tessellation_shader GL_OES_texture_border_clamp GL_OES_texture_buffer GL_OES_texture_cube_map_array GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_OES_texture_view GL_EXT_blend_func_extended GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_EXT_texture_sRGB_R8 GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_clip_cull_distance GL_EXT_disjoint_timer_query GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_EXT_clip_control GL_EXT_color_buffer_half_float GL_EXT_texture_compression_bptc GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_depth_clamp GL_EXT_texture_query_lod GL_MESA_bgra "
===================================
Using modifier ffffffffffffff
Modifiers failed!
Using modifier ffffffffffffff
Modifiers failed!
Rendered 120 frames in 2.000056 sec (59.998315 fps)
Rendered 240 frames in 4.000061 sec (59.999082 fps)
```

Coreforge, 2022-03-12:
That's more than I have on the cm4 then. I'll take a look at the Xorg stuff once I get kmscube to work reliably with more than 1 core enabled.

jcdutton, 2022-03-12:
> Userspace is mad, crashes at the same spot each time: [ 565.882] (EE) [ 565.882] (EE) Backtrace: [ 565.884] (EE) 0: /usr/lib/xorg/Xorg (OsLookupColor+0x188) [0x559488b4c8] [ 565.884] (EE) [...]

"Bus error" is normally an unaligned write or read.
But if kmscube is working, the PCIe bus and the GPU driver is probably OK.
Worth trying the command line version of glmark2 also.
Were there any dmesg messages when the bus error happens?

Coreforge, 2022-03-12:
On the cm4, there isn't anything useful in dmesg that I noticed. I think Xorg does some things over /dev/mem though (?), so there is probably something in there. I might try again later if I can narrow it down a bit.

jcdutton, 2022-03-12:
glmark2-drm    is the non-X command line version. Should work just like kmscube but does more tests.

pgwipeout, 2022-03-12:
Just tested with a 4G board, which shouldn't faceplant with a DMA failure. Same error at the same spot.
So the question is what is `OsLookupColor+0x188` and what is it doing.
No dmesg output at all, but with drmdebug there's a soft reset recovery on the card at the time of the failure.

pgwipeout, 2022-03-12:
Curious:
```
master@soquartz:~$ sudo glmark2-drm
=======================================================
    glmark2 2021.02
=======================================================
    OpenGL Information
    GL_VENDOR:     X.Org
    GL_RENDERER:   AMD TURKS (DRM 2.50.0 / 5.17.0-rc5-00097-gccb1df4cf6b5-dirty, LLVM 12.0.1)
    GL_VERSION:    3.1 Mesa 21.2.2
=======================================================
[build] use-vbo=false: FPS: 60 FrameTime: 16.667 ms
[build] use-vbo=true: FPS: 60 FrameTime: 16.667 ms
[texture] texture-filter=nearest: FPS: 60 FrameTime: 16.667 ms
[texture] texture-filter=linear: FPS: 60 FrameTime: 16.667 ms
[texture] texture-filter=mipmap: FPS: 60 FrameTime: 16.667 ms
[shading] shading=gouraud: FPS: 60 FrameTime: 16.667 ms
[shading] shading=blinn-phong-inf: FPS: 60 FrameTime: 16.667 ms
[shading] shading=phong: FPS: 60 FrameTime: 16.667 ms
[shading] shading=cel: FPS: 60 FrameTime: 16.667 ms
[bump] bump-render=high-poly: FPS: 60 FrameTime: 16.667 ms
[bump] bump-render=normals: FPS: 60 FrameTime: 16.667 ms
[bump] bump-render=height: FPS: 60 FrameTime: 16.667 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 60 FrameTime: 16.667 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 49 FrameTime: 20.408 ms
[pulsar] light=false:quads=5:texture=false: FPS: 60 FrameTime: 16.667 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 30 FrameTime: 33.333 ms
[desktop] effect=shadow:windows=4: FPS: 59 FrameTime: 16.949 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 59 FrameTime: 16.949 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata:Bus error
```

jcdutton, 2022-03-12:
@pgwipeout 
So it "bus error" on the "buffer" one.
sudo glmark2-drm -b buffer
That will just run the "buffer" one.
Just to see if it repeatably fails on that one.

So, there are probably some unaligned write/read left to fix in the driver.
But this is very good progress. Great work.

pgwipeout, 2022-03-12:
```
root@soquartz:~# glmark2-drm -b buffer
radeon: The kernel rejected CS, see dmesg for more information (-16).
=======================================================
    glmark2 2021.02
=======================================================
    OpenGL Information
    GL_VENDOR:     X.Org
    GL_RENDERER:   AMD TURKS (DRM 2.50.0 / 5.17.0-rc5-00097-gccb1df4cf6b5-dirty, LLVM 12.0.1)
    GL_VERSION:    3.1 Mesa 21.2.2
=======================================================
radeon: The kernel rejected CS, see dmesg for more information (-16).
[buffer] <default>:radeon: The kernel rejected CS, see dmesg for more information (-16).
radeon: The kernel rejected CS, see dmesg for more information (-16).
<snip>
radeon: The kernel rejected CS, see dmesg for more information (-16).
 FPS: 60 FrameTime: 16.667 ms
=======================================================
                                  glmark2 Score: 60
=======================================================
```
Nothing in the dmesg, let me turn on drmdebug.

pgwipeout, 2022-03-12:
I apologize in advance for this, here's the last output before it dumps (it dumps at the same place every time).
```
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata:[  288.489559] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_MMAP
[  288.491841] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_BUSY
[  288.492741] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_BUSY
[  288.493514] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_WAIT_IDLE
[  288.494320] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_MMAP
[  288.506527] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_BUSY
[  288.507336] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_BUSY
[  288.508096] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_WAIT_IDLE
[  288.508984] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_MMAP
[  288.518727] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_BUSY
[  288.519765] [drm:drm_ioctl] comm="glmark2-d:rcs0" pid=974, dev=0xe201, auth=1, RADEON_CS
[  288.521283] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_WAIT_IDLE
[  288.522136] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, DRM_IOCTL_MODE_PAGE_FLIP
[  288.522970] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.523655] [drm:radeon_irq_kms_set_irq_n_enabled [radeon]] vblank0 interrupts enabled
[  288.524596] radeon 0000:01:00.0: [drm:drm_vblank_enable] enabling vblank on crtc 0, ret: 0
[  288.524567] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59648, wptr 59664
[  288.525512] [drm:radeon_crtc_page_flip_target [radeon]] flip-ioctl() cur_rbo = 0000000011fbf557, new_rbo = 0000000033003173
[  288.526227] [drm:evergreen_irq_process [radeon]] IH: D1 vblank
[  288.527197] [drm:evergreen_irq_set [radeon]] evergreen_irq_set: sw int gfx
[  288.527991] [drm:drm_mode_object_get] OBJ ID: 61 (2)
[  288.528237] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.528774] [drm:drm_mode_object_put.part.0] OBJ ID: 61 (3)
[  288.529779] [drm:drm_mode_object_put.part.0] OBJ ID: 60 (2)
[  288.530447] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59664, wptr 59680
[  288.531495] [drm:evergreen_irq_process [radeon]] IH: CP EOP
[  288.532209] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59680, wptr 59696
[  288.532269] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.533263] [drm:evergreen_irq_process [radeon]] IH: D1 vblank - IH event w/o asserted irq bit?
[  288.533723] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.534439] [drm:evergreen_irq_process [radeon]] IH: D1 vblank
[  288.535507] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59696, wptr 59696
[  288.547908] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59696, wptr 59712
[  288.549003] [drm:evergreen_irq_process [radeon]] IH: D1 vblank
[  288.549726] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59712, wptr 59728
[  288.550749] [drm:evergreen_irq_process [radeon]] IH: D1 flip
[  288.551472] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.552262] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59728, wptr 59728
[  288.559484] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_CREATE
[  288.561072] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_MMAP
[  288.563469] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_BUSY
[  288.564608] [drm:drm_ioctl] comm="glmark2-d:rcs0" pid=974, dev=0xe201, auth=1, RADEON_CS
[  288.564584] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59728, wptr 59744
[  288.566185] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, RADEON_GEM_WAIT_IDLE
[  288.566243] radeon 0000:01:00.0: [drm:vblank_disable_fn] disabling vblank on crtc 0
[  288.567036] [drm:drm_ioctl] comm="glmark2-drm" pid=973, dev=0xe201, auth=1, DRM_IOCTL_MODE_PAGE_FLIP
[  288.567689] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.568931] [drm:radeon_irq_kms_set_irq_n_enabled [radeon]] vblank0 interrupts disabled
[  288.569887] [drm:evergreen_irq_process [radeon]] IH: D1 vblank
[  288.569895] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.571089] [drm:radeon_irq_kms_set_irq_n_enabled [radeon]] vblank0 interrupts enabled
[  288.572039] radeon 0000:01:00.0: [drm:drm_vblank_enable] enabling vblank on crtc 0, ret: 0
[  288.572914] [drm:radeon_crtc_page_flip_target [radeon]] flip-ioctl() cur_rbo = 0000000033003173, new_rbo = 0000000011fbf557
[  288.574193] [drm:evergreen_irq_set [radeon]] dpm thermal
[  288.575037] [drm:drm_mode_object_get] OBJ ID: 60 (2)
[  288.575514] [drm:drm_mode_object_put.part.0] OBJ ID: 60 (3)
[  288.576026] [drm:drm_mode_object_put.part.0] OBJ ID: 61 (2)
[  288.581241] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59744, wptr 59760
[  288.582335] [drm:evergreen_irq_process [radeon]] IH: D1 vblank
[  288.583053] [drm:evergreen_irq_process [radeon]] evergreen_irq_process start: rptr 59760, wptr 59776
[  288.584076] [drm:evergreen_irq_process [radeon]] IH: D1 flip
[  288.584803] [drm:evergreen_irq_set [radeon]] dpm thermal
[... cut ...]
[  290.332275] [drm:drm_crtc_helper_set_config] [CRTC:52:crtc-5] [NOFB]
[  290.332862] [drm:radeon_atom_encoder_dpms [radeon]] encoder dpms 33 to mode 3, devices 00000008, active_devices 00000000
[  290.333980] [drm:radeon_atom_encoder_dpms [radeon]] encoder dpms 21 to mode 3, devices 00000001, active_devices 00000000
[  290.335576] [drm:drm_mode_object_get] OBJ ID: 58 (2)
[  290.336039] [drm:drm_mode_object_put.part.0] OBJ ID: 58 (3)
[  290.336612] [drm:drm_release] driver lastclose completed
Bus error (core dumped)
```

jcdutton, 2022-03-12:
@pgwipeout 
At least the problem is predictable. :-)
You will probably need to compile glmark2 in debug mode, then run it under gdb to find out which line the bus error is on.
Hopefully it will narrow it down to a syscall to the driver, and then we will know which function in the driver code has the alignment problem.

Coreforge, 2022-03-12:
The buffer benchmark by default uses "map" as the vbo update method. I haven't looked very closely into glmark2s code, but my guess is that it maps the buffer on the GPU into userspace and accesses is that way, while not making sure alignment is correct. I don't know if there is a way to enforce alignment on mapped memory from the kernel though, or if all userspace software would have to be changed.

Coreforge, 2022-03-12:
Try running `glmark2 -b buffer:update-method=subdata` instead

jcdutton, 2022-03-12:
If it helps, one can force alignment checking on x86 and even get the compiler to check for it with -Wcast-align
You add an extra line of inline asm code to switch it on around the section you wish to check.
An example page explaining it:
https://www.xszz.org/faq-41/question-20190628214421.html

[161 more replies]